Ralim/IronOS #1242: Pinecil reboots on usb-C 65W PD  , using 150W usb-C PD-charger

River-b, 2022-02-24:
Pinecil reboots when plugged into the 65W port of this device. Does not happen when plugged into the 100W port.
I can repeat this everytime on two different Pinecil. One pinecil has 2.17  and the other has the Beta new animation logo firmware.

I could make a video if needed.  It happens within first ~10 sec of starting the device.

 - Power adapter being used:
 3-prong grounded 150W USB-C charger. This does not happen on the 100W ports, but only on the PD65W port. 
 Charger comes with 3 Usb-C ports. the two 100W ports work fine. the single 60W/65W PD port  the Pinecils reboot within the first 10-15 seconds, the screen turns black, but then comes back and after I press some buttons, it seems to work. 
- Power cable is Pine64 60W usb-c cable.
- same cable and pinecils have no issues in my other PD65W charger. 

This is same charger on Amazom, Hunda & Huwder are same company. 
https://www.amazon.com/Charger-HUNDA-Delivery-Charging-Compatible/dp/B08THBXQJK

I have the Huwder one but it is same, bottom 3 ports are all PD,  65W, then 2x 100W PD. 
https://huwder.com/products/3c-1a-150w-pd30-pps-charger-station

Ralim, 2022-02-24:
To confirm; a video may be good to know what you mean by pressing a button makes it come back on, I would expect this to be a pd issue but that doesn't sound like one?

River-b, 2022-02-24:
the time of the reboot varies, 6 sec sometimes 10 sec. . After the  initial reboot, Pinecil seems to work, but I haven't used it more than a few minutes after the test.  

sometime when I plug it in ,  it will not reboot until I press the +  button but I press it withing the first 15-25 sec I think. I will make a video now

I also noticed that one time I did not get a reboot , but it was 1 out of maybe 25 times I plugged it in to see behavior.

River-b, 2022-02-24:
do you think this is a Faulty  USB-C charger, I just received it today, supposed to be high end, it's not one of the cheap ones.

River-b, 2022-02-27:
https://www.youtube.com/watch?v=WgZ6CrsbtUA

River-b, 2022-02-27:
I tried 5 different cables and I flipped the cables over too. it reboots almost everytime  using the Pine64 usb-C cable, and another certified 60W  C-C cable, and a 100W C-C cable.

River-b, 2022-03-01:
I have a USB PD trigger now. it is the FNB48 device.

Ralim, 2022-03-01:
Looking through this video, I'm really starting to suspect that this power supply is just not happy at all.

I would try with the trigger board, but if you see that intermittent behaviour its definitely the supply.

Even without the trigger board looking through this I think its the power supply, wondering if the supply has a bug with _its_ PD implementation or if its a factory issue (like bad soldering so it fails under load).

Seems to only be buggy with negotiation part though, so maybe just a poor PD controller or weak solder joint on a CC pin (or both) ?

River-b, 2022-03-01:
I have a PD Trigger now and a couple USB testers. if you are around on Chat channel later, could help me set up testing. I would like to see if this is Firmware on the charger (not done to PD specs )  or if it's not, then I could maybe open it up to look for some places to reflow.  this is a heavy charger, it's not GaN.

River-b, 2022-03-10:
@Ralim   there was a suggestion that maybe this could be problem with its PD output method, and somehow caused by its power distribution?  therefore would this be something I can not fix at all  on this Power supply?

closed 2022-08-05 by River-b