Ralim/IronOS #1243: Pinecil reboots on weak DC power supply

adokitkat, 2022-02-26:
**Describe the bug**
Pinecil (2.17 European firmware) reboots constantly when plugged into USB-C (Thunderbolt 3) port on a laptop. This didn't happen before the last update. I know the power from the USB port is not enough for soldering, but it was nice to be able to change settings at least.

**To Reproduce**
1. Update Pinecil to firmware version 2.17
2. Connect Pinecil to USB port on a PC/laptop (USB-C port in my case)
3. Pinecil reboots and screen flashes

**Expected behavior**
Pinecil to boot up normally as it did before. 

**Details of your device:**
 - Device: Pinecil
 - Release: 2.17.E065BE3
 - Power adapter being used: USB-C Thunderbolt 3 port on a laptop

River-b, 2022-02-27:
(1) when you switch back to 2.16  does the constant reboots  while connected to 5V PD on laptop stop?  
(2) when you say constant  do you mean pinecil is rebooting over and over many times  when  2.17 is loaded?
(3) are you able to test 2.17  on other PD 5V devices to confirm such as Android phone with PD usb-c  or another laptop/PC ?

Ralim, 2022-02-27:
Hello,
You have not provided exact details about what device you are using for the power, PD is controlled by the OS and the hardware, so both can change how this works out. 

There are _lots_ of variations of PD on desktop and a lot of them cause issues because they keep trying to power cycle the device because "it isnt working" because we actively reject USB, we also reject power swap and a bunch of others.

I can not reproduce this on the 3 computers I have with USB-C, so suspect this may be down to differences with your device.

adokitkat, 2022-03-05:
Hi again @Ralim and @River-b.

So I've tested it and here's what I've found.

Pinecil with IronOS 2.17 flashed via Pine64 updater, 3 different USB-C cables and 1 USB-A to C cable used:
* connected to Dell G5's Thunderbolt 3 port running Windows 10 - 5V, rebooting and screen flashing
* connected to ThinkPad E14's only USB-C port (also used as a charging port) running  Pop!_OS 21.10 Linux (Ubuntu based) - 5V, rebooting and screen flashing
* the same ThinkPad E14 but Pinecil connected to USB-A port - 5V, **no rebooting and flashing, works as expected!**
* connected to Google Pixel 6 Android smartphone- 5V, works as expected
* connected to 45W and 65W USB-C PD chargers - 12V, works as expected

I didn't try to downgrade to 2.16 yet, however before on 2.16 Pinecil connected to Dell G5's Thunderbolt port on Windows 10 worked well.

Ralim, 2022-03-06:
Hi,
So I use a Lenovo T14 laptop for a fair bit of the development and I've never run into this issue. I suspect this may be to do with TB as I use only AMD machines.
Screen flashing like that is the upstream port forcefully power cycling the device as a hard reset.

Can you confirm (a) this is reproducible on the English build (not multilingual) of 2.17, (b) test on 2.16 to check it doesnt occur.

Would be good to try and make a guess as to the issue, as I dont have any thunderbolt capable devices to test on available at the moment.

dmitrygribenchuk, 2022-03-28:
Hello. I have the same issue with Lenovo E14 (Intel Core) and plugin into type C socket.
Looks like a bootloop.
Also have it on 2.16 RC2, 2.15 RC1. I have not found a version witch works correctly.
Check the cable and socket on my phone - quick charge works correctly. 
Connect to another power supply (or USB port of this laptop) - all working fine.

https://user-images.githubusercontent.com/15670247/160390060-d5ac466f-cea7-46ae-b027-ce8752f0cc4d.mp4

dmitrygribenchuk, 2022-04-21:
UPD. All working fine on 2.15
[image]

Ralim, 2022-04-27:
I have not been able to reproduce the issue.
Further diagnostics would require a pd trace of what the laptop is rejecting.

Most likely cause is the laptop is trying to get the Pinecil to negotiate something that it's refusing to.

dmitrygribenchuk, 2022-04-28:
@Ralim I can provide my laptop via Anydesk to investigate this issue for example.

Ralim, 2022-05-01:
It requires hardware to capture the USB PD transaction that your device is rejecting.

cbagwell, 2022-05-01:
I am seeing same reboot behavior on a 2018 MacBook Air USB-C.  It wouldn't reboot when hooked to laptop when using stock Pinecil firmware but started rebooting after upgrade to 2.18.

Similar behavior as videos but I do not get the double blink of first video and just a reboot every 1s.  Voltage toggles between 5.0V and 5.1V.  If I hold + button, I see it print PD State 6 just long enough to read and then switches to state 11.  That print lasts about about 0.5 second and then reboots.

I upgraded my firmware because the stock firmware would constantly reboot when using my 30W MacBook Air power adapter and that was much more importable issue for me to solve; and it did solve it btw.

Ralim, 2022-05-08:
Solving compatibility to the macbook chargers is what happened before these issues with thunderbolt usb-c ports showed up.
I haven't been able to borrow a macbook to test this with so I dont have any further information on this at the moment.

damex, 2022-05-20:
> It requires hardware to capture the USB PD transaction that your device is rejecting.

what hardware is needed for that and what are we looking for?
could 'usb tester' like https://www.chargerlab.com/power-z-kt001-usb-power-tester-with-pd-qc-trigger-and-mfi-identification/ suffice for the job? (I will need to get something compact and obtainable from Aliexpress)

BF3000, 2022-08-11:
Pinecil v1 updated to 2.19. Constantly reboot itself after 10s in qc or simple usb 5v modes.
Work in qc mode if I enable tip heating, until it heating stops .
Work normally with pd or barrel jack.
And yes both pd and qc power comes from Baseus 65w charger, one of the best things around.

BF3000, 2022-08-11:
Looks like Advanced settings - Power pulse - something higher than zero fix the problem.

BF3000, 2022-08-11:
Looks like Power pulse 0.2w or higher can partially fix the problem.

Ralim, 2022-08-11:
@BF3000 
That sounds more like a power supply issue. What power adapter are you using?

BF3000, 2022-08-11:
Baseus GaN 65w
One of the best available.
And the problem is, it work the same way if connected to PC.
Looks like my PInecil constantly wakes itself with that option.

River-b, 2022-08-11:
Not that this is the issue, but wanted to point out that you could get 3 models of the same 2-prong USB charger and 2 work fine, and one has a little too much current leak or other issue where they will charge a Phone fine for example, but on Pinecil cause reboots due to  enough current leak that pinecil is sensitive to. Some of this can be eliminated using a 3-prong grounded charger. All 2-prong usb chargers are allowed to  be sold even though they have current leakage and no ground prong to send that current to because they are wrapped in plastic for what is called Double Insulation.

> Baseus GaN 65w One of the best available. And the problem is, it work the same way if connected to PC. Looks like my PInecil constantly wakes itself with that option.

River-Mochi, 2023-02-25:
My  Lenovo Legion 5P laptop, both of the USB-C ports cause pinecil to reboot so I can not use them to do firmware updates. I have to use the USB-A ports on the laptop to update Pinecil V2.
Another Pine64 member reported the same behavior on the exact same laptop.

Ralim, 2023-02-26:
@River-Mochi 

Can you capture the USB PD traffic please for a failed negotiation?

River-Mochi, 2023-02-28:
> Can you capture the USB PD traffic please for a failed negotiation?

1. KM002 software and firmware is updated now so it's slightly different from past.
2. could you tell me what settings would be best for "**Data Recorder**" mode?
3. Do you also want the **PD Analyzer** recorded which is separate?

### Per #2:
[image]

### Per #3
[image]

Ralim, 2023-03-01:
Only need the raw PD messages; which show in the PD analyzer tab.

Patronics, 2023-10-23:
I'm getting the same issue with my M1 Macbook Pro's USB C connection, sometimes (but not always) when plugging it in. For me I didn't see this issue when first plugging it in (with firmware 2.20 preinstalled), but that may have just been the randomness since it doesn't always happen). I tried both updating to 2.21 and to 2.22 RC 2, the issue was the same on both of them. 

Plugging it in while holding `-` to do the update seems to consistently work, at least (and after reinstalling the firmware update the interface is usable). The USB C cable I'm using is 6 feet long, but the pinecil (with "detailed idle screen" enabled) reports 5.2 volts so it doesn't seem to be a voltage drop issue.