Ralim/IronOS #1270: Acceleration Sensing not working on TS80P

TheUnlimited64, 2022-03-27:
**Describe the bug**
I've got a new TS80P from miniware and I recently flashed IronOS onto it. I noticed the sleep mode turning on even when im actively using it, so I tried to adjust the sensibility to 1 and 9. Nothing changed and shaking the device pretty aggressively, but it didn't woke up

**To Reproduce**
Normal using?

**Expected behavior**
When shaking/using the Iron it should be awake

**Details of your device:**
 
 - Device: TS80P
 - Release: 2.17.E065BE3
 - Power adapter being used: 45w PD Powerbank (Ravpower), 90w PD (Ravpower), 65w PD (Huawei)
 - If this is an accelerometer related issue, please include its model number here: MTPUA00523 (?)

bkeinert, 2022-03-27:
Very likely related to/duplicate of https://github.com/Ralim/IronOS/issues/1229

Ralim, 2022-05-08:
Please update to 2.18 release and check if it is working out better for you.

closed 2022-07-03 by Ralim