Ralim/IronOS #1273: device not shutting down if timeout ends before sleep temp is reached

discip, 2022-04-02:
**Describe the bug**
The device does not enter *shutdown mode* if the **`shutdown timeout`** expires before the **`sleep temperature`** is reached.

**To Reproduce**
1. Set **`sleep temperature`** to **90 °C**.
2. Set **`shutdown timeout`** to **1M**.
3. Heat up your device to **320 °C**, eventually move it to prevent it from falling asleep. 😊 
4. When the temperature is reached, let the device sit without being moved.
5. Now watch it entering the **`sleep mode`** but after the **`shutdown timeout`** expires, it goes back to the main screen without shutting down.

**Expected behavior**
Device should regardless of reaching the **`sleep temperature`** shut down after **`shutdown timeout`** expires.

**Details of your device:**

iron  |  Firmware Version  |  ACC  |  Power Supply
-|-|-|-
**TS80P**|**https://github.com/Ralim/IronOS/commit/0b8192b690145e9b5f2675e627fc3836a864839d**|**LIS2DH12**|**[Anker Nano II 45W](https://de.anker.com/collections/wandladegerate/products/a2664311)**

Ralim, 2022-04-02:
I'm not 100% following here;
I would expect:
1. Start heating up
2. Sleep timer expires
3. Device enters sleep mode
4. Shutdown timeout expires
5. Device shuts down.

The firmware doesn't support skipping 2/3.

What order of events are you seeing?

discip, 2022-04-02:
> 1. Start heating up
> 2. Sleep timer expires
> 3. Device enters sleep mode
> 4. Shutdown timeout expires
> 5. Device shuts down.

Yes that is what one would expect.
But unfortunately this is not working like that , if the conditions *specified in the opening post* are met.

> The firmware doesn't support skipping 2/3.

Actually these steps are not skipped. They are executed as intended.
But instead of executing 5 the device just goes back to the main screen after 4 occurs.

I don't know, if you followed the *reproduction procedure* above, but as I pointed out earlier, I suspect that the device does not shut down due to the fact that it takes longer to cool down to the desired **`sleep temperature`** than the **`shutdown timeout`** lasts.

sandmanRO, 2022-04-21:
Hello Gentlemen,
I'm so glad to see the project is still going strong. Regarding this issue, the fact that the iron does not shutdown while the iron is still hot is not a bad thing. Nevertheless, it should shutdown eventually when it's safe to do so. The scenario is like this:
1. We heat up the tip then we let it to rest.
2. After the sleep timeout (the sleep countdown starts from the last button action/movement) the iron goes to sleep, that is, the tip temperature goes towards the preselected sleep temperature.
3. Either the iron is in sleep mode (the temperature could be anywhere between sleep temperature and boost temperature) or in idle mode (the temperature could be anywhere between room temperature and boost temperature as the idle could be induced by user at any moment), the shutdown's countdown starts and when it times out it terminates the sleep/idle stage, that is the tip will start cooling down from whatever temperature is towards the room temperature. HOWEVER, the iron will not fully shutdown until the tip temperature drops down below a safe temperature (let's say 45 deg Celsius). Until then the display will keep showing the actual tip temperature - during this time the iron could appear it is in idle mode when in fact it is just waiting for the tip to cool down to a safe temperature. Once the temperature is under the safe threshold the display turns off and the iron goes in full shutdown.
Of course, we are assuming the iron is not moved during this time as any movement would cancel the shutdown.

Ralim, 2022-04-21:
So my issue here is I haven't been able to replicate an issue.
Which makes me suspect that we have different ideas of the operation of the iron.

For me following those steps I observe the following:

1. Enter soldering mode
2. Unit heats up to 320C
3. Unit enters sleep (Shows ZZzzzz)
4. Unit then shuts down (can see it return to the idle screen)
5. Unit then slowly cools down
6. Once unit falls below 50C, the oled turns off.

This is to me, the desired operation? We absolutely want the screen on until we fall below the 50C threshold for "may hurt if touched".

Which I think is the same as you @sandmanRO ? (tell me if I'm wrong)

discip, 2022-04-21:
@Ralim 
Have you set the sleep temperature to **90°C** and the shutdown timeout to **1 minute**?

sandmanRO, 2022-04-21:
Hello @discip, you say the iron never shuts down even after the tip temperature drops under 50 Celsius (disregarding how long it takes the tip to cool off)?

sandmanRO, 2022-04-21:
When I say "shutdown" it means the iron is in idle stage with the display turned off.

discip, 2022-04-21:
@Ralim
> 1. Enter soldering mode
> 2. Unit heats up to 320C
> 3. Unit enters sleep (Shows ZZzzzz)
> 4. Unit then shuts down (can see it return to the idle screen)
> 5. Unit then slowly cools down
> 6. Once unit falls below 50C, the oled turns off.

After having checked for the described behavior, I got the point that this behavior was intended. 🤦‍♂️😓
@sandmanRO Thanks. 👍
## So no issue at all! Sorry for the inconvenience!

closed 2022-04-21 by discip

Ralim, 2022-04-22:
No problem at all.  Thank you both :)
Love being questioned, since what is logical for me sometimes isnt for others 😓 And I've definitely made mistakes before.

3eGo, 2022-04-23:
i have a problem with the firmware of the ts100.
in the latest version you have added the time of the bootup logo which can also be infinite.
when i set that setting i get no result of the logo staying on screen all the time or at least until i press a button.
am I doing something wrong or is it not working correctly? thank you very much in advance of the reply

discip, 2022-04-23:
@3eGo
Please open a new issue regarding your concern.

Unfortunately I do not have the TS100, to be able to help you with this.
Maybe someone else owning the TS100 is willing to help you out.