Ralim/IronOS #1277: Undervoltage

yueduz, 2022-04-19:
[image]
Every time I turn it on, it prompts: "Undervoltage", and it needs to press the "-" key twice to start working. And I measured the input voltage with a multimeter to be 4.966 V. I think he is too strict about the voltage requirements. And I wish this prompt could be turned off.

Ralim, 2022-04-19:
Operation at 5V is generally not supported.
It will let you if you wait out the timeout, but it wants you to wait for voltage negotiation to complete before starting soldering.

What power supply and firmware version are you using? (The information you cut out of the template)

yueduz, 2022-04-19:
[image]
[image]

Ralim, 2022-04-21:
Aha, so that that looks like its a Quick Charge power adapter which is not recommended as soldering at 9V is very slow.
As Quick Charge takes a minimum of 1.5 seconds to negotiate; trying to start soldering before the negotiation has finished will indeed return the error that you are trying to solder on too low of a voltage. That is by design; as soldering on 5V is not recommended.

yueduz, 2022-04-21:
My charger can support up to 12V. Then he should wait until the negotiation is completed before starting welding, why does he automatically start when the negotiation is not completed?

Ralim, 2022-04-21:
It will start if either you have turned on auto-start mode in the settings or you press the button at the front of the device to start soldering.

yueduz, 2022-04-21:
After I turned off the auto start, I looked at the time from 5V to 12V, and he was almost exactly 5 seconds. So can PD timeout provide a longer timeout value?

Ralim, 2022-04-21:
QC does not start until PD has timed out.
If you are going to only use the older QC standard then you can set the PD timeout to 0 to start QC earlier

yueduz, 2022-04-21:
I set it to automatically start welding mode on power up and set the PD timeout value to 1. I no longer get the undervoltage prompt and everything works fine. But I don't know why. Set a longer timeout value and get an undervoltage prompt instead.

Ralim, 2022-04-21:
With a PD timeout set to be short you will negotiate QC fast enough to not run into the issue.

yueduz, 2022-04-22:
> With a PD timeout set to be short you will negotiate QC fast enough to not run into the issue.

But when I set the pd timeout to 0, he starts at 5V, then after a while he says the heating is out of control. He doesn't seem to negotiate qc.

You might think this is not a problem. But I think he should only prompt for air pressure when both PD and QC negotiation fails.

If you think this is not a bug, you can close the issue.

River-b, 2022-04-22:
This looks like a Clone  Pinecil. where did you get this Pinecil was it from AliExpress and not official Pine64.com or pine64 official reseller?   was it Black color to begin with but you switched to Red?   all authentic Pinecil come in Black color  not all red (which can be purchased as separate case).

haarp, 2022-04-22:
Is it possible or sensible to actually allow operation at 5V? I'm aware that this would result in only 3W of power output. Can it reach soldering temps with that power? Would be nice to have that option in an emergency (i.e. broken power bank, all you have is a shitty phone charger)

River-b, 2022-04-22:
need at least  3amps to get most out of Pinecil. 
12v 1.5amp is on the low end.   also I think you might have a fake/clone pinecil. 
if you are a member of Pine64 Discord  at all, look in Pins of   Pinecil channel.  I put   list of different power paramenters that work best for pinecil and other helpful  links for power supply questions. 

You can not use a Phone charger 5V  to operate  Pinecil.  this is only good enough for maybe  updating firmware  on Phone connection or PC usb-c connection.

River-b, 2022-04-22:
**Ways to Power Pinecil & get Maximum Performance**

● Usb-C port      - usb-c  cable to charger that is labeled  **PD** 65W or greater, minimum 3amps.

   For DC barrel jack - Check Polarity, some chargers have it reversed on the jack which could damage your device 
● DC5525 port - old laptop charger  on back 18v-21v,  min. 65W+,  min 3amps + .  if it doesn't have a DC 5.5mm/2.5 jack, can often buy adaptor to convert
● DC5525 port - variable voltage laptop charger  24V 3A ,  min. 65W or more,  has dial to change voltage, set to 21V until you do the special MOD  (after mod of cutting trace on PCB, you can crank to 24V safely) 
● DC5525 port - any 18V-21V tool battery,  make your own adaptor to the battery or simply buy 'Power Wheels' adaptor for your brand battery, splice DC5525 jack to it, works great & portable.
● Pinecil Wiki - https://wiki.pine64.org/wiki/Pinecil#Power_Supply_Compatibility ,  it will work at lower voltages but takes more time rise to Temp & maintain temp
● How to measure DC jack - https://www.digikey.com/en/blog/barrel-connector-measurement-tricks (edited)

River-b, 2022-04-22:
> Is it possible or sensible to actually allow operation at 5V? I'm aware that this would result in only 3W of power output. Can it reach soldering temps with that power? Would be nice to have that option in an emergency (i.e. broken power bank, all you have with you is a shitty phone charger)

if you want an alternative that is not listed in the wiki, just experiment and you will see.  but also one of the best emergency alternative   is using  the 18v-21v tool batteries that people have around and splicing it into   DC 5525 jack .  Can use Power Wheels  adapter made for specific  brands of tool batteries and a DC 5525 jack. 
18V  tool batteries like Ryobi/Rigid  heat up very fast and work well - plenty of  Power Wheel adaptors on Amazon/Ebay  to convert it to use for Pinecil  or make your own adapter,  BigClive has a video on how to make your own adapter.

River-b, 2022-04-22:
Also,  Clone/fake  Pinecil  are not supported by Pine64.  the hardware is made from unknown source and suspect.  they have copied all Pine64 and Pinecone logos on the fakes.

River-b, 2022-04-22:
> [image] Every time I turn it on, it prompts: "Undervoltage", and it needs to press the "-" key twice to start working. And I measured the input voltage with a multimeter to be 4.966 V. I think he is too strict about the voltage requirements. And I wish this prompt could be turned off.

The Minimum voltage on  official Pine64 for the Pinecil device  is   12V.     see the wiki for this.   This is a soldering iron that you need to actively keep temperature  up and not a device like a phone battery that you could wait for hours to charge up.  Therefore  getting a charger that can support  as close to 21V  as possible is ideal   but at minimum,  12V usb  is doable but  painful because every time you touch the solder the temperature will drop and you have to wait for temp to go back up slower. 

The reason your 5V charger does not work is that  it is not the  correct device for Pinecil.  There are many USB chargers out there and generally,  those small tiny phone chargers are not made for Pinecil.  However ,   USB-C  laptop chargers generally are made with right power   look for ones that say  PD 65w  (and 3amp or 3.25 amp).  The PD  protocol is important (PD = Power delivery protocol). 
https://wiki.pine64.org/wiki/Pinecil#Power_Supply_Compatibility

Pine64 pinecil  was never intended to work with weak  5V phone charger supplies because they can not pump out enough power  per second to keep up with demands of soldering iron. 
It would be like asking why does my car not run on a 9V battery  - because car is not rated for 9v , and many cars can not even run on most 12V batteries, you have to get a battery that is rated for the size of the car  and  some 12V batteries are more powerful than others.

Pinecil is Rated to run on **12V -21V**   (aiming for 3amps or more if possible).   if you experiment  and get it somehow working with anything less, that would be hobbling the pinecil and not supported.   Pinecil will also work with 24V  DC chargers if you first make a  small modification to the PCB - this is detailed in the Feb 2022   Pine64  Newsletter.

yueduz, 2022-04-22:
i buy here ：https://item.taobao.com/item.htm?spm=a1z09.2.0.0.29ba2e8dZmoRCn&id=633955322505&_u=n2cvdot5990

River-b, 2022-04-23:
> i buy here ：https://item.taobao.com/item.htm?spm=a1z09.2.0.0.29ba2e8dZmoRCn&id=633955322505&_u=n2cvdot5990

pretty sure that is  fake clone Pinecil.  all real authentic Pinecil are  only made in Black color with blue finger grip.  
Discussion section of Github, see article I wrote about fake pinecil. [link](https://github.com/Ralim/IronOS/discussions)

there are no authorized resellers of Pinecil on  Aliexpress or those types of places. if you are in china then you could have ordered directly from Pine64.com since it ships from hong kong I believe.   The Fake Clone sellers are now  all over place. I stopped counting but last I checked over 25 places were selling clone fakes.

closed 2022-04-23 by yueduz