Ralim/IronOS #1283: Faster sleep mode in certain orientations

haarp, 2022-04-22:
Hello,

I love this firmware and have been using it for years now. Thanks for your work! After starting to make use of a soldering stand, an idea came to me that might, or might not be worth the effort.

Right now, sleep is based on a movement timer. This works, but it needs comparably long timers to avoid false positives. But the iron can also measure its orientation, so what if the iron "knew" when it was put away based on that? e.g. Lying flat on a table, or at a certain angle in a stand. We could then reduce the movement timers as long as the orientation is held.

This could result in small energy savings (for battery users) and tip life improvement. Also it would be kinda cool for the iron to almost instantly sleep/wake up when put away/picked up.

We'd needs two new config menu entries:

- Fast Sleep orientation: Put the iron in your soldering stand or flat on the table and press a button to set this angle
- Fast Sleep timer reduction: Factor by which sleep (and shutdown?) timers are reduced. 1(off), 1/2, 1/4, 1/8, 1/16

Thanks!

discip, 2022-04-23:
Similar to #813.

River-b, 2022-04-23:
This is a slightly separate suggesiton but do you have Pinecil?  if it's a Pinecil, you have the option of installing a low cost Hall Effect sensor to U14 spot on the PCB.  look at the Hall sensor documentation in  documentation folder  [Link here](https://github.com/Ralim/IronOS/blob/master/Documentation/HallSensor.md) 

Hall sensor will expand your sleep options even more. by  using a neomydium magnet you place on your  holder stand, it will cause the Pinecil to auto ZZZzz  sleep when it is store in the stand. but it is only an option if you have a Pine64 Pinecil Iron as this option is not for miniware products.

But yes these orientation potentials are interesting too.