Ralim/IronOS #1285: not working infinite logo on the ts100.

3eGo, 2022-04-23:
i have a problem with the firmware of the ts100.
in the latest version you have added the time of the bootup logo which can also be infinite.
when i set that setting i get no result of the logo staying on screen all the time or at least until i press a button.
am I doing something wrong or is it not working correctly? thank you very much in advance of the reply

federicodilo, 2022-04-23:
In order to get your options effective you need first to exit to the main screen.
It should work for you too.

3eGo, 2022-04-24:
yes but I've always done it that way.
then when I return to the settings it shows me that it is selected on the infinite logo mode.
the problem is that the setting just doesn't work unlike the others.

federicodilo, 2022-04-24:
I find this quite strange since it works as intended on my TS100.
Are you using a custom animation? Have you tried other animations/static images?
Can you please provide a video of the problem?

3eGo, 2022-04-24:
no i am not using a custom animation and i also tried to change other static imagesit but nothing, it just keeps not going.
now I try to make a video with all.

3eGo, 2022-04-24:
https://user-images.githubusercontent.com/104205730/164997023-dc4ba6c2-4500-437a-ad19-033e51e554c3.mp4

discip, 2022-04-25:
@3eGo 
Maybe you could try reflashing the firmware.

federicodilo, 2022-04-25:
Curiously enough it lasts 5 s, I suspect it depends on how the timers are internally managed.
Maybe @Ralim has some ideas about it. Can you please take a photo of the inside?

3eGo, 2022-04-25:
what i have to do?

3eGo, 2022-04-25:
I also tried to put other firmware, change versions and more, but nothing has ever changed

Ralim, 2022-06-08:
Can you please try the firmware from https://github.com/Ralim/IronOS/actions/runs/2460121981 when you have time 😁

discip, 2022-06-11:
@3eGo
Please respond to the developer’s request.

And could you also please test this version  https://github.com/discip/IronOS/suites/6892990972/artifacts/267234503?

thanks in advance

discip, 2023-01-05:
@3eGo
Are you still interested in this issue getting solved? 🤷🏻‍♂️
Please test the latest firmware, prior to responding.

discip, 2023-02-03:
@3eGo
Please try the latest firmware to make sure your issue is solved.
And please do not forget to close this. 😊

3eGo, 2023-02-05:
oh sorry, I hadn't seen the emails arrive, now i check

3eGo, 2023-02-05:
yes it works

closed 2023-02-05 by 3eGo