Ralim/IronOS #1292: Horizontal offset of display TS100

lanefloch, 2022-05-03:
Just got a new TS100 off amazon and flashed it. But it seems the screen has a large offset horizontally. Not sure if there is a setting to compensate for this. Thanks for your help!
[image]
[image]

Ralim, 2022-05-03:
There is not a setting as of present.
That would almost look like the screen is not the correct model (as its expecting a different horizontal offset).
If you flash the miniware firmware does it work?
What DFU version is your TS100?
Is it actually a TS100 or a clone?

discip, 2022-05-03:
Try to use a magnifying glass to inspect, if the oled is physically shifted inside the device.

lanefloch, 2022-05-06:
Ok upon further inspection the entire OLED is physically shifted within the Iron. Thanks for the help!

closed 2022-05-06 by lanefloch