Ralim/IronOS #1294: MHP30 - "heating on power-up" does not work

MEMS-lab, 2022-05-05:
Hi,
I am using the MHP30 hotplate with the latest ironOS in an application, where I want to set the PCB on the hotplate and not have to touch the buttons after setting this up. So my thought was to use the "heat on power up" function, get everything ready and in a last step plug in the power. Then the MHP30 should ramp to the pre-set soldering temperature. However, it does not work. The hotplate powers up with the red LED blinking, but no power is drawn (0.0W) and the temperature does not increase. This is independent from the max power setting and soldering temperature used.

How to re-produce the error (very simple):
1. set solder temperature, then
2. > solder settings > Heat on power up: set to "S" (default is 0) [hotplate should now heat on power-up to soldering temperature]
3. unplug MHP30, then plug in again
4. Nothing happens (yes, red LED blinks, but no heating; I have to exit heating mode in order to enter the menus, so it is in the right state, but something is preventing it from heating)

My device:
MHP30 v2.17.E065BE3, 12-02-22, I use it with the original 65W C-USB power supply

Many thanks for help!
P.s.: Kudos and thanks to everyone here for the outstanding work!

Ralim, 2022-06-07:
Hia,
When you have time; it would be fantastic if you could give the firmware build from here a try: https://github.com/Ralim/IronOS/actions/runs/2454288106

MEMS-lab, 2022-06-07:
Hi there,

I just installed and ran it. Still no heating on start-up. The light is red and animated as before, but no power [0.0 W] is drawn (detailed display).

Sorry for not being able to report success yet! Please give it another try!

Many kind thanks!

Jan

MEMS-lab, 2022-06-07:
Hi,

As add-on: this firmware revision prevents any heating of the MHP30! So you might be on to something. I.e. if I set the startup heating to “0”, so no heating on startup, reset to factory conditions and just hit “A” to heat to the default 320 °C, no heating occurs.

Reinstalling the order firmware made the MHP30 work just fine again.

My guess is that some variable or register is different for the MHP30, which was correct for the normal heating, but now is “wrong” for both the standard heating and startup heating. Looking at the revision before and the difference for the normal operation might have an answer. Just guessing though.

Many thanks for the efforts!

Jan

Ralim, 2022-06-08:
Hi,
Just to check did you use https://github.com/Ralim/IronOS/actions/runs/2454469240; it looks like you replied by email so you may not have seen the edit?

discip, 2022-07-03:
@MEMS-lab
Was your issue solved?
Please close if it was. 😊

MEMS-lab, 2022-07-04:
No, it is not working, unfortunately. I had provided a detailed response, what I think might be the reason. I am happy to try a new version.

discip, 2022-07-04:
Did you try the one provided by @Ralim in the post before mine?

Ralim, 2022-07-06:
@MEMS-lab 
I suspect you tried the older firmware based on what you had said; guessing you did not see the edited link.
Could you please confirm which firmware you used?
Alternatively can you try the 2.19 test release from https://github.com/Ralim/IronOS/actions/runs/2603598835

The root cause for this is that before we start heating on the MHP30 we need to take a series of measurements of the the "hotplate" to measure what one is fitted. And until these complete any output control for the heater is forced off as it interferes with the measurement. There was a race on startup as to which order things would complete in.

discip, 2023-01-05:
@MEMS-lab
Are you done with this issue?
Please do not forget to close. 😊

MEMS-lab, 2023-01-08:
The power-on on start-up is still not working. As suggested before in the comments on what I think is the issue in the code, if might make sense to discuss what was changed and why it is expected to work now, and I can then help to pin-pont why the “fix is not a fix” yet.

Thanks for still trying to find a solution to this for the MHP30!

discip, 2023-01-08:
> The power-on on start-up is still not working. ...

@Ralim
Have you tried if this works on your **`MHP30`** yet?

discip, 2023-01-08:
> The power-on on start-up is still not working. ...

@Ralim
Have you already tried whether this works with your MHP30?

Ralim, 2023-01-23:
Last I tested it works but its been ages since I reflashed it. Will test in next week or less and figure it out.

discip, 2023-02-02:
@Ralim
@MEMS-lab
Has anyone had time to try this out yet?

discip, 2023-02-06:
@Ralim
@MEMS-lab
Would be very nice if someone could test whether this is solved. 😊

discip, 2023-02-16:
@Ralim
As per your statement here: https://github.com/Ralim/IronOS/pull/1306#issue-1263163792
> * [x]  The changes have been tested locally
> * [x]  There are no breaking changes
> 
> * **What kind of change does this PR introduce?**
> * Bug fix for [MHP30 - "heating on power-up" does not work #1294](https://github.com/Ralim/IronOS/issues/1294)

... this should be solved.

So closing this until similar complains occur.

closed 2023-02-16 by discip