Ralim/IronOS #1295: Bricked TS100: "A device which does not exist was specified"

mikkan-39, 2022-05-05:
So I bought my second TS100 and tried flashing the firmware as usual. However, this one got bricked instantly. DFU seems to be still accessible, but nothing else.

**Steps:**
1. Entered DFU mode, a drive with some folders appears
2. Dragged the last TS100 .hex file to the drive
3. The drive immediately dissapears and windows throws and error: "A device which does not exist was specified"
4. After a minute it re-connects as a new drive, which is completely inacessible.
5. Panicked and tried flashing different firmware versions, including manufacturer's, also I have tried using different USB cables, 3 different windows machines and a mac (which throws a very similar error).
6. After rebooting the iron seems to be dead, unless I enter DFU

**Expected behavior**
I expected the firmware to be flashed as usual

**Details of your device:**
 - Device: TS100
 - Release: First I flashed 2.18, but tried several other versions, as well as the stock firmware

**Additional context**
It seems that I can still enter DFU. ver 3.48. I have seen a similar issue on this repo, but their solutions did nothing for me.

Ralim, 2022-05-08:
Hia, this is unfortunately due to bugs in the factory DFU.
The best known work around is to use Ubuntu Linux for flashing.
Sadly fixing DFU bugs in the vendor firmware is entirely out of scope for this repository.
The best we can do is what others have tried and report what worked for them.
I've never had any issues using the double file copy method on Linux.

iasonaszotos95, 2022-05-11:
I have the same issue. It also tells me that the file is too large to transfer. @mikkan-39 let me know if you find a workaround!

mikkan-39, 2022-05-11:
> I have the same issue. It also tells me that the file is too large to transfer. @mikkan-39 let me know if you find a workaround!

Hi! In the end, my workaround was to use Linux :)

iasonaszotos95, 2022-05-11:
It actually worked when I tried it on my Laptop. Just a heads up if anyone else encounters this issue

discip, 2022-05-11:
@mikkan-39 
If this is solved for you, please don’t forget to close the issue. 😊

thanks

closed 2022-05-12 by mikkan-39