Ralim/IronOS #1299: Remains relatively cold when heated

lorond, 2022-05-11:
**Describe the bug**
Does not heat as expected.

Brand new just unpacked miniware TS100 device. On the stock 2.20 firmware it heats well.

**To Reproduce**
1. Plug the power, screen shows value around 225-230
2. Tap A button
3. It heats up in 3-4 seconds and shows value around 320
4. Actual temperature is low (about 55*) and it dosn't solder anything
5. It the boost mode (hold A) screen shows 420, actual temperature rise to around 100*

Temperature calibration does not change anything.

**Details of your device:**
 - Device: TS100 2.51B
 - Release: 2.18
 - Power adapter being used: 24V 3A

**Additional context**

Maybe CHan is an issue? Looks like it reports value with two numbers precision, not one. According to documents, device detects it as 227.9*, not 22.79:

> **CHan**
> This is the handle temperature in °C. This is used for cold junction compensation of the tip temperature. This is shown in degrees Celsius x10, so 200 == 20.0 °C

Debug menu output:
v2.18.70565DE
- Vin 23.6
- ACC LIS2DH12
- PWD DC
- Max 608

when cool (around 22-23*):
- RTip 940-955
- CTip 227
- CHan 2279

when heated up to around 310 (on screen, actual around 50*):
- RTip 2947
- CTip 308
- CHan 2293

lorond, 2022-05-11:
Some photos
PCB:
[image]

Actual temperature:
[image]

Ralim, 2022-05-12:
Hello,

Firstly; thank you for the photos + good details

Indeed CHan looks _very_ wrong and would be what is causing this behaviour.
The question is _why_ is it wrong.
Looking at your PCB, it looks correct; U6 on the far right is the temperature sensor being used here (TMP36GRT - marking T6G)

Biggest suspects to check are:
(1) Soldering of temperature sensor U6
(2) Contact of the connection from the little daughter board to the main PCB
(3) Soldering of the STM32

If they all look fine, you can try reading the voltage on the temperature sensor output (pin 1; bottom left pin in your photo) with respect to gnd (can use power input -ve). To check what its putting out.

lorond, 2022-05-31:
@Ralim Still waiting for multimeter to arrive. Will post information a bit later.

River-b, 2022-07-03:
could try to reheat/reflow each of the items  ralim mentions and then test again. may find out which one had bad solder or cold solder joint.   if you do it one at a time, you will find culprit because the Chan will change.

lorond, 2022-08-26:
@Ralim if I understood your right, you talking about that pin:
[image]
[image]

It's about 2.8218V for cold (temp of environment) state and 2.8689V for hot state (300* on stock firmware).

lorond, 2022-08-26:
Just tested another TS100, that has IronOS working well.
First - it base on STI, mine work on GigaDevice, but model seems to be same - F103 T8U6:
[image]

STI's model voltage on same pin is 0.7548 for cold (~23*) and 0.8080 for hot (320*).
The hot-cold voltage difference seems to be close - 0.0471 on GD and 0.0532 on STI

Ralim, 2022-08-26:
Yes those STM unit measurements are correct.
The 2.8ish Volt measurements are out of specifications for this temperature sensor.

Did you test two different irons, or one iron and swap processor modules?

ZJJoker, 2023-01-23:
Hello.
I have the similar problem with a bit diffirent prerequisites. 
Once I've used an impropiate power supply: while soldering touched with tip another soldering station AC-powered, had a spark. After that had to replace MCU, acellerometer and transistor package. 
MCU marked as STM (not GD, hope not remaked).
Measured Vout from sensor ~0.775 V (pointed directly to the sensor and then to the MCU - the same).
Factory firmware - ok, corresponds to a thermal imager .
Rel 2.16 or 2.20:
RTip ~ 830
Ctip ~115
Chan ~1148

Tip resistance ~ 8.6 Ohm.

Possible to solder by setting about +100 degrees. So, Treal ~ 300, Tset~ 410.

Could you propose something to resolve please?

------------
Best regards,
ZJJoker

Ralim, 2023-01-23:
Your handle temperature is reading as being at 114C.
Assuming that's wrong your damage is the cold junction sense part of the circuit.
On the ts100, this is the TMP36 component.

This also makes sense as the Miniware firmware overwrites this to 25 C if your outside a fairly small window rather than have them accept faulty units.

ZJJoker, 2023-01-23:
Ralim, thanks.
If I got it right, do you mean that replacement of the tmp36 sensor could have a result?

The only thing makes me doubt, is that corresponding to the datsheet the Vout seems to be ok:
[image]
So ~0.775 ~= 20-25 С degrees. And the Vout still changes while heating or cooling the sensor.
The 1 kOhm resistor between  sensor and MCU is also ok.

Ralim, 2023-01-23:
@ZJJoker 

So if you are reading the correct output voltage from the TMP36; and your 3.3V rail to the MCU is actually 3.3V (+-1%) then I would be suspect of connections between the TMP36 and the MCU or damage to the MCU ADC (relatively unlikely).

> Just have mesured the Vs = 3.8, not 3.0 as in datasheet.
Vs should be 3.3V iirc?

ZJJoker, 2023-01-23:
[image]
According to the scheeme Vs=3.9, thats ok.
By the way,  the Vout is also ok.
Does the firmware show the handle temperature when switched (without heating)?
Just have tried freezing the handle without the tip - Vout decreased to ~0.6 V. After installing the tip - LCD shows 114.
And by the way, is there any chance to get RAW values of ADCs with stlink?

Ralim, 2023-01-24:
Ah yeah ok, its on the 3.9V rail.

The decoded raw handle temperature is in the debug menu. See https://ralim.github.io/IronOS/DebugMenu/ 
`Han C`

ZJJoker, 2023-01-25:
No, I don't mean DECODED tempeature, but RAW ADC value - to check if there's any toruble with hardware.
I can't still get:
Vout=0.775 ~ T=25 by datsheet
Real:
Vout = 0.775 -> ?? F1(V) =ADCval -> F2(ADC) = 114 С (or 1148 СHan).

If F2 works fine with others, the only weakness in my case I see in F1, that converts Vout to RAWadc.

Ralim, 2023-01-25:
To get a raw value you just need to build a tweaked version of the firmware.

The formula used on a TS100 is straight forward and linear:
https://github.com/Ralim/IronOS/blob/07d85c82112a33d6c3b2541674667ebffe5060fb/source/Core/BSP/Miniware/BSP.cpp#L79

So if your getting 1148 we can flip that fairly trivially and say
`(1148×994)/100 + 4965`
Which suggets adc reading is `16 376`

We can then scale that into a voltage by doing
`(16 376/(4096×8)) × 3.3`
Which gives the reading of 
`1.649 V`

So either:
(1) ADC pin is reading 1.649 V
or
(2) ADC Vref is not 3.3V

River-b, 2023-01-27:
@ZJJoker 
If you need the TMP36,  they sell several versions of it  on  LCSC.com
this might be it:   https://www.lcsc.com/product-detail/Temperature-Sensors_Analog-Devices-TMP36GRTZ-REEL7_C129489.html

Search  for TMP36  on the website and you can pick the form that matches.  I don't have a TS100 but you can tell by the photo  if  it looks like the one you have or not. 
I have ordered from  LCSC.com before  and even though it ships from China, it's fast and not expensive for shipping.  They carry  same or more parts as  Digikey/mouser.
LCSC.com and JLCPCB.com (to order custom PCB)  are the same company.

ZJJoker, 2023-01-29:
@Ralim 
Thanks for very detailed explanation. I have studied the source and now I do understand the algorythm. 
But... First, I've tried to resolder the MCU to be sure soldering is not the issue. Better I had not)) After my experiments, the MCU is detected by PC, is able for flashing and saving config, but OLED is dead. Maybe I've spoilt it while unglueing for measuring voltage on the resistor which was under the LCD.
Finaly, I've measured even the CLK and DATA  signal with an oscilocope) - If my grand dad ever heard that in the last century (I'm using osiloscope to repair a soldering iron)- he'd say I've got crazy.
So, now I've ordered some spareparts to revive the hardware and take a pause.
Thanks once more.

ZJJoker, 2023-01-29:
@River-b 
Thanks for the link, the store of of great choice of components!
Though I'm still sure tmp36 is not to blame, as it seems to works according to the datasheet. Maybe I really had some trouble with the MCU, as for now I've remembered, once the iron heated to the red, as if it had no temperature control at all, but I've never had SenErr oк smth like that. After that (I don't remeber what exactly I've done then), It was adequately responding for the temperature settings, but for -90 degrees to the value set.
My investigation to be continued, as for now thats a matter of a principle.

discip, 2023-01-29:
@chino
> what's that awesome temperature device you're using there!?

Definitely from UNI-T!
Maybe this one: [UTi260B](https://de.aliexpress.com/item/1005001486771461.html?spm=a2g0o.detail.1000014.29.3103139cD8g4ps&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.320564.0&scm_id=1007.40000.320564.0&scm-url=1007.40000.320564.0&pvid=851c3ba9-9d7f-40a3-b69c-891fe79c8066&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.320564.0,pvid:851c3ba9-9d7f-40a3-b69c-891fe79c8066,tpp_buckets:668%232846%238110%23356&pdp_ext_f=%7B%22sku_id%22%3A%2212000031135989948%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=2%40dis%21EUR%21521.52%21291.96%21%21%21%21%21%400b0a23b116750312115971576e38c5%2112000031135989948%21rec)

River-b, 2023-01-30:
> OLED is dead.

LCSC.com  also sells this same OLED  screen for about  $2  I think.  I ordered some before for Pinecil  so if  the ts100  has the same screen, you can replace that too.  however  maybe first check  all the solder  joints on the  OLED  screen.  sometimes the soldering  on it is  very bad.
my understanding is the OLED screen on all of these kinds of Irons is the one part that has to be   hand  soldered by a human. 
This leads to poor  soldering  joints sometimes on them  or they are marginal, it may work  but is not reliable.   on one of my irons  I  re-heated it  and fixed the  fingers of the OLED screen.

discip, 2023-02-16:
@lorond
Is your issue solved yet?
If so please consider closing. 😊

thanks

lorond, 2023-02-16:
@discip sorry for so long time to answer. Hope, I will continue research soon.