Ralim/IronOS #1300: Pinecil stuck on 9v QC3

ZeyHex, 2022-05-15:
**A problem**
When using Pinecil64 paired with Xiaomi MDY-11-EZ, the soldering iron gets stuck in 9v mode, although the adapter can output up to 20v using the QC3 protocol.

**Describe the solution**
The problem is probably related to incorrect initialization of QC in the project code, I did not read it, however, in my personal projects, I already used the https://github.com/GyverLibs/QuickCharge library to work with QC3 and I was able to get any voltages that were required . Maybe implement it in IronOS as well?

Ralim, 2022-05-15:
So one of the big problems with QuickCharge is the timings are not consistent across brands.

Generally the 20V via QC3 support is tested against the Pine power supplies and the timings are adjusted to suit them. QC is not recommended for use at 20V as generally supplies are not rated for the current and more than one has been damaged by this in the past.

Do you know what the timings are for your your supply/can you capture the signal levels on the USB data lines when it's negotiating ?

ZeyHex, 2022-05-15:
The timings don't match, it's true, but there is an overlapping range for all of them that works. It is this range and durations that are used in the library that I recommended.

Yes, of course, the device is certified by Qualcomm.

Ralim, 2022-05-15:
I'll have a look at that library and compare; it is notable that this adaptor can't be used at 20V; you would need to use a custom QC3 voltage for safe operation.

I suspect your adaptor may not be liking the weak pulldown that this firmware enables for compatibility with other chargers.

ezshua, 2022-08-09:
Hello,

I have the same power supply and the same problem with voltage stuck at 9v. But I solved it by replacing the power cable with an older one. 

The first cable had red USB3 standard connectors (with additional contacts in the Type-A connector) and the voltage did not rise above 9V, and the second cable with white USB2 standard connectors (only 4 main contacts) and the voltage rose to 20V. After that, the soldering iron showed a power take-off of 33W during heating, which is the maximum for this power supply!

Probably, when using an old cable, another mode of selecting the supply voltage is activated. Is it possible to make a software solution in the form of a custom setting to bypass this situation?

BF3000, 2022-08-11:
Same problem. It appears QC voltage was limited to 9v in options. I set it to 20v and its worked.

ezshua, 2022-08-15:
If you pull the wire out of the charger a little (about 2-3mm), then the Pinecil calmly scrolls the voltage to the limit set in the settings, I have up to 20V. I think the PD is turned off and we just switch to the QC in a hardware way.

ezshua, 2022-08-26:
Get access to PD debug menu:

PD Debug
State 12 No VBus
1  5V  3. 0A
2  9V  3. 0A
3 15V  1.80A
4 20V  1.35A

and now Pinecil stuck on 8.9V.
I set QC limit voltage to 20.4V, but without changes.

I think the initialization of the PD does not go to the end and does not allow the initialization of the QC to start, and therefore it gets stuck at 9V. It's not enough for my knowledge to point out the reason more precisely ...

Ralim, 2022-08-26:
Hi, your using USB PD not QC, so QC voltage wont change  anything here.

The iron is limiting to 9V to protect your power supply.
As it cant do a high enough current on the higher voltages to be used at those voltages.

discip, 2023-02-16:
@ZeyHex
Is your issue still present?
Otherwise please close. 😊

thanks

ezshua, 2023-03-07:
v2.20 working fine with MDY-11-EZ and MDY-12-EH at 19.83V ! 
thx!

discip, 2023-03-10:
Closing, as OP does not answer and the issue seems to be solved.

closed 2023-03-10 by discip