Ralim/IronOS #1308: Boot logo swap still shows Miniware logo for about 1/3 of a second on TS80P

skeeballassault, 2022-06-07:
**Describe the bug**
Just got my new TS80P, and when adding a custom boot logo, the Miniware logo still shows for about a third of a second before cutting to my custom logo. Videos I've seen have shown the Miniware logo not showing up at all so that appears to be intended behavior - this isn't a world ending bug but it will certainly bother me.

**To Reproduce**

**Expected behavior**

**Details of your device:**
 
 - Device: TS80P
 - Release: 2.18
 - Power adapter being used: Included power adapter for TS80P
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

Ralim, 2022-06-07:
On the TS80P on later DFU versions, miniware added this to the bootup before ironos starts. So IronOS has no control over this.

The only way to remove it is to entirely replace the DFU bootloader; noting that this comes with some risk. So it is best to have a SWD programmer on hand just in case.

discip, 2022-06-08:
@skeeballassault
If you are brave enough to try replacing the bootloader,
here is the place you should go: [IronOS-dfu](https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/Bootloader.md)

closed 2022-07-03 by Ralim