Ralim/IronOS #1310: TS100 DFU 3.48 black screen after flash

ozjuly19, 2022-06-09:
**Describe the bug**

Completely blank screen when trying to use the system after trying to flash IronOS (anything but DFU mode) powered via wall or usb flashing stock is a no go tried every version I could get my hands on also tried older versions of IronOS modified versions and searched through every issue I could get to regarding similar issues in github still no fix

I have tried different computers (2x Windows 11 1x Windows 10 1x Ubuntu 20.04) and multiple cables with no differing effect everytime you try to copy a new FW over (.hex) it always returns .ERR or in one case with stock it returns .RDY but the issues still persist as if nothing is different

I have also fully disassembled the iron in hopes of finding loose connections on the screen or pull resistors and did not find any issues if this isn't a realistic fix at this point I'm all for replacing this device if so advised

**To Reproduce**

Flashed fw and a working iron

**Details of your device:**
 
 -  TS100 GD32F103 (STM32 not sure how this relates I see it when it plugs into ubuntu)
 -  DFU v3.48

ozjuly19, 2022-06-09:
[image]
[image]

ozjuly19, 2022-06-09:
After fiddling with the device for what was intended to be the last time I am now able to flash Mini 2.20 but still not TS100_EN.hex (or IronOS) so whats currently working is:
 - Stock v2.20

Non functional
 - Custom (IronOS) version: any

closed 2022-06-09 by ozjuly19

Ralim, 2022-06-09:
Ah sorry to hear about this; the GD32 seems to be a little bit more flaky than the ol' STM32.
I found I had to do the double copy to get things to work with it, but after that they should _stay_ working.

Sorry to hear about this though 😢