Ralim/IronOS #1311: Ts100 massive problems

NurmiMiika, 2022-06-10:
1st problem: When i try to flash the firmware to my brand new ts100 it always says .ERR. 

2nd problem: When i plug in power normally, i only see black screen.

**What ive done**
Press A button - plug in microusb - copy the file to the showing ts100 ”folder” on pc - doesnt disconnect and connect again - shows me .ERR

**Expected behavior**
1st problem: When i copy the file it should disconnect and connect back and say .RDY

2nd problem: When plugging in the battery 4s, it should power up and show either normal menu or ralim menu.

**Details of your device:**
 
 - Device: ts100
 - Power adapter being used: the one that comes with it and a 4s lipo

**Additional context**
I have tried multiple pc:s, cables and with different downloads of the firmware but it doesnt show me .RDY.

regards. Nurmi Miika

NurmiMiika, 2022-06-10:
Dfu: 3.48

discip, 2022-06-11:
@NurmiMiika
Maybe this could help you: https://github.com/Ralim/IronOS/issues/1310

And please don't forget to close this issue if solved. 😊
kind regards

NurmiMiika, 2022-06-12:
Seems to not work still.

NurmiMiika, 2022-06-12:
Tried it just like so and didnt work either....

Ralim, 2022-06-13:
@NurmiMiika 
Sadly this is the nature of the vendor (Miniware) DFU code.
We cant really fix this; only speculate on what has worked for other people.
Make sure you fully extract the zip, and that you are trying to flash the .hex file.
Otherwise its worth reading around the previous issues and trying things people suggest.

NurmiMiika, 2022-06-13:
I understand... I've read multiple issues and have tried to get useful information but nothing. I contacted the place i bought it from and now ill just have to wait and test when i can...

River-b, 2022-07-04:
you tried different usb cables and also tried to double and triple copy over the file ?  even if you see error keep trying the copy paste.  also try using a different   PC/laptop for this & different cable.  we have seen people who switched to different USB port or different computer or different cable and it works.

NurmiMiika, 2022-07-04:
Yes i have tried... Many pc's and cables.

NurmiMiika, 2022-07-04:
Also many copies

discip, 2022-07-04:
Have you tried flashing the stock firmware?

NurmiMiika, 2022-07-04:
Yep that also... Doesnt work and contacted miniware and they are sending a new one now...

NurmiMiika, 2022-07-04:
Gonna try to get this one to work also

discip, 2022-07-04:
Does it boot up in DFU mode?

River-b, 2022-07-04:
perhaps there is issue with your ts100, are you in Return period for it?  if you can still return it, might I recommend Pine64 Pinecil? as that device, was designed to use IronOS and Ralim has more power to dig into issues.
if you can't return it, then will need to look into this more.
have you looked at it and any cold joints or solder spots that might need  reflow/reheat?

NurmiMiika, 2022-07-04:
When plugged to bc and pressing a button... Dfu shows up on screen

NurmiMiika, 2022-07-04:
Ill have to look at the joints... Luckily had warranty still so getting a new one rn

NurmiMiika, 2022-07-04:
@discip ill have to try downloading from that way also...

discip, 2022-07-04:
Yes but maybe the bootloader is broken somehow?

### BUT before going this path, please read the documentation carefully, because you might have to use external hardware provided by stm or 3rd party, to revive your iron if anything goes sideways!

NurmiMiika, 2022-07-04:
Ill try with that. Seems like there is no instructions on the site for the bootloader / runtime

River-b, 2022-07-04:
when you get the replacement Iron, use that and just reheat/reflow a bunch of joints. I did that to one old iron and it started working.   I mostly reflow but did add some flux/new solder to some spots.

NurmiMiika, 2022-07-04:
@River-b ill do that when i get home with my other iron

discip, 2022-07-04:
Here you go:
https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/Bootloader.md
https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/BackUp.md

River-b, 2022-07-04:
the legs of solder on oled screen are done poorly it seems from what I've seen in photos and youtube videos.

NurmiMiika, 2022-07-04:
@discip i dont know if im dumb or, but i cant find stuff like dfu util or runtime.hex

NurmiMiika, 2022-07-04:
I would for real need somebody to guide every move... Regards complete noob

discip, 2022-07-04:
Here it what you need: https://github.com/Ralim/IronOS-dfu/releases/download/v0.2/TS100.zip

But I just realized, that you would have to flash **`runtime.hex`** as you would do with a regular firmware.hex.
Nonetheless you could try it. 😁

NurmiMiika, 2022-07-04:
I tried and it only says .err 🥸

discip, 2022-07-04:
Have you tried using a Linux machine yet?

NurmiMiika, 2022-07-04:
Dont have one unfortunately

discip, 2022-07-04:
What about: https://www.virtualbox.org/?

NurmiMiika, 2022-07-04:
I unfortunately cant use these...

discip, 2022-07-04:
Alright than you will have to either ask one of your parents, siblings or friends, or just wait for the other iron to arrive and hope that one is functional. 😅

NurmiMiika, 2022-07-04:
Ill ask my godfather cause hes the only one that knows 😃 Hopefully the one that will grt to me works... Idk if i will even try to download ironos to it just bc i want even one to work

discip, 2022-07-04:
### Please do not abuse the name of god!

NurmiMiika, 2022-07-04:
I translated the right term from my language and that was it🧐

discip, 2022-07-04:
Nevertheless you can not do that.

NurmiMiika, 2022-07-04:
Sorry... My bad😶

discip, 2022-07-04:
I wish you get a good soldering iron.

NurmiMiika, 2022-07-04:
Hope so :) I wont close this problem cause ill propably need help with that later on when i try to get it workin

discip, 2022-07-04:
Do as you are pleased, but don't forget to close it as soon as your problem gets solved! 😊
thanks in advance

River-b, 2022-07-04:
just saw this - go to very bottom of this page to the **FAQ**  section for miniware - it mentions some steps when you get the ERR   error. 
https://github.com/Ralim/IronOS/blob/dev/Documentation/Flashing.md

River-b, 2022-07-04:
Depending on which OS you are using, there are detailed instructions in the Flashing.md link I gave above  that tell you exactly how to use dfu  to flash  depending on which OS  you have on your PC/laptop  windows/mac/linux.  and also if you have a Miniware product like ts100  or a Pinecil.   

Pinecil iron works best with IronOS firmware because it was designed to use it.  Unfortunately, TS100 comes with miniware's own firmware so sometimes there are issues getting TS100 to cooperate with open source firmware like IronOS.

jnm, 2022-07-07:
> ### Please do not abuse the name of God!

I know this is quite off-topic, but "godfather" is not an abusive term:

> In [infant baptism](https://en.wikipedia.org/wiki/Infant_baptism "Infant baptism") and denominations of [Christianity](https://en.wikipedia.org/wiki/Christianity "Christianity"), a **godparent** (also known as a **sponsor**, or *[gossiprede](https://en.wiktionary.org/wiki/gossiprede "wikt:gossiprede")*)[[1]](https://en.wikipedia.org/wiki/Godparent#cite_note-Roth2005-1) is someone who bears witness to a child's [christening](https://en.wikipedia.org/wiki/Baptism "Baptism") and later is willing to help in their [catechesis](https://en.wikipedia.org/wiki/Catechesis "Catechesis"), as well as their lifelong [spiritual formation](https://en.wikipedia.org/wiki/Spiritual_formation "Spiritual formation")[[2]](https://en.wikipedia.org/wiki/Godparent#cite_note-Fitzgerald1994-2) (https://en.wikipedia.org/wiki/Godparent).

Cheers :peace_symbol:

River-b, 2022-07-07:
> ### Please do not abuse the name of God!

@discip ,   godfather is a correct term and is use often and commonly in USA. 

> Nevertheless you can not do that.

There is nothing wrong with  NurmiMiika saying godfather.  it's like saying uncle or aunt.  if you have someone your parents selected as a guardian for you during your baptism, they are officially called a godfather or a godmother. These terms are used commonly in USA and are a normal and respectful ways to refer to the special person your parents picked to witness  a baptism of the baby and to act as a guardian for the child and to help the parents with the child. 
Godfather and Godmother are incredibly respectful positions and people are usually honored to be chosen for the role as a guide and mentor to the child.

River-b, 2022-07-07:
> 

there was nothing wrong with what you said.  I just missed it because I was focused on helping you with soldering iron, but godfather is a respectful and correct word. that word is very respectful way to call the person in your life that your parents choose as your godfather.  it is a little bizarre someone is trying to tell you not to use the word. there is absolutely nothing wrong with it.

NurmiMiika, 2022-07-07:
Yep exactly... But ye i dont rly care and its fine ;D

discip, 2022-07-07:
@NurmiMiika 
## Sorry my bad!
I was not aware of the fact there is another meaning of what I thought it was.

discip, 2023-02-16:
@NurmiMiika
Have you managed to solve this issue?
If so please close. 😊

thanks

closed 2023-06-04 by Ralim