Ralim/IronOS #1313: Feature request for modification of ts 100

showlyshah, 2022-06-14:
Hello, 
i have a ts 100 , in which i have installed ralim firmware before. later some day, its oled got dim and now it is totally dead. i searched for .69 inch oled in many places.. only chinese sellers  have them. and i cant buy from them since shipping from china is blocked in my country ( india ) .. those available in my country are so much expensive ( 13 usd if converted ) .. while bigger display with higher resolution are available at much cheaper price ( 1.2-1.3 usd ) in my country.  Those displays use the same ssd 1306 control ic . 
i want to mod my iron with a .91 or .96 inch display ( 128*32 ) . can you please point out what modifications ( on source files ) i should do in order to use a bigger display with higher resolution (  i can do the hardware modifications myself ) so that the images and fonts will be displayed properly in  a 128*32 size display.

Ralim, 2022-06-14:
So the only supported way to do this would be to use a subset of that display like the original.
In which case you would just need to change the init sequence in the OLED.cpp file

showlyshah, 2022-06-15:
sorry but i dont have much programing experience. can you help me in making  necessary  programming adjustments for the below linked oled screen .. ( i can do hardware mods for it ) 
https://www.ktron.in/product/0-91-inch-white-color-oled-display-i2c/