Ralim/IronOS #1317: TS100: graphics is slightly overlaped

sorgelig, 2022-06-21:
**Describe the bug**
[image]

just updated to v2.18 and noticed graphics overlap at the right edge on symbols C and V.

I have also Pinecil soldering iron. It doesn't have such problem.

federicodilo, 2022-06-24:
Just to confirm that's a model related issue, mine is fine.
[image]
 [image]

sorgelig, 2022-06-24:
my TS100 is at least 2 years old. How to provide more info about specific model?
I don't know how OLED is addressed, but can it be some race condition of graphic buffer filling and pulling it out over I2C/SPI to OLED?

Ralim, 2022-06-24:
This is super odd.

Could I ask you to try going back to 2.17 and testing and then coming back forwards to 2.18 and testing? Just want to (1) rule out bad flash and (2) see if it's new.

I'm really not sure what going on here; we pre render the whole screen and then draw it in one transaction. So I would expect it to occur on other devices too.

I'll have to dig out my ts100 and see if I can make it happen for me.

sorgelig, 2022-06-25:
i've tried already v2.17 - same issue. Went back to v2.18. Tried different languages (Russian, English). It only happens in Detailed view mode. Non-detailed view don't have such issue.

sorgelig, 2022-06-25:
after more precise inspection it seems screen is broken. In some cases it displays that column, but in other cases doesn't.

closed 2022-06-25 by sorgelig

Ralim, 2022-06-26:
Ah I'm sorry to hear it's the screen.

The screens are fairly cheap and not too hard to replace if you decide to try.

sorgelig, 2022-06-26:
it's very minor break. So i will use it as is till it completely break. Then i will use my spare TS100, haha :)
Also want to switch to Pinecil, and it's good i can re-use whole set of TS100 tips with it.