Ralim/IronOS #1321: Pinecil v1.1 includes vbus mod

GabyPCgeeK, 2022-06-27:
I've looked around and haven't found any reference to this so here it is. 

I bought a Pinecil and IronOS reported "PD No VBus" and upon opening it I found out that it has a v1.1 board. The FBUS302 vbus pin is routed to an unpopulated resistor.

[image]

[image]

River-b, 2022-06-30:
hello, could you please post photo of the Back side of this new PCB.  I want to see the production number.  it would be on the side where there is an empty spot to add hall sensor near the U14  spot on the PCB.

River-b, 2022-06-30:
indeed it looks like the PCB has been updated

GabyPCgeeK, 2022-06-30:
[image]

[image]

[image]

[image]

[image]

River-b, 2022-07-02:
2213 ,  this has production date of 13th week of 2022  , so April 2022. 
Indeed it looks like the PCB changed. 
Just to confirm,  did you buy this directly from  Pine64.com  or the official Pine64 store on Amazon USA?

would have to see if we could get confirmation from Pine64 of this PCB change so we won't have to cut the trace anymore to get the DC 24V  modification safely.   For now It is good idea to check with the IronOS firmware first like you did and if it says "PD No Vbus"  - then do not cut the trace. 

Also could you  show photo of the outer Case?  does the case still say 12V-21V  or has it changes to 12V-24V ?

GabyPCgeeK, 2022-07-03:
Bought it from Amazon on Jun 21.

[20220702_230712.jpg](https://user-images.githubusercontent.com/28073116/177024289-db0876b7-6fb4-4922-88d9-e4b09f67b4e9.jpg)

River-b, 2022-07-03:
nice high res photos. thanks

discip, 2022-07-03:
@GabyPCgeeK
Since this is no issue you maybe want to close this and move it to the discussion. 😊

thanks

closed 2022-07-20 by GabyPCgeeK