Ralim/IronOS #1327: Can't upload logo to sequre sq-001

bananafan2015, 2022-07-08:
Hi, I recently bought a sequre sq-001 and flashed it with the latest release of iron os(2.18). After that, I wanted to add a custom logo so using the python script I made one but no matter what I tried when I pasted the file it reboots and renames the file to .NO

**To Reproduce**
 Steps to reproduce the behavior:
1. Plug in soldering iron in dfu mode 
2. Paste converted png.hex file in poped up drive
3. Drive reboots with a file with .NO extension

**Expected behavior**
After pasting the file in the d drive called firmware drive should reboot and the pasted file should be named .OK

**Details of your device:**
 
 - Device: sequre sq-001
 - Release: 2.18
 - Power adapter being used: powering of USB
 

**Additional context**

bananafan2015, 2022-08-04:
Hi I understand that you may be busy but I've been waiting for an answer to my latest comment for 21 days. Could you take a look and help me with my problem.

Ralim, 2022-08-04:
The SQ-001 is not supported. And their bootloader has different quirks to the Miniware one.
I dont have one and I have no idea why their DFU firmware would reject this.

discip, 2022-08-04:
@bananafan2015
Maybe you could try replacing the bootloader with the one created for IronOS?
https://github.com/Ralim/IronOS-Meta

But first read the instructions attentively.

discip, 2022-08-04:
@Ralim
Would your bootloader fix that, or is this device not supported?

discip, 2022-08-04:
Would your bootloader fix that, or is this device not supported?

Ralim, 2022-08-04:
The device is not officially supported; it _should_ work but I have zero idea around the recovery if it goes wrong etc.

bananafan2015, 2022-08-06:
Ok thanks for the help I guess I'll just use it without a custom logo.Also what bootloader is discip talking about?

discip, 2022-08-06:
@bananafan2015
Here you go:
https://github.com/Ralim/IronOS-dfu

discip, 2023-01-05:
@bananafan2015
Did you get this going?
Please do not forget to close if this was solved. 😊

bananafan2015, 2023-01-06:
Unfortunately not,i dont want to risk bricking my iron so I'm closing the problem.

closed 2023-01-06 by bananafan2015