Ralim/IronOS #1328: MHP30: No Accelerometer detected

markbirss, 2022-07-10:
At startup message displayed that - No Accelerometer detected

**To Reproduce**

**Expected behavior**
Detect Accelerometer

**Details of your device:**
 
 - Device:MHP30 
 - Release: v2.18.2.70565de
 - Power adapter being used: 65W USB-C 20V 3.25Amp
 - If this is an accelerometer related issue, please include its model number here:

Debug menu shows: ACC None

**Additional context**
First time user with new device

Bought from BangGood June 2022

Ralim, 2022-07-10:
Hello,
To confirm was the accelerometer functional in the stock os?

If it was then most likely they have changed what model accelerometer is fitted.

Two options:
(1) open device and look at what they have fitted
Or
(2) make firmware to scan the i2c bus to find it.

If you have time and are willing to go looking that would be great but otherwise I can try and make you a firmware to scan.

Jarves020, 2022-07-17:
Hi,

i have the same Problem.
Tilt Sensor not working under IronOS 2.19.
Debug Menu shows "ACC None".
Flashed MiniWare 2.06 Firmware and Tiltsensor works.

Disassembled the unit and made Pictures of the PCB with readable Part Markings.
PCB Version is V1_6_A.
I guess that the accelerometer is the Part in the middle of the PCB back side (second Picture) marked with "AMVM 693"?

[image]
[image]

markbirss, 2022-07-17:
@Jarves020 thank you for opening it up and taking the pictures.

Adafruit has a handy list and pdf with some common I2C stuff but it not listed there either
only other 3-axis accelerometer's

https://learn.adafruit.com/i2c-addresses/the-list
https://cdn-learn.adafruit.com/downloads/pdf/i2c-addresses.pdf

wildblue, 2022-08-20:
I have the same problem.

Ralim, 2022-08-20:
Can you please test the firmware from this action:
https://github.com/Ralim/IronOS/actions/runs/2893341755

If you get no errors, can you report which model it recognises?
If you get no errors, can you report if it works?

Jarves020, 2022-08-20:
Hi,
tested the firmware.
It works. The move counter in the debug Menu counts up on Movement.
Heating stops if you tilt it in any direction and the status LED switches to green, but no Alert is Displayed.
The Display also goes to sleep after a while in Standby and wakes up on Movement.

Acc reported is SC7A20.

Thanks!

[image]
[image]

Ralim, 2022-08-22:
Thanks for testing.

I've since updated that patch to just add the SCA, could you double-check for me that this firmware also works: https://github.com/Ralim/IronOS/actions/runs/2901939828

And if so I'll merge it in and call it done :)

Jarves020, 2022-08-22:
HI,

i can confirm, that https://github.com/Ralim/IronOS/actions/runs/2901939828 works as well.
Acc is detected as SC7A20 and heating stops if the Plate is tilted!

discip, 2022-08-22:
@markbirss
Have you had any chance to test the firmware provided by @Ralim lately?
If that works for you as well, please do not forget to close this issue. 😊

thanks in advance

closed 2022-08-22 by Ralim

markbirss, 2022-08-23:
@discip 
Just tested all good, thank you all