Ralim/IronOS #1330: Menu animation prevent moving quickly

bombela, 2022-07-17:
**Describe the bug**

When navigating the menu items, there is some animation expressing a move between the items. The animation cannot be skipped. And is effectively limiting to at most one button press per second.

**To Reproduce**

Open the settings menu. Try to navigate faster than one action per second.

**Expected behavior**

Move to the next menu item as fast as I press the buttons.
Alternatively, an option to turn off menu animation similar to turning off icon animations.

**Details of your device:**
Pinecil, updated to IronOS v2.19.

**Additional context**

The original firmware on my Pinecil did not have animation when moving between menu items. This made navigating as fast as you can press the buttons. But somehow the software would lockup randomly. Once I didn't noticed the iron stayed on (and hot) for a full day. Hence why I upgraded the firmware.

Fun remark: This soldering iron has a computer more powerful than both my CNC router and my 3D printer. That is both amazing and horrifying at the same time.

Final note: the soldering iron is the best I have ever used. It heats up fast, and maintains temperature incredibly well (when it doesn't locks up that said).

River-b, 2022-08-03:
have you tried to do a reset in menu? and this is noticeably slower to navigate buttons than your last Firmware which was which version?

River-b, 2022-08-03:
Try this Beta  Firmware Ralim just fixed.  Let us know if it fixed it.
https://github.com/Ralim/IronOS/actions/runs/2788482416

Download Zip at bottom of that link, unzip it,  then  use Pine64 Updater , custom > browse to file  and then Install it.  or on Linux just use DFU-util.

bombela, 2022-08-03:
I am away from my pinecil for a little while for now. I will reply to what I can right now.

I haven't tried to reset the settings. I don't remember and did not write down the version of the original firmware. It came with the pinecil that I bought on Amazon US. The UI did not have animated icons in this version.

I have had my pinecil for a few months. When it looked up, the display would freeze. But it would still heat up at whatever temperature was set. The only way to turn it off (and the heat off) was to unplug the power cable.

I am using a 19V laptop PSU. A Dell one, top of the line in 2011. 130W I think. More than the pinecil needs for sure. I used a barrel jack initially. Then a silicone cable with a usb-c connector because it is just much more flexible. As you guessed, you don't want to plug this cable into any random usb-c device. I did wonder if the barrel jack was somehow slipping in and out momentarily and maybe triggering the lock up. But it still happened with the more secure usb-c connector. I haven't tried another power supply.

I didn't not have any lockup with the new firmware and the same power supply so far.

River-b, 2022-08-04:
> not have any lockup with the new firmware and the same power supply so far.

Did you try the Beta Firmware,  did it Fix the Issue? do you have Fast menu now?

PR #1347 (2022-08-12): Animation non block

Buttons can now skip an animation in progress
Animations are faster
* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**
Should close #1330
closed 2022-08-12 by Ralim