Ralim/IronOS #1334: Allow startup heating during boot logo

aaronjamt, 2022-07-29:
**Is your feature request related to a problem? Please describe.**
I've noticed that when a startup image or animation is enabled, the `start-up behavior` setting doesn't apply until _after_ the image/animation has finished and returned to the main menu.

**Describe the solution you'd like**
Ideally, there would be a setting that allows you to make the `start-up behavior` wait until the boot image is done _or_ start immediately

**Describe alternatives you've considered**
There could also be additional options under the `start-up behavior` menu that let you choose "heat up immediately at power up" or "enter sleep mode immediately at power up"

**Additional context**
This wait applies even when the `boot logo` setting is set to infinity, meaning anyone with that setting could easily plug in/turn on their iron then wait for it to heat up, forgetting that they have to hit a button first (not that I know anyone who would do such a thing... :eyes: )

aaronjamt, 2022-07-29:
Looks like this is partially fixed by moving lines 979-983 of `GUIThread.cpp` before line 975 (`BootLogo::handleShowingLogo((uint8_t *)FLASH_LOGOADDR);`) but that causes the boot logo to show up immediately and only once you exit soldering mode does the boot logo show.

Ralim, 2022-07-29:
So at the moment this is by design, the iron never heats unless the screen is explicitly showing this.
As otherwise people would set an infinite boot up logo and not realise the tip is burning hot when they press the button to dismiss the bootup logo.

aaronjamt, 2022-07-29:
> So at the moment this is by design, the iron never heats unless the screen is explicitly showing this. As otherwise people would set an infinite boot up logo and not realise the tip is burning hot when they press the button to dismiss the bootup logo.

That makes sense... what if when the bootup logo is set to infinite it waits, otherwise starts immediately? Could also have it only heat to sleep mode temperature before the bootup logo finishes, then switch to whatever you configured for autostart (obviously only if you enabled autostart and didn't set it to the `R` setting?

aaronjamt, 2022-07-29:
@Ralim How does something like [this](https://github.com/aaronjamt/IronOS/commit/7a34e857ce3b388ec3614c6c1c82e1399b6f0736) work? If all of the following conditions are met, it will start heating during the boot logo to your sleep temperature or 50°C (whichever is lower, since the boot animation shouldn't be long enough for the iron to get above that temperature unless you have a very fast iron, in which case this doesn't even matter anyways):

* Boot logo is enabled
* Boot logo time is NOT infinite
* Autostart is enabled
* Autostart is NOT set to `R` mode (AKA sleep mode without heat)

aaronjamt, 2022-07-29:
I just tested those 4 conditions (well, except for the "boot logo enabled" one which isn't really possible to test because the original code will start the iron heating, whether mine does or not) on my Pinecil and it doesn't start heating until after the boot logo finishes unless they are all satisfied. I'm going to open a PR for both this issue and #1335 (would you rather I make one PR for both or would you rather they be separate? I can do either way)

commit c28a8620 (2022-07-31): Merge pull request #1338 from aaronjamt/boot-anim-preheating

Allow preheating iron during boot logo (fixes #1334)
closed 2022-07-29 by aaronjamt