Ralim/IronOS #1335: Slow boot animation on Pinecil

aaronjamt, 2022-07-29:
I was trying out the animated boot images/GIFs on my Pinecil and found that both the `IronOS.gif.dfu` and `IronOS_L.gif.dfu` play back slowly (I looked at the `.gif` files in this repo and they are each 2.73 seconds long, but the iron takes 6-7 seconds to show the main menu, which is quite annoying when I want to just plug in my iron and start soldering). Is this something anyone else has noticed and can it be fixed? For now I'll just use the static `IronOS.png.dfu` but I would love to see the animated one working.

aaronjamt, 2022-07-29:
I believe I found the issue: in `BootLogo.cpp`, line 55, it says `osDelay(interFrameDelay * 5);`. Removing the `* 5` makes the animations play at the same speed as they do in the original `.gif` files. I'll submit a PR in a couple minutes (trying to fix the other issue I opened as well).

aaronjamt, 2022-07-29:
Reopening as I was completely wrong in my diagnosis. Turns out the animation file I was using is out of date. The `IronOS-Meta` repo needs a new Github Release with the latest versions of those files.

Ralim, 2022-07-30:
Clicked publish on the release. Sorry for that.

discip, 2022-07-31:
@aaronjamt 
This can be closed now. 😊

aaronjamt, 2022-07-31:
> @aaronjamt 
> This can be closed now. 😊

Oops, didn't see Ralim's comment, closing.

closed 2022-07-31 by aaronjamt