Ralim/IronOS #1342: RFE: Add soldering instructions for noobs and reference

FreifunkerEZ, 2022-07-31:
**Is your feature request related to a problem? Please describe.**
People need to learn how to solder. It would be really nice, if the iron would come with some explanations.

**Describe the solution you'd like**
I'd like if there was a point in the settings/menu that said "How to solder". A QR code to a good resource would be nice, but very much not possible, eh? 

So maybe the text from https://www.makerspaces.com/how-to-solder/ (or a better source) can be edited and inserted there.

A few lines of text should find some space and it would be a nice gimmick.

**Describe alternatives you've considered**

**Additional context**

discip, 2022-07-31:
@FreifunkerEZ
Good evening,
are you serious?
Are you aware that April 1st is still a while away?

Did you hook up your iron to a screen of some sort?
And have you tried to read at least the explanation text of the individual settings using the irons screen? Even in fast mode?

Assuming you did so, you are really asking for a **whole instruction** to be put on such a **tiny screen** and want _in all seriousness_ to make people ***learn how to solder*** by being forced to read letter by letter of of a **`96 x 16 pixel`** resolution screen?

Leaving aside the consideration that generally only people who know how to use a soldering iron, actually buy one!

Aside from that, the capacity of the **TS80P** is almost exhausted.
And as you already put it yourself, this *'gimmick'* at least *imho* is not worth the effort .

FreifunkerEZ, 2022-08-01:
Ya, I was kinda serious, but while I was actually aware that a QR-Code would not fit onto the display, I was somehow totally not concerned with letters. 

LoL.

I'll close this then ^^

Sorry.

closed 2022-08-01 by FreifunkerEZ

River-b, 2022-08-03:
Perhaps we could fit  web link to Github/IronOS  so people could come here to read documents on Menu/settings/other goodies.

River-b, 2022-08-03:
but there is not room in pixels for QR code I think not.