Ralim/IronOS #1353: Add GlowBoost mode to the pinecil

thanos-the-tank-engine, 2022-08-04:
The current boost mode on the pinecil, despite being quite powerful, could be a lot more powerful. GlowBoost would be an optional feature to swap out the current boost mode for one that when active simply shunts the power going to the tip instead of regulating it, allowing you to use the absolute full power of the iron.

**Benefits of glowboost**

- heat up ground planes more quickly
- easier to desolder large metal components, including soldered heatsinks
- allows you to use the pinecil to light cigarettes
-
**Cons:**

- fire hazard
- might kill your tips

River-b, 2022-08-04:
could Glowboost mode possibly allow a new user to accidentally destroy their pinecil or at least their tip?

discip, 2022-08-04:
@thanos-the-tank-engine
You are joking, right?

If not, please consider closing this request.

thanks

thanos-the-tank-engine, 2022-08-05:
> could Glowboost mode possibly allow a new user to accidentally destroy their pinecil or at least their tip?

yes that's why it would be buried in the settings with  a disclaimer before you turn it on

Ralim, 2022-08-05:
I would only add this as a compiler define; as I think I mentioned in the chat; rather than a setting itself to turn off the limiter.

River-b, 2022-08-05:
my jaw has dropped.

River-b, 2022-08-05:
> I would only add this as a compiler define; as I think I mentioned in the chat; rather than a setting itself to turn off the limiter.

does this mean there will be a special Glowboost  version that people have to compile themselves?

CRImier, 2022-08-05:
seconded, this is perfect for my water boiling usecase

prokrypt, 2022-08-12:
another vote from me

**More benefits:**
 - emergency light source
 - use all the watts, all the time, because you paid for it.

prokrypt, 2022-08-12:
> * cauterize wounds

OMG BRILLIANT!

discip, 2023-02-16:
@thanos-the-tank-engine
Are you still interested in this option to be implemented?
If not, please do not forget to close this issue. 😊

thanks

closed 2023-02-16 by thanos-the-tank-engine