Ralim/IronOS #1356: delete Logo.md  install directions, send everyone to IronOS-Meta for that

River-b, 2022-08-05:
I would like to completely delete the Logo.md file completely that is in this IronOS here:

https://github.com/Ralim/IronOS/blob/dev/Documentation/Logo.md

and just put a link here in https://github.com/Ralim/IronOS/tree/dev/Bootup%20Logo
It would sends everyone over to IronOS-Meta  since they have to go their anyway to get the Images and Python scripts.
This way the Instructions for making Boot Logo images and storage of them is only updated in one place, in the IronOS-Meta. and linked to from IronOS firmware.

[image]

I already put the same Logo.md file  completely in the README.md over in IronOS-Meta and did PR for it.

River-b, 2022-08-06:
Also Joric recommended "I'd recommend removing custom sidebar from wiki. Stock sidebar supports anchors/sections automatically and looks generally better now.

discip, 2023-02-16:
@Ralim
Most likely you missed that. 😄
Should this be done like that?

River-b, 2023-02-16:
This is so old I need to review this to see if it's even relevant now.  I haven't looked at the instructions in a while and this is from 6 months ago.

closed 2023-07-21 by Ralim