Ralim/IronOS #1360: Request to increase decimal places to 2 for calibrate voltage

River-b, 2022-08-08:
this ticket is for Potato. he wants to know if  there could be 2 decimals for the calibrate volts.
 right now it showsonly one decimal spot like  20.1v    but since the fluctuations are slow he does not see/know if it's working but if it has 2 decimals this would be more obvious.

20.13 V would be prefered behavior,  adding 2 decimals.

Ralim, 2023-06-04:
At the moment the firmware for all devices is only setup for 1 decimal places.
Entirely doable, just needs some refactoring of all devices to handle 2dp.

discip, 2023-06-05:
@River-Mochi
I have been on this for some time now, but unfortunately I can't get it to work.

Here is a little workaround where both values are visible while adjusting the voltage:
[finer-resolution-V-calibration](https://github.com/discip/IronOS/actions/runs/5181805223)

River-Mochi, 2023-06-05:
I will let people know.   the ticket was submitted for someone else request.  he probably didn't realize it would be  so hard to do.