Ralim/IronOS #1361: Improving soldering UX

nyanpasu64, 2022-08-09:
**Is your feature request related to a problem? Please describe.**
The current soldering UX (along with issues):

- Hold + to enter boost mode.
    - (high risk) Holding boost for long periods of time is ergonomically painful. Additionally you can accidentally release the + button for short periods of time, resulting in a short-press to + followed by a long-press, setting your *permanent* temperature to the maximum possible value.
- Long-press - or short-press + and - together, to exit soldering mode.
    - (annoyance) Long-pressing - is slow, and pressing two buttons is awkward and requires two hands.
- Press + to enter temperature change mode.
    - (annoyance) Changing temperature is slow and fiddly compared to rotating a (quantized or continuous) analog dial, which the Pinecil does not have.
    - Should a temperature preset system be added? I'm not sure.

**Describe the solution you'd like**
My initial redesign:

- Press + to enter boost mode. Press - to exit boost mode.
    - (medium risk) You can press Boost, and forget it's enabled.
- Pressing - outside of boost mode exits soldering mode.
    - (high risk) I found myself short-pressing - to disable soldering. This works in regular mode (and is substantially more convenient than holding -), but if boost is active it instead turns off boost only. I actually pressed - once, left the iron to cool off, and half a minute later I was surprised to see it had stabilized at 319 degrees rather than cooling off. This can lead to people burning themselves or their surroundings on the iron.
    - Rejected alternatives:
        - Make short-pressing - only turn off boost, not soldering.
            - (medium risk) Users may see temperature declining, think the iron is cooling off, but it stops cooling at the regular temperature. (I did not encounter this problem myself, so I don't know how likely it is for users to burn themselves or their surroundings this way.)
        - Make long-pressing - only turn off boost, not soldering.
            - (high risk) This changes long-pressing - from turning off soldering to turning off *boost* mode only, and users will expect the iron to cool down even when it's not.
- Long-press + to change temperature.
    - (annoyance) Difficult to change temperature, long-pressing + is untested.
        - I don't know if this is a big deal. I don't change temperatures very often, to the point I haven't even gotten to test how natural it feels to long-press +, but it may be an issue for people who do this often.
    - (medium risk) If you try to short-press + to enter boost mode, but long-press by mistake, subsequent presses will adjust temperature. I don't know how likely this is to happen.

My second safer redesign:

- Make each + press only boost for 10 seconds (make it configurable?).
    - (annoyance) This may be annoying to keep an iron in boost, since you have to repeatedly press +. I don't know if it's a problem. The boost timer feels quite nice to use when playing with solder, but I haven't done any real desoldering projects with this new firmware.
    - (confusing) The 10 second boost timer is not communicated with the user.
    - Should the + press last for 10 seconds _after_ the iron reaches the boost temperature (if it's not yet reached)?
        - What if the boost temperature is unattainable at the current power level and iron usage? Should boosting be enabled indefinitely?
        - I think this will make the boost timer even harder for the user to understand.
    - (future) Difficult to extend this UI to more than 2 temperature "modes".
        - I *really* like how you can use this two-mode UI without looking at the screen (short-press + to boost for 10 seconds, and short-press - to exit). I don't think a multi-temperature UI can allow you to reliably enter low/med/high temperature without looking at the screen (though I expect it to be possible to turn off the iron without looking, and treat otherwise as a bug). So I don't think a multi-temperature soldering UI is even a good goal to have.
        - Perhaps multiple temperature presets could be accessible from the _temperature change_ UI rather than soldering UI? I don't have any concrete proposals yet.
- Pressing - *always* exits soldering mode. To turn off boost you have to press - then +.
    - (good) More convenient than long-pressing -. I exit soldering mode relatively often, but it's not a *big* deal to long-press - *after* finishing a solder job.
    - (low risk) This may be difficult to disable boost when undesired, so the iron remains in boost longer and wears out slightly faster.
    - (major annoyance) If you don't want to boost, you have to wait up to 10 seconds for the iron to begin returning to regular temperature.
- Long-press + to change temperature.
    - (annoyance) Difficult to change temperature, long-pressing + is untested. (see above)
    - (medium risk) If you try to short-press + to enter boost mode, but long-press by mistake, subsequent presses will adjust temperature. I don't know how likely it is for users to long-press + by mistake; I haven't done it yet. But users may long-press + trying to boost the old way.

I've implemented this at https://github.com/Ralim/IronOS/compare/dev...nyanpasu64:IronOS:redesign-solder-controls and tested it in some toy soldering (not an actual project), and I'm fairly satisfied with how it behaves (though I don't know if repeatedly pressing +, or long-pressing to set temperature, is problematic). I haven't implemented a configurable boost duration though; that would require learning new subsystems and a lot of busywork.

**Describe alternatives you've considered**
I'm still not fully satisfied with disabling boost on a timer, and mapping *no* buttons to turning only boost off.

- How should "always exit soldering, not boost" interact with locked mode? Locking out "exit solder mode" makes boost mode *impossible* to interrupt early (unless the back button is programmed to exit boost mode *and* display a "cannot exit soldering mode" warning).

> Users may see temperature declining, think the iron is cooling off, but it stops cooling at the regular temperature.

I don't know if this problem even exists in practice or not. For example, since short-pressing - does nothing outside of boost mode, users won't be trained to use it to exit soldering in general. Should I test making pressing - *only* exit boost mode, and see how I like it?

- Should soldering and cooldown mode be better visually distinguished? Would doing so make users less likely to think they exited soldering, when they exited boost only?
    - For example, should cooldown, soldering, and boosting all show the *target* temperature alongside the current one?

**Additional context**

iMartyn, 2022-08-09:
How about using the accellerometer for boost control?  Perhaps vigorous shaking?  Just a thought.

nyanpasu64, 2022-08-10:
I made a build that switched short-pressing - to exiting boost (not solder) mode, and it feels fairly natural to use. So it's *probably* not a usability hazard (or hopefully safety hazard either).

I don't plan to redesign the temperature display or add motion controls for boost; those are more involved tasks than I want to tackle now.

And I think motion controls to enable heat or while the iron is hot are a bad idea. Shaking to activate boost is impossible mid-solder-joint, may splatter solder and cause burns, and may be activated by mistake, eg. when cleaning the iron tip in a sponge, putting it down or picking it up quickly, or the iron falls off a table. IMO motion controls are inherently ambiguous and unreliable, and not suitable for core functionality in a safety-critical product.

nyanpasu64, 2022-08-10:
I wish the Pinecil had an analog temperature dial. It would be so much more fluid to adjust temperature directly for desoldering, maybe you wouldn't even need boost then.

Right now, temperature adjustment is awkward: you have to enter temperature mode (and the tip starts cooling down), long-press to slowly creep the temperature up and down, then wait for a timeout to return to soldering mode then wait longer for the tip to heat up again. The only real improvements I can think of (on the current hardware) are to make the tip not cool down in temperature adjustment mode, or fiddle with the short/long press handling.

mouse wheel temperature control when

nyanpasu64, 2022-08-12:
Tried a soldering project with my new firmware;

- ...the back button short press is dangerous, i did it thinking i turned off soldering, it just turned off boost
- boost is enough for short periods of heavier heating (and 10 second boost timer is definitely better than holding +, unsure if better than no timer at all). I ended up raising the permanent temperature rather than using boost for long periods. a dial would be better for switching temperature, and probably possible over Bluetooth on the Pinecil V2.
  - Are temperature presets possible to implement (in the soldering or temperature screens), to make temperature switching less bad if you don't have a dial?
- When soldering large components connected to large planes, good contact area from iron to workpiece isn't enough, you need a thick iron tip like a cone tip, to transfer more heat up to the contract point without resorting to inordinate boosting (a thin tip will often fail to melt the joint or solder itself to the joint).

discip, 2022-08-12:
@nyanpasu64
1.
> you have to enter temperature mode (and the tip starts cooling down)

Until now I was not aware of this fact. 😲
Thanks.

2.
> The only real improvements I can think of (on the current hardware) are to make the tip not cool down in temperature adjustment mode

Sounds good! I am with you on this one.

3.
> I ended up raising the permanent temperature rather than using boost for long periods.

That is what I was about to recommend to you when I first read your post.
And in addition to this and to **2.** you have the option to adjust the step width of the **`short`**- / **`long`**-press action!

- For **`short`** you are allowed to rise the increment up to **50°C**.
- And for **`long`** you may even set it up to **90°C** per step.

@Ralim
I suggest to change the behavior of *'temperature adjustment mode'* as proposed in **2.**, but with the exception that it would not start heating if *'temperature adjustment mode'* is entered from the main menu (That is: while not heating! [by long pressing **`A`**])

thanks in advance

nyanpasu64, 2022-08-13:
Forgot to mention another oddity I'd have to fix in my current redesign: I forgot to disable boost mode when you change the regular temperature.

I'll consider setting the temperature step to 5 degrees and long step to 20, to make large adjustments faster.

discip, 2022-08-24:
@nyanpasu64
I don't know if you were following the discussion about keeping the tip heated during temp adjustment, but here is what the developer stated on the matter: https://github.com/Ralim/IronOS/pull/1368#discussion_r945206836.

After you have read the statement and since you found a way to achieve what you wanted, I think this issue can be closed, unless you are able to implement what @Ralim suggested.

sandmanRO, 2022-08-25:
What I did in my custom build was this: I preserved the current behavior for boosting (that is long click). The boost is supposed to run just for a short period of time so having it run only as long as you hold down the A button makes sense - if higher temperatures would be needed for longer periods of time you would just increase the "normal" soldering setpoint, right? However the risk described by @nyanpasu64 of accidentally increase the setpoint temperature is way to real (it happened to me as well). So what I did this: in soldering mode, instead of a single short click on A/B for adjusting the temperature setpoint I have to double click (+2 clicks within 300ms) which sets the above risk to a very low probability to occur...it does not eliminate it completely but it's less likely you would do an accidental double click followed by a long click when attempting to enter in boost mode.

nyanpasu64, 2022-08-25:
If you want to keep boost on press-and-hold (if you go that route, I'd prefer a streamlined temperature change UI, but don't know how to make it, or if it's already achievable but not the default settings), you could map changing temperatures to - only, though this will break existing workflows, and will be a pure regression for people who don't use boost.

sandmanRO, 2022-08-25:
That was just my 2 cents thought. When you do a custom build it's easy...you implement it the way you think it suits you best. Ralim, discip and the rest of hardcore team have a way more challenging job of implementing a behavior that will suit most of the users, which is not always easy.

RealTadango, 2022-09-16:
Boost for me is only used for a short periods, like 5 sec. For this the long press works just fine. 

What you describe is more like you want to have a few presets with an quick and easy toggle? Maybe add someting with HOLD + and use - to toggle presets? That would make sense for me.

discip, 2023-01-05:
@nyanpasu64
Have you settled for the current behavior? 😃
If so, please close this issue.

nyanpasu64, 2023-01-06:
This is kinda unusual but I've been using an unmerged firmware fork with 10 seconds boost and short-press back exit (rather than triggering temperature adjust), on my Pinecil for, like, months now. I rarely use boost (since I set my temperature step to 10 and long-press to 20, though still want analog temperature control over USB or Bluetooth), but frequently use short-press back to exit soldering mode. What should I do now?

discip, 2023-01-08:
@nyanpasu64
Maybe you could adapt your own fork if you know what was modified to get the OS you need? 🤷🏻‍♀️

discip, 2023-02-15:
@nyanpasu64
Can this be closed? 😊

nyanpasu64, 2023-02-16:
i don't know, whatever ok

closed 2023-02-16 by nyanpasu64