Ralim/IronOS #1363: V2 pinecil,  Advance menu, can not manually set Power to OFf

River-b, 2022-08-09:
**Describe the bug**
Advance Menu, can only set Power Limit to (Off)  - if you do factory reset

**To Reproduce**
Using (+)  I can  set from 0W -95W  but (Off is not an option)

**Expected behavior**
to be able to use (+)   to select  Off or 0W-95W 

 - Device: Pinecil V2, manufactured July 25, 2022 , first Pilot batch.
 - Release: 2.18 
 - Power adapter being used: both a PD65W- 20V and also Apple 140W-28V EPR

**Additional context**
My Pinecil V1 with 2.19 works fine and allows me to select OFF.

I would try to update the firmware  on V2, but Gamiee said there is no method yet?  is there a harder method that I could try that doesn't use the updater?

River-b, 2022-08-26:
one more note,  when I scroll through 5w-95w,  the Dot scrollbar on the right side does not move it stays stuck at the top.

JoeyNinja, 2022-08-26:
Able to reproduce this issue on the V2 with Anker 737 PowerCore 24K battery bank. However holding (+) until the dot scroll bar flashes, immediately releasing and then pressing (+) once will select OFF.

discip, 2023-02-15:
@River-b
Is this issue still present?
If not please close this. 😊

thanks

River-b, 2023-02-16:
I don't think this has been fixed yet but it's been so long that I need to find a pinecil and check it again to be sure it wasn't miraculously fixed in a recent patch. I will check this week.

River-b, 2023-02-20:
@discip  this  funny little problem is still there and repeatable.
I have the newest beta "More Unique" installed and  
I can set Power limit to anything 0W - 95W, but I can't set Off  again unless I "restore defaults" and then just don't touch it. 
It's a low low priority issue. I  just notice it does this.

River-b, 2023-02-20:
> However holding (+) until the dot scroll bar flashes, immediately releasing and then pressing (+) once will select OFF.

your trick worked for me too. I was able to reproduce this trick and indeed I got it to Off.