Ralim/IronOS #1364: V2 Pinecil  PD Debug menu does not work on EPR charger

River-b, 2022-08-10:
**Describe the bug**
PD debug menu does not work as expected. can not access information

**To Reproduce**
Plug into Apple PD 3.1   140w/28V, using certified EPR 240W/40Gbps cable
Hold down (+)  and keep holding while plugging in USB-C cable

**Expected behavior**
that eventually holding down up to a minute, it will stop twitching and allow me to see the settings in the PD Menu.

**Details of your device:**
 - Device: Pinecil V2, made July 25, 2022, sold starting Aug. 2, 2022
 - Release:  2.18
 - Power adapter being used: Apple ERP 140W/ 28V , PD 3.1 

**Additional context**
using Apple 140W charger, tried 2 different EPR cables.  flipped C tip both ways and both ends.  Still same behavior . PD menu Twitching , and when I release I can not get to PD debug settings, it reverts to the main menu to change temperature.  
even holding down for more than a minute and flipping all tips and trying 2 cables. does not work.

Note: PD debug menu works fine on my PD65W charger, only doing this on Apple charger.

I suspect I have a defective V2 usb-c port.  Reason I say this is I get a Clicking Sound sometimes with I plug in a USB-C cable,  like something is Loose on usb-c .  
Tried both of my V1 models , No click no matter  how hard or how many times I push in the cable.  what could be causing this Clicking noise?  should I open this up? 
But V2  has a clicking noise pushing in the USB-C cable sometimes if I think it's at certain angle.  could this be causing my PD debug menu to not work?   This is not the normal sound from inserting the cable into the port. And I checked it's not from me accidentally pressing a button, but noise is similar to button press. 

I'll try to record click but I only have 2 hands

River-b, 2022-08-10:
Video showing this
https://www.youtube.com/shorts/ZCFLSnipclc

River-b, 2022-08-11:
tried to test to see if  I get the 5V like Not_Daniel.  I do notice that everytime I plug into EPR charger, it initially shows me a large **5**  and then a few seconds later it switched to show 28. 
- also sometimes when I plug in,  it will start, then black screen, then menu appears again

JoeyNinja, 2022-08-26:
PD Debug menu appears to function normally when connected to [Anker 737 PowerCore 24K](https://www.anker.com/products/a1289) with [EPR cable](https://www.inter9ers.com/store?store-page=INTER9ERS-i9-All-In-One-Super-Cable-4ft-EPR-240W-40Gbps-USB4-p464220399).
Video showing menu: https://youtu.be/QUc8sbSZgnw

River-b, 2022-08-26:
thanks joey  good to know this is only Apple chargers that PD debug not working on. 
I am a little jelly you have the first  Anker EPR  140w/28V  Powerbank Battery.

River-b, 2022-08-29:
Based on MagicMeter's deep analysis of packets of the Apple  EPR charger (which was made to work with Apple laptop and not other EPR things).  Could this explain why PD Debug menu is not working correctly. 

The magsafe cable is designed to work with Apple 140W charger & Macbook. It does interesting things, cable consumes a little power, Macbook asks for 20V- 4.7Amps first, and _only_  after that does Macbook ask to enter EPR mode; a soft reset is sent to test cable, and then power adaptor enters EPR mode and sends higher power to Macbook.

see his notes on the Packets:
https://usbchargingblog.wordpress.com/2022/04/22/apple-140w-usb-c-pd-3-1-epr-power-adapter-part-1-how-it-charges-a-16-inch-macbook-pro-2021/comment-page-1/

River-b, 2022-09-03:
@Ralim  whatever you did  worked.

just loaded beta  2.19.3E02733 (01-09-22)   firmware.   Apple 28V charger now works on PD Debug menu, no issue with coninuous reboots and exiting out of PD Debug menu.

River-b, 2022-09-23:
@Ralim I want to close this ticket  but wanted you to see this note first since we _don't know_  what fixed the problem.

- I reverted back to *2.18 .A1A569A (17-07-22) , and _all problems came back_.  the V2 works for heating,  but  the **PD Debug** menu  contantly reboots using APPLE 140W charger + EPR Cables.
-  Rebooting is repeatable and does not stop.

- Trying  at least  2 different versions of **2.19 Beta**,  problem goes away , PD Debug works as expected, no infinite reboots  like in the Video Posted.  all PD Debug  works in Apple 140w + EPR cable  for 2.19 beta.   it works on newest 2.19 beta from 2 days ago as well. 

- should I close ticket or keep it open until we know what the fix was?

ReFil, 2022-09-28:
I think this may be connected to an issue I'm encountering with two brand new Pinecil V2s

Using an apple 140w power brick with either iron it will power on but cut out seemingly at random. Looking at the detailed soldering screen you can see it step down through the voltages right before the soldering iron reboots 20v-15-12-9-5. Switching to a slimq 100W brick with the same cable doesn't present any issue. I think it's the cable isn't emarked and it's causing a problem, although I've used this cable at 100w with this apple brick and a zypds board. Would a potential mitigation be to have an option to disable EPR in settings?

Ralim, 2022-09-28:
I suspect this might be fixed in the newer firmware.
If your running the stock V2 firmware this is most likely the cause.
If so, it would be great if you could try the newer firmware.
We are still working hard as time permits to get the flasher tool updated; at the moment we have a basic temporary one for Linux and Windows. Best support for this is being done by @River-b in the community chat.
If you cant join the chat / dont want to etc; just comment and Ill do some typing up here.

ReFil, 2022-09-28:
Yeah they're running stock firmware. Not sure where this community chat is, I'm sure I can figure out a temporary flashing software

River-b, 2022-09-28:
> 
@ReFil   community chat links are in the [Pinecil Wiki ](https://wiki.pine64.org/wiki/Pinecil#Pinecil_V2_Authenticity_Checker). you can choose the one you like discord/matrix/telegram

ReFil, 2022-09-28:
I don't need the extra power. My issue was related to constantly rebooting making the Pinecil unusable. I found a different brand cable worked better but still inconsistent. will be picking up an epr cable to see how that works

River-b, 2022-09-28:
@Ralim   

1. I asked Gamiee if we should get the Tmp_flasher (win/linux)  up on Wiki  or Github for easier support presently as more and more people get EPR chargers, and it's a bit tedious to type out instructions again. 
he said not to as it's temporary  and you and gamiee have one coming soon.  If soon is more than 2-3 weeks from today,  I think we should have  tmp_flasher somewhere.  
What do you suggest,  should I just use Pinecil Wiki  with some instructions? since I can take it down quickly ?  or post it in an  Issues ticket for people to use? 
2.  I'm not sure we need to have  a disable  EPR feature, up to you, just using a lower  PD60 cable does this already.

River-b, 2022-09-28:
> 

go to Pinecil wiki for link,  join live chat it's easier.

ReFil, 2022-09-28:
Thanks, I will do

River-b, 2022-11-10:
I'm closing this ticket because with  V2 and Apple 140W  EPR charger,  with  Beta 2.19  and Beta  2.20  I no longer have this issue. 

PD  debug menu is working as expect

River-b, 2022-11-10:
I'm closing this ticket because with  V2 and Apple 140W  EPR charger,  with  Beta 2.19  and Beta  2.20  I no longer have this issue. 

PD  debug menu is working as expected now with APPLE 140W charger .  unknown what fixed it  but  it works now due to whatever happened in the newer  betas.

River-b, 2022-11-10:
> 

@ReFil   community chat links in Pinecil Wiki  also tons of good info in the wiki on Power supplies.  https://wiki.pine64.org/wiki/Pinecil#Community_links

closed 2022-11-10 by River-b