Ralim/IronOS #1369: Why is IronOS not recommended for the TS100?

erantapaa, 2022-08-14:
The front page README shows that IronOS is not recommended for the TS100.

I installed a version of the Ralim firmware a couple of years ago and have been using it since without any problems. Was I just lucky? Is there a version of the firmware that is recommended for the TS100, if not in this repository then perhaps somewhere else?

paulfertser, 2022-08-14:
Hi,

On Sun, Aug 14, 2022 at 02:48:22PM -0700, Erik Rantapaa wrote:
> The front page README shows that IronOS is not recommended for the TS100.

The README says that buying new TS100 units is not recommended because
they have buggy "DFU" bootloader. If you already have a device and its
bootloader works for you or if you can use SWD to reflash to a real
proper DFU bootloader then it's not an issue.

Ralim, 2022-08-14:
Yep as @paulfertser said, the TS100 and TS80 are no longer recommended to be purchased.

 The software will still work fine, just getting it on the new devices with clone chips can be a real uphill battle. Given that other products exist that don't ship flawed bootloaders, they are the preference.

River-b, 2022-08-15:
you could buy a Pine64 Pinecil as that is  designed to use IronOS open source firmware and therefore they work with Ralim to make sure his  firmware will work on the Iron. 

The TS100 irons come with their own firmware from Miniware company.  
Loading free IronOS is optional thing people do because they like IronOS features better. However, miniware does not check in with Ralim's IronOS or care if their product will work with IRonOS when they sell it.  They sell it to work with their Proprietary firmware.

River-b, 2022-08-15:
" Please note that Miniware started shipping TS100's using cloned STM32 Chips. While these do work with IronOS, their DFU bootloader works terribly, and it is hard to get it to successfully flash larger firmware images like IronOS without timing out. This is the main reason why the TS100 is no longer recommended."

erantapaa, 2022-08-15:
Ok, thanks for the clarification!

closed 2022-08-15 by erantapaa