Ralim/IronOS #1370: change QC default in Pinecil to  QC - 12V

River-b, 2022-08-15:
Pinecil comes default with QC set to 9.0V
I recommend changing default in firmware to QC -12V  so that when people plug into  PinePower QC port they can at least get it to  use working voltage and Pinepower desktop supports 12V/1.5amp and lower QC.

Since the recommended Pinecil  official says 12V-24V, I  think QC should default to 12V
New people do not know that with a new Pinecil that they have to manually change the QC setting.

if people are trying to do something alternate with slower lower  9V - QC,  then they can manually go into Power Settings > QC to lower it. 

I tested this with V2  and V1 pinecils connected to QC port on PinePower desktop.

Ralim, 2022-08-15:
For V1 this could be done, but for V2 it cant as 12V/6.2 ohm is over 1.5A.

Generally I prefer to keep QC lower at 9 as there are QC chargers out there that cant do 1.5A properly at 12V.

discip, 2022-08-23:
@River-b
Based on this statement:
> Generally I prefer to keep QC lower at 9 as there are QC chargers out there that cant do 1.5A properly at 12V.

It would be nice if you would close this issue. 😊

closed 2022-08-23 by River-b