Ralim/IronOS #1373: Boot Logo for V2 Pinecil

River-b, 2022-08-19:
A motivated member of Pincil chat really wants to get custom boot logo on  V2.
They are willing to do research and leg work and experiment.

Any and all resources/links/ advice  so they could experiment with this themselves? 

Or  should they just wait for the official write-up on how to flash?

thanks

esmith13, 2023-01-31:
any progress with this? i'm guessing you can't just use blisp on windows with the [hex] boot logo file then?

River-b, 2023-02-02:
> any progress with this? i'm guessing you can't just use blisp on windows with the [hex] boot logo file then?

not yet,  another place for  Boot logo question might be Blisp github - the flasher.   https://github.com/pine64/blisp

Ralim, 2023-02-03:
I started a branch on blisp to support DFU files to let us do this but its not yet complete.
IronOS and IronOS Meta side this is practically done. Its just I need to do the blisp side support for DFU files.

gordon-quad, 2023-03-14:
Literally unusable without this feature.
While blisp part is getting implemented, as a workaround is it possible to build a firmware image with the boot logo embedded?

River-Mochi, 2023-03-14:
> Literally unusable without this feature.

this gave me a good chuckle while having my coffee. Art is indeed more important for some :)

wbokslag, 2023-04-12:
I commented on an issue in the IronOS-Meta repository, I suggest a workaround which does allow you to use your own logo by compiling it into the cpp codebase and flashing that. 

See my comment there: https://github.com/Ralim/IronOS-Meta/issues/32

doegox, 2023-07-08:
> > Literally unusable without this feature.
> 
> this gave me a good chuckle while having my coffee. Art is indeed more important for some than soldering :)

We're using extensively the logo feature to know which iron belongs to who, when we bring multiple ones at soldering villages or hardware CTFs... This feature is more than art ;)

gamelaster, 2023-07-08:
@doegox oh this is indeed good usage.... I will check out what we can do :)

Ralim, 2023-08-01:
Updating on this:
Three parts are needed for this to work:

1. IronOS support for reading the logo
2. IronOS-Meta support for generating the `.dfu` file
3. `blisp` supporting burning `.dfu` files

1. Done as of `80c4b58976268849b6d1c8d38df60ac5a547f5c3` (Had a few bugs to Iron out)
2. Done -> https://github.com/Ralim/IronOS-Meta/releases/tag/latest
3. Done except for Windows support (WIP now) -> https://github.com/pine64/blisp/pull/45

If your on OSX or Linux and happy to build `blisp` yourself, then logos should be fully working. Or hold tight for the prebuilds once we sort Windows out.

SoLongSidekick, 2024-01-28:
> Updating on this: Three parts are needed for this to work:
> 
> 1. IronOS support for reading the logo
> 2. IronOS-Meta support for generating the `.dfu` file
> 3. `blisp` supporting burning `.dfu` files
> 
> 
> 1. Done as of `80c4b58976268849b6d1c8d38df60ac5a547f5c3` (Had a few bugs to Iron out)
> 2. Done -> https://github.com/Ralim/IronOS-Meta/releases/tag/latest
> 3. Done except for Windows support (WIP now) -> [DFU file support pine64/blisp#45](https://github.com/pine64/blisp/pull/45)
> 
> If your on OSX or Linux and happy to build `blisp` yourself, then logos should be fully working. Or hold tight for the prebuilds once we sort Windows out.

Any update on this?

Ralim, 2024-01-29:
This was already released in 2.22 and pineflash and blisp support flashing the `.dfu` files that IronOS-meta can produce.

closed 2024-01-29 by Ralim