Ralim/IronOS #1376: Get the message that it can not find the motion sensor at first start on mhp30

wildblue, 2022-08-20:
**Describe the bug**
At first start or after reset my device complains that it can not find a motion sensor. The temp calibration is not done after startet in menu.

**To Reproduce**
Reset device, repoqer device.
Start calibration

**Expected behavior**
Find a sensor, do calibration

**Details of your device:**
Mhp30
 
 - Device: [e.g. TS80/Pinecil etc]
 - Release: [eg 2.15.40087E6]
 - Power adapter being used:
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**
v2.19

Ralim, 2022-08-20:
Hello,
Is the issue that the Accelerometer cant be found or that you are trying to do the temperature calibration?

wildblue, 2022-08-28:
both are issues in my eyes...

discip, 2022-08-28:
Sorry, but you seem to have misunderstood the question.
The question is not if this are issues, but which of the 2 you are facing.

Ralim, 2022-08-28:
@wildblue 
Can you try the latest build from actions for `dev` branch (currently this one https://github.com/Ralim/IronOS/actions/runs/2941275691)

As that should have more accelerometers included in the build.

I have no idea what you mean around temperature calibration or what you issue is, so more information would be great 😄

discip, 2023-02-15:
@wildblue
Please don't forget to close solved issues. 😊