Ralim/IronOS #1379: Some help menus revert to home instead of looping

Fluffybunny96, 2022-08-22:
When you let the animation scroll of the "Start up behavior" instead of looping it reverts back the the home screen.

**To Reproduce**
Ensure menu loop speed is on "Slow". This bug still occurs on "Fast" but occurs after 1 & 1/2 loops.
1. Soldering settings
2. "Start up behavior." OR "Allow locking buttons"
3. Enter the menu and let help text scroll
4. instead of looping like all other menus this one reverts back to home screen.

_I expect this behavior has something to do with a home screen timeout idle but i cannot find where this setting may exist._

**Expected behavior**
Menu help text should loop

**Details of your device:**
 - Device: TS100
 - Release: 2.19.A3BBABC
 - Power adapter being used: DC 18v

Ralim, 2022-08-22:
This is the timeout : https://github.com/Ralim/IronOS/blob/3a44259735fbbfed48ccab97889b9c7d6777b1aa/source/Core/Src/settingsGUI.cpp#L1024

It is currently hard coded at 30 seconds of no button press.

Does this line up roughly with the amount of time you feel has passed before it times out and drops you out of the menu back to the home screen?

Fluffybunny96, 2022-08-22:
Yes it roughly lines up.
30 seconds may not be long enough as it barely allows the help text to finish on the longer descriptions.
It would be good if there was a setting exposed to change the timeout maybe?

discip, 2022-08-22:
@Fluffybunny96
Setting the speed to **`fast`** should solve your problem, the **`slow`** pace is too slow anyway, imho.

Or the value ***`30`*** could be increased by some sane amount.

However, we should definitely avoid adding another setting option explicitly for this, as this would eat up storage space, which the **`TS80P`** has been struggling with for some time.

Fluffybunny96, 2022-08-22:
Understood, that makes sense.
Maybe the best option is to slightly extend the sleep to 45 seconds?

Ralim, 2022-08-23:
Hia @Fluffybunny96 

Yeah I think bumping sleep timeout makes most sense; since adding another option is more mess.

Can you give the files on this run a try: https://github.com/Ralim/IronOS/actions/runs/2910995538

Fluffybunny96, 2022-08-24:
Works great now.
The other thought I had was instead of creating another setting you already have a sleep timeout setting maybe it can just be linked to that setting?

PR #1384 (2022-08-24): Bump timeout for menu actions
Should Close #1379
Bumps menu timeout.

This is really only there so that an iron left in the menu will eventually shutdown.
So fairly harmless to have this be longer.
closed 2022-08-24 by Ralim