Ralim/IronOS #1387: Calibrate Temperature

River-b, 2022-08-24:
**Is your feature request related to a problem? Please describe.**
Currently,  IronOS allows you to execute "Calibrate temperature?"  even if **no** tip is inserted.   This produces a very low offset number than is not in the normal 700-1000 range. 

 For people who have not looked up the instructions for calibration or do not understand that the tip needs to be installed for this calibration, this leads to a very incorrect calibration and the Iron will not operate correctly (temperature on screen would not be the temperature of the tip).

**Describe the solution you'd like**

1. IronOS  checks if tip is **Installed**, and **not** allow calibration to proceed if it detects No tip is inserted. instead of Offset number, it displays  "Install Tip first"
2. Change the name from "Calibrate temperature?"  to "**Calibrate Tip Offset?**"

**Additional context**
Changing the name has 3 benefits and could be done now independently of doing action #1 if that is more difficult to code.

1. Some may not look up the detailed instructions when doing calibration, and then worry that the number displayed at the end is a very high temperature. The name change woud disuade people from assuming the Offset is a number related to F/C temperatures and then worrying that there is something broken on the pinecil/firmware.
2. The new name will encourage people to ask in chat or look up details before proceeding, because "Tip Offset" requires more investigation/thought than "Temperature" (cold junction compensation is more complex and calibration could be worse if done randomly without following exact environmental steps).
3. People will not incorreclty press "Calibrate temperature?" many times in a row thinking it will be better and more precise if they do it more times when it does not as all elements on the PCB are getting warmer, and this leads to Offset numbers that vary greatly and the calibration is worse than before.

Ralim, 2022-08-24:
(1) Eugh that's a dumb bug. Will fix.

(2) Thats in the translation files for each language. Do you want to change it or shall I, also does @discip have some wisdom on this?

River-b, 2022-08-25:
If @ dicip is willing to give me some advice I could do it.  also if he could help change some of the other languages.

1.  I just have to Change it in Translation.EN file ? I know how to do this I'm familiar with them from the vietnamese one 
2. how do people know to change it in the other 30  files to  "**Calibrate Tip Offset?"**   or do we just use google translate?

River-b, 2022-08-25:
TranslationEditor for English is displaying this error.  Did not touch Hall Sensor area, but it has a **Length problem.**
see photo.   I will submit the PR with the Calibrate change, but  Hall sensor need looking at if Length is too short.

[image]

discip, 2022-08-26:
@River-b
In my opinion **`Calibrate Tip Offset`** will confuse users even more, since the term ***offset*** could also mean the distance from the farthest end of the tip to the actual iron *[length of the tip]* *(View of a new or unexperienced user.)*

So we either would need a *precise* menu item name like **`Calibrate Cold Junction Compensation`**, which will not fit on the screen due to its length, or we leave it as it is and add the corresponding information to the description (scrolling text).

Another idea I had, was to remove the manual calibration option all together and make the calibration to be executed on every startup in the background as long as a tip is installed. Also adding an option to set **`Auto temp calibration`** on or off.

discip, 2022-08-26:
@River-b
@Ralim
In my opinion **`Calibrate Tip Offset`** will confuse users even more, since the term ***offset*** could also mean the distance from the farthest end of the tip to the actual iron *[length of the tip]* *(View of a new or unexperienced user.)*

So we either would need a *precise* menu item name like **`Calibrate Cold Junction Compensation`**, which will not fit on the screen due to its length, or we leave it as it is and add the corresponding information to the description (scrolling text).

Another idea I had, was to remove the manual calibration option all together and make the calibration to be executed on every startup in the background as long as a tip is installed. Also adding an option to set **`Auto temp calibration`** on or off.

River-b, 2022-08-26:
idea is for people to use it less.  
right now we have people coming to Pine64 live chat rooms, using it 10-15  times in a row and seeing wide varience of the offset number and drawing wrong conclusions. 

 if it says "calibrate Tip Offset,"   they might stop and read the procedure in document before making  Calibration worse.

Other issue we see is people wrongly assume the number at the end is _temperature_ since it's called "calibrate temperature"  and then they start a panic in the live chat  saying " this is bad, pinecil is broken, showing 900 degrees message at end of  calibrate temperature,  this is a serious bug Ralim needs to fix this asap"  
...because they did not look up instructions first, and not know the 900 is Offset number and nothing to do with °C

- hope is calling it Tip offset, they will first look up instructions or ask in chat the procedure. This is only supposed to be done if they have Temperature off of  more than 5 °C  and not often, since more chance of making calibration worse. and they would need to have tool to measure if the tip is way off in temperature.  Or if the tips are visibly burnt or wearing out too fast, this is another sign that calibration is wrong. 

"calibrate temperature" makes it too easy and then people just randomly use it, thinking they know what they are doing but more often it's calibrated incorrectly which leads to worse problem.  I also changed the scroll message in the Menu to make it more clear, does not need to be done if you are within 5 °C  which might help.

River-b, 2022-08-26:
@discip  are you able to fix the  translation editor tool?  it's not giving enough room for Hall Sensor  section which is why it shows in Red color if you use  translationEditor.html  on EN   english one.

River-b, 2022-08-26:
Automatic on every re-boot is problematic.  if they  were using the iron for a bit, and just unplugged to swap tips the handle and components inside would be warm.  Warm handle does not meet requirements that  Both handle and tip are Equal in temperature and  at room/ambient temperature  (for proper calibration).

River-b, 2022-08-26:
@discip  
about your idea,  "Cold Junction Compensation"  also called "**Seebeck Effect**"  since Thomas Seebeck discovered this.

could call it "**Calibrate Seebeck Effect?**"  which displays as 

> Calibrate
> Seebeck effect?

it's also sometimes called  "cold junction reading"

River-b, 2022-08-26:
[Seebeck Effect](https://www.tegam.com/what-exactly-is-cold-junction-compensation/#:~:text=At%20the%20cold%20end%20the,the%20basis%20of%20a%20thermocouple.)

discip, 2022-08-26:
1. If we start to apply your principle *(use cryptic names to make user look it up)*, than we just have to replace every menu item name with a number so people definitely will have to look up what setting they are about to change.

   Please get me right.

   There will always be people who keep making wrong assumptions due to them being lazy, including myself!

2. If the components need to be as close to room / ambient temperature as possible, we should definitely calibrate the offset as early in the boot sequence as possible.

   Btw: You don't need to power off your iron to swap the tips.

3. I read about the **`Cold Junction Compensation`**.

I stick to my suggestion, but would like to add some more **`ifs`** to the already existing ones https://github.com/Ralim/IronOS/issues/1387#issuecomment-1228523045.
e.g:
* Is there a way to measure absolute time?

   Most likely not, but the idea is to log the latest temp and depending on that at boot, check if enough time has passed for the tip to cool down to room temperature and if so, calibrate the offset. Maybe @Ralim has a better solution for that.

discip, 2022-08-26:
1. If we start to apply your principle *(use cryptic names to make user look it up)*, than we just have to replace every menu item name with a number so people definitely will have to look up what setting they are about to change.

   Please get me right.

   There will always be people who keep making wrong assumptions due to them being lazy, including myself!

2. If the components need to be as close to room / ambient temperature as possible, we should definitely calibrate the offset as early in the boot sequence as possible.

   Btw: You don't need to power off your iron to swap the tips.

3. > [Seebeck Effect](https://www.tegam.com/what-exactly-is-cold-junction-compensation/#:~:text=At%20the%20cold%20end%20the,the%20basis%20of%20a%20thermocouple.)

   I read exactly that article.

I stick to my suggestion, but would like to add some more **`ifs`** to the already existing ones https://github.com/Ralim/IronOS/issues/1387#issuecomment-1228523045.
e.g:
* Is there a way to measure absolute time?

   Most likely not, but the idea is to log the latest temp and depending on that at boot, check if enough time has passed for the tip to cool down to room temperature and if so, calibrate the offset. Maybe @Ralim has a better solution for that.

River-b, 2022-08-26:
2.
In  V2 Pinecil we **do** have to unplug  when swapping tips, because it auto-senses on re-boot which Tip ohm we are using since Pinecil now has both 6.2 ohm tip and  8 ohm tip  (technically would not need to  unplug if  same ohm tip size is swapped)
I also unplug for safety  when swapping tip.  but V2 , it's  only going to detect tip ohm on **unplugging/reboot**. 

auto-calibrate on every reboot  would be problematic because the handle or tip could be warm, the person just used it. 

Technically these new  shorter tips may work in other TS100 style irons too, but not sure,  shorter tips do require more  powerful charger.

 -  In V1 pinecil,  we are getting a manual menu change that allows us to manually select 6.2 ohm or 8 ohm  Tip,  every time we change tips since  V1 does not have the hardware to auto-detect Tip  on boot-up.

discip, 2022-08-26:
> In V2 Pinecil we **do** have to unplug when swapping tips, because it auto-senses on re-boot which Tip ohm we are using

Ah, I see. Did not know that. But this is covered by 4.

> auto-calibrate on every reboot would be problematic because the handle or tip could be warm, the person just used it.

That is why I wrote this:

> but the idea is to log the latest temp and depending on that at boot, check if enough time has passed for the tip to cool down to room temperature and if so, calibrate the offset

discip, 2023-02-15:
@River-b
Do you think this can be closed after all the latest changes?

discip, 2023-02-15:
@River-b
Do you think this can be closed after all the latest changes? 😊

closed 2023-02-16 by River-b