Ralim/IronOS #1394: mhp30 not using qc3

wildblue, 2022-08-28:
when connecting the mhp30 with a usbc to usba cable i get only 5v, 3w.
seems there issomething missing for qc3.

**Describe the bug**

**To Reproduce**

**Expected behavior**

**Details of your device:**
 
 - Device: [e.g. TS80/Pinecil etc]
 - Release: [eg 2.15.40087E6]
 - Power adapter being used:
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

discip, 2022-08-28:
Please fill in the missing details to help the developer finding the cause.

> * Device: [e.g. TS80/Pinecil etc]
> * Release: [eg 2.15.40087E6]
> * Power adapter being used:
> * If this is an accelerometer related issue, please include its model number here:

Ralim, 2022-08-28:
MHP30 is missing QC support, and is only designed for PD support to the best of my knowledge.
The unit I have here at least, does not have the required hardware for QC negotiation.
Nor have I seen a QC3 port capable of 20V5A (yet).

BF3000, 2022-08-30:
> MHP30 is missing QC support, and is only designed for PD support to the best of my knowledge. The unit I have here at least, does not have the required hardware for QC negotiation. Nor have I seen a QC3 port capable of 20V5A (yet).

And newer will I think. You can't transfer 100w of power with random usb cable, so there is a reason why Qualcomm adopted PD standart as a base for QC4. Cables with id chips.

wildblue, 2022-08-30:
okay, i may have made a typo at the usb-a capabilities. but even my 6 year old phone is able to draw 1.5A at 12V over that connection

Ralim, 2022-08-30:
Yes, but 18W is not very much power compared to the 60W the MHP30 is designed for.

discip, 2022-08-30:
@wildblue
Arguing back an forth will not change the hardware!
Since this is clearly no software related issue please close it! 😊

Thanks

wildblue, 2022-08-30:
i know. normally issues do not get closed by me. i woyld have done that after getting the information that the mhp30s hardware is not capable of running on a normal usb power supply.

i am quite confident to have read in the documentation of iron os that it supports qc2, and qc3. you may have to document that (not) feature of ironos different to avoide making other people having the same (non) issue with the software.

and at this moment a had closed this issue, but with the original firmware my device is not throtteled at 5V 3W. How do i fix that misbehaviour  of my hardware with ironos?

discip, 2023-02-15:
@wildblue
Did you manage to overcome this issue somehow?
Please do not forget to close this issue. 😊

thanks

closed 2023-07-21 by Ralim