Ralim/IronOS #1407: Ts100 tips support

77LoRd77, 2022-09-08:
Hello! Made a quick test using a modified ts100 tip in ts80pro

All seams to be ok exept for temperature

The reading is way of( the iron is melting solder when shoving only 100-120°)

Any ways to add support and identification of the ts100 tips?

Ps. 1. 4 pins jack was used since it was the only one laying around
2. The tip's contacts seams to be SS or something else that i am not able to solder properly(soldering is not stiking to it) will fix it latter(maybe weld it), but good enought for a test
[image]

Ralim, 2022-09-10:
Hello,

I would love to understand why you would do this change?

For the TS80P there are a few things that would need to be changed to support this:

1. Change the tip resistance to 8 ohms from the 4.5 that the TS80 uses
2. Change the lookup table that is being used as the tips have different response curves
3. Change the PWM timing for the tip to compensate for different characteristics

Additionally you would also need to change:
1. The resistance used for the op-amp feedback gain
2. Not use any soldering in attaching the jack to the tip, as you need like materaisl for the thermocouple operation to work correctly
3. Otherwise, you would need to look at relocating the cold junction compentation resister closer to the tip junction you have added.

In the current assembly using soldering you have an unknown thermocouple response as you have introduced quite a few different metals into the construction of the "tip".

Adding support for this to the firmware would be a non-trivial undertaking with the current means of assembly.

discip, 2022-09-15:
@77LoRd77
It seems that you missed the comment above. 😊

77LoRd77, 2022-09-15:
Hello yes i did miss it, thank you

The reson why i would like to do that is due to the cost of ts80 tips, they are 3x the price of ts100
I like the usb c power option and the quick tips replacment of ts80

I am not sure how bad the soldering will affect the thermocouple response, but i presume it should not be supper bad and tips should perform more or less adequate event with a slight temperature variation of -+15° here and there

But mayby its too much work to modify the tip of ts100 everytime, maybe pinecile is a better option

discip, 2023-02-15:
@77LoRd77
> But mayby its too much work to modify the tip of ts100 everytime, maybe pinecile is a better option

> Additionally you would also need to change:
>
> 1. The resistance used for the op-amp feedback gain
> 2. Not use any soldering in attaching the jack to the tip, as you need like materaisl for the [...]

Since you seem indecisive and the changes to make this work are huge, I'm wondering if you're still interested in proceeding.

Please consider to close this. 😊

thanks

77LoRd77, 2023-02-15:
Yeah, bought a set of tips for ts80 in the end

closed 2023-02-15 by 77LoRd77