Ralim/IronOS #1409: help with TranslationEditor needed

discip, 2022-09-14:
@discip  its would be better to put this into its own issue than bumping a really old thread.
I cant see any difference between the two you have posted other than formatting, so you might need to clarify what the difference is that you are looking for as well

_Originally posted by @Ralim in https://github.com/Ralim/IronOS/pull/249#issuecomment-1244938277_

1. > other than formatting

   Yes, that's exactly it! Only the format.
The actual one takes up much space (scrolling wise). Some languages were formatted like that (narrower) before I updated them to match the output of **`TranslationEditor`**.

2. Sorry, I don't know how to make a visual comparison between the two. 😅

   actual one:
https://github.com/Ralim/IronOS/blob/dev/Translations/translation_EN.json
**vs**
reordered one:
https://github.com/discip/IronOS/blob/test/Translations/translation_EN.json

discip, 2022-09-20:
@Ralim
Have you had time to check if this is somehow achievable?

discip, 2022-09-20:
@Ralim
Have you had time to check if this is somehow achievable? 😊

Ralim, 2022-09-24:
I dont know to be honest; this is going to come down to how javascript is serialising it out.
As its just a formatting change, I dont see the big issue though?

discip, 2022-09-27:
@Ralim
Is your answer meant to cover both topics?

It would really help if I could use **`TranslationEditor`** to make editing all the language files easier. (2)

closed 2022-10-22 by discip