Ralim/IronOS #1410: LED power and possible SPI connectivity

Herushan, 2022-09-15:
Was asked to write an issue (by River B.) on adding controllable LED lights to the Pinecil V2. Have a design with just power currently and wonder if utilizing the empty U10 pads would work for PWR and GND? If that does not work the discord folks said the 5V rail might be an option as well and where is the best place to connect to that then?

In discord someone wanted SPI controlled LEDs and I would like to know where is the best place to connect to SPI and if it would be possible for the OS to support them?

River-b, 2022-09-16:
could you add a sample image of the LED design and also  the link to the  suggested  LEDs from brian that can connect to SPI?

Ralim, 2022-09-17:
Hia,

Absolutely you can tap power off U10 to the best of my knowledge. You should otherwise be able to find a 4-5V regulated rail after the DC/DC that you can use as well.

In terms of controlling the LED's; Ill need to check if we have the hardware SPI exposed anywhere. If its possible to map the hardware SPI pins out; definitely happy to look at adding support.

River-b, 2022-10-04:
@Ralim  there is a 4.2V rail on the schematic.
There is also 3.3V  regulator. Nusacat mentioned the 3.3V reg has a 300ma limit, how much is left?
Could we use either 3.3V or the 4.2V rail or  should we skip the 3.3V rail for only the 4.2V  for power on LEDs?

Ralim, 2022-10-04:
If possible I would use the 4.2V rail. There should be headroom on the 3.3V rail but wouldn't want to risk instability.

River-b, 2022-10-30:
@Herushan / ahall   made the first protptype  LED light today for  Pinecil V2,  red wire is going to CAP  C8 for 4.6v and black wire is going to GND  leg  of  unused U10 spot.

since it is GND,  he thinks it should also be okay for V1 original owners to also use the same U10 spot/pin  for GND even though U10 is occupied by TMP sensor in all V1's.  He was concerned about using U10 for power bc of affecting  readings for V1 users and it was not clear on schematic if  U10 is using 6.3V or is that a typo?  For now he just soldered to side of C8 CAP  for  4.6V power.

[image]
[image]
[image]
[image]

River-b, 2022-10-30:
if we can not use SPI  to control  on/off then  another possibility is small on/off switch on the bottom.