Ralim/IronOS #1415: V2 always reads 6.2V for Tip Res in debug menu

SolderEater, 2022-09-21:
I have 2 Pinecil V2's.  One running 2.19 beta, one running 2.18 stock firmware.  I'm not hot swapping the tips, but always unplug the pinecil between tests.  With 4 different long tips that measure between 7.5 and 8.3 ohms on my HT118A meter, both pinecils always show 6.2 Ohms for Tip Res.  I even tried a T12 that shows 8.0 on the nose on my meter.  Pinecil still says 6.2.  It also shows 6.2 with a 6.2 tip.  With no tip inserted, it says 0.  All the rest of the time it always shows 6.2 for Tip Res.  I also took it apart and cleaned the contacts and their mounting spot with alcohol.  Even scratched them up a bit.  Still 6.2 on all tips.

Ralim, 2022-09-22:
[Pinecilv2_EN.bin.zip](https://github.com/Ralim/IronOS/files/9621936/Pinecilv2_EN.bin.zip)
Here is the test firmware. I've swapped out the tip res option in the debug menu to instead show the raw reading.
Would be fantastic to know what reading you are getting for your tips :)

SolderEater, 2022-09-22:
Now it reads Tip Res 4696, 4800, 4700 for long tips.  An outlier is 4317 for a long knife tip that only measures 7.2 on my meter.
  4664 for a T12   and 3495 for a short 6.2 tip.

Ralim, 2022-09-22:
[Pinecilv2_EN.bin.zip](https://github.com/Ralim/IronOS/files/9624562/Pinecilv2_EN.bin.zip)

Hia,
Please give this a try; I've shifted the threshold down a little.

SolderEater, 2022-09-22:
That fixed it.  At least on the first long tip I tried.  Now says 8.0.  I'll try more tips later today and let you know if I see any issues.  If I do have any I will also give you the raw reading of any tips that still read 6.2

SolderEater, 2022-09-23:
This is working correctly now with the 5-6 tips I have tried.  Thank you very much.

closed 2022-09-23 by SolderEater