Ralim/IronOS #1417: Pinecil V2,  corrupted Snowy I2C screen using beta 2.19 firmware

River-b, 2022-09-25:
**Describe the bug**
using 2.19 Beta for Pinecil V2, get snowy screen when V2 is left to idle in either main screen or in Debug mode. shaking or flipping V2 brings it back to normal view, no need to unplug. 

**To Reproduce**

Ralim, 2022-09-25:
Can you please confirm if this occurs for you on : https://github.com/Ralim/IronOS/actions/runs/3120485217
What git hash is in your `2.18 stock` firmware you are referring to?
For the V2 it occurs on, what is the manufacturing date?

River-b, 2022-09-25:
2.18.A1A569A   which was firmware.bin  that Gamiee included with our  tmp_flasher zip. 
of 10 tests with 3 different chargers,  I did not get corruption on  2.18  but I just got  corruption now  using APPLE  PD140 and EPR cable while in DEBUG menu.  let me confirm a few more times.

Ralim, 2022-09-25:
Just use an EPR charger for testing, no point testing across multiple. EPR puts most strain on I2C.

River-b, 2022-09-25:
ok !

River-b, 2022-09-25:
I am only able to test  2.18 with the EPR  Apple PD140W since it is only EPR I have that natively does not cause 2.18 to reboot.

River-b, 2022-09-25:
i just tested with  2.18.A1A569A  , 2 runs left idle for 30 Min each time, on Debug, Time(uptime) view,  and it just stay on showing the time number increasing, no  Snow corruption,  no black screen, all good.

Ralim, 2022-09-25:
For me I've had 2.20.0DDBE80 going for over an hour without any issues. 😓

River-b, 2022-09-25:
FYI  ran  2.18 again on V2  for 58 min in debug mode,  with no corruption

I will test Beta posted

River-b, 2022-09-25:
After 50 minutes on  two  V2, 2.20 beta,  everything looks good no Snow or I2C corruption. 
-  I still see the Uptime display though it did not go to black screen.  is this intentional it stays on?
-  if it's idling for  long time,  it is desired it  does Shutdown (black screen) per shutdown setting that is custom selected in sleep settings.  my sleep shutdown is 5 min.  but in Debug mode it stay **on** even though I  did not touch the Iron for 50 minutes.

Ralim, 2022-09-25:
Debug mode should not have the auto shutdown mode; so its expected the screen stays on, this is by design as the debug screen is for debugging issues, and not normally entered.

All other screens should have some form of shutdown though.

VICLER, 2022-09-25:
I have the same issue on pinecil v2 with stock 2.18. I'm using PD 20V. After some idle time the screen is full of snow. Shacking does not help, only unplug.
Where can I download a newer beta? 
@River-b (and why is there a 2.2 beta, but the last release is only 1.19, where is 2.0 and 2.1?)

River-b, 2022-09-25:
> I have the same issue on pinecil v2 with stock 2.18. I'm using PD 20V. After some idle time the screen is full of snow. Shacking does not help, only unplug. Where can I download a newer beta?

This was not happening for me in stock 2.18 , only  people who were using 28V  PD140W  chargers and Beta 2.19.
 could you give use more info to make sure this is related issue?

1.  which screen/menu   are you in when you leave the  iron to idle and it becomes snow?
2.  about how long does it take to become snowy?
3. what is exact  full number  2.18  version you have?
4. what is make/model of your power adaptor?
5.  could you test the following:  when it get's snowy,  not only shake, but just flip it over couple times, that sometimes brings screen back. also and (-) button a few clicks. Does that work to bring back the  normal screen?

SolderEater, 2022-09-25:
I had snow with 2.19 using Rocoren EPR charger.  After 2.20 flash I still had snow.  Then reset to defaults and no snow.  I kept looking at detailed idle screen as the culprit, but turning it back on and still no snow using 2.20.  So far 2.20 looks good.

SolderEater, 2022-09-25:
My 2.20 on EPR power is snowing.  To trigger it it seems all I need to do it turn on detailed idle screen.  At the idle screen, maybe 5 minutes after it goes black I gently wake it up with movement.  I'll have a screen that looks like this  https://drive.google.com/file/d/1-oxvlKp9NsFNCkr0zVFC3Uht2Ay0Lp06/view?usp=sharing   It looks like it's writing random setting screens to the wait screen after the screen blanks out.

River-b, 2022-09-25:
> My 2.20 on EPR power is snowing. To trigger it it seems all I need to do it turn on detailed idle screen. At the idle screen, maybe 5 minutes after it goes black I gently wake it up with movement. [...]

Flux,

1. To ble clear turning on Detailed idle screen mode makes the snow happen?  but if you are not in detaield screen you do not get snow?
2. what screen are  you testing to idle or  in?  you can just copy and paste or Drag and drop the images directly here. Github allows it.

River-b, 2022-09-25:
@Ralim  I am also getting Snowy  screen when  I have V2  set to Detailed screen.  it takes 

> I have the same issue on pinecil v2 with stock 2.18. I'm using PD 20V. After some idle time the screen is full of snow. Shacking does not help, only unplug. Where can I download a newer beta?

beta is in the post

JoeyNinja, 2022-09-25:
Using Anker 140W EPR Battery.
On 2.18: I've had snow appear on the main screen. 
On 2.19.FFD2C97: Main screen is fine, snow eventually appears on debug menu (tested on uptime screen). Have detailed idle enabled.
On 2.20.0DDBE80: So far no snow anywhere. Have detailed idle enabled.

River-b, 2022-09-25:
Tested two different V2 V2-a is apple 140w. V2-b is Kovol 240w.

V2-a is detailed idle setting 2.20. main screen goes to black after a few minutes. When I pick it is slightly corrupted. Flipping it cleans & fixes display (photo).
V2-b 2.20 is normal no detailed settings. Main screen goes black as expected when I pick it up sometimes it looks good & sometimes it has strange words that I have no idea where they came from [image]
[image]
[image]

JoeyNinja, 2022-09-25:
Had a similar thing happen when idling on 2.20 non-detailed screen. Backwards letters for some reason.

[image]

River-b, 2022-09-25:
It seems better with 2.20 but if it can't be perfect, it does not seem to affect functionality, it just makes V2 look like it's broken.  I lightly flip V2, then it refreshes the screen and everything is back to normal.  All functions are good.  I'm happy to live with this if it can't be fixed.

Potato suggested in the chat that maybe a real fix might need an extra  Resistor or CAP on the PCB to help with EPR chargers  stressing the I2C?  for me this only happens if  I leave it to Idle  while I'm working and just pressing (-) button or gentle flip brings it back to normal.
If it is affecting operation for anyone, then that is different.

River-b, 2022-09-25:
It's also not consistent, out of 5 tests now  with 2.20 ,  my V2  that has  normal view (no detailed)   seems fine if I leave it idle sometimes ( it might be how long I leave it idle).  Sometimes  it will go black and I pick it up and it's normal, other times  it goes black and I pick it up and half the  main screen is corrupted or has strange words.

the other V2 with detailed screen enabled is more consistently having  Snow every test but it's limited snow compared to 2.19 which was most of the screen.

River-b, 2022-09-29:
2.20 still have corruption. 
Just had this happen while during active desoldering , not even idle. Using Kovol 140 charger & cheaper Epr cable. 
I was in soldering temp screen & soldering, noticed it switch to corrupt parts of user interface menu. 
When it corrupts it seems to like to switch to parts of word interface or user interface screen . [image]

River-b, 2022-09-29:
- 2.20  all good
- using my 2nd V2 for last hour of soldering
- did not get corruption at all
- using same kovol 140 charger and cheap EPR cable
-  @Ralim  I'm wondering if there is something wrong with my new  V2 .  both of them soldering  with all same variables, one is fine no corruption during soldering at all, and the other one is corruption within minutes of active soldering and it happens every time.  the Good one only gets corruption if I leave it to idle a long time but  not while actively soldering.
- but how  to I prove this or fix this?   this same V2 that gets more corrupted screen,  also strangely shows  less volts on every single charger I have.  if  the good one shows 28V, the bad one shows 27.5V,  similar if I use 20V charger.

Ralim, 2022-10-03:
The most likely issue here is just the OLED / I2C bus. At the moment we are running the I2C bus faster than required so my first plan is to test lowering it down. Just want to unbox my gear and get some good looks at the I2C bus to see if we are running it on the edge of stability. (Will try adjusting the timing to see).

I would expect something like 5-10% variance once we add up all the part tolerances; so if we are near the edge as it stands that would explain this. If so; we can just drop speed by like 10-20% without any drama. Partially not helped by less than clear docs on the I2C in the BL706 when I was setting it up.

Main way I plan to look at this is:

(1) Make test firmware with different I2C settings and compare
(2) Look at the bus data stability and timing 
(3) Most likely adjust timing or slow down

And finally;4, where I have a mild hunch that _something_ in the I2C doesn't like display on/off toggles. I've wondered if the voltage rails go a little unstable. Have thought about having a bit of a backoff where after toggling on/off we force a few ms of bus silence for it to stabilise.

I suspect the snow is mostly the I2C in the OLED getting out of sync to the data. As dropping a bit during commands would cause this skew.

River-b, 2022-10-03:
I could also test this week only soldering with PD20V-65W charger to see if  this  eliminates snow?

mikerocklewitz, 2022-10-27:
I have this oled issue after adding hall sensor on one for sure. Not sure about the other yet havent rly used it much. Appears randomly.

River-b, 2022-10-31:
update on further usage:  using the V2  for a while now and only using PD65 cable all week,  confirming that corruption of the oled screen happens often during soldering, even just a few minutes of soldering, not when it's hot or long use. 

 In the normal course of using it or placing it in stand and back and forth, I have to flip it over  or shake it  pretty much  every time I  look at screen,  the OLED is messy  and I have to flip it to clear/refresh screen.  
 I find myself  grabbing my V1  more and avoiding the V2 so I can work  with a steady screen and know whether it's hot/heating or sleeping and needs (+) button hit.

mikerocklewitz, 2022-10-31:
> update on further usage:  using the V2  for a while now and only using PD65 cable all week,  confirming that corruption of the oled screen happens often during soldering, even just a few minutes of [...]

Can confirm.  Happens moreso when soldering.  Cannot reproduce it on the bench or in a stand.

River-b, 2022-11-01:
maybe  could I get a build from "test-slow-i2C" branch? 
I could test that  even if it's slower screen draw to see if  corruption completely goes away.
This would  either confirm or deny  the hypothesis that it's related to I2C clock  as it might not be. 

I could report back after some soldering which is where it affects me the most.

River-b, 2022-11-11:
Could people with snow corruption screen let me know .  Did all of you install a  Hall Effect sensor at U14 or not?  thanks

mikerocklewitz, 2022-11-11:
> Could people with snow corruption screen let me know .  Did all of you install a  Hall Effect sensor at U14 or not?  thanks

I have installed a hall sensor.  But the snow effect is not consistent.  Very random actually.  I can never make it happen. Sometimes it happens soldering. Sometimes it doesnt.

VICLER, 2022-11-11:
> Could people with snow corruption screen let me know . Did all of you install a Hall Effect sensor at U14 or not? thanks

I have not installed anything. I have this corruption on a new device out of the box

SolderEater, 2022-11-13:
I still have it on an unmodified V2 running 20.5416783.  It does seem improved over the original firmware.  It doesn't start to snow nearly as quick as it used to.

jjwebb, 2022-11-14:
I found this issue page by googling "12.18.A1A569A". I have this same issue with an off-the-shelf Pv2 I just bought, with that firmware. Screen goes snowy after idle time or soldering. Only way to fix it is to unplug it, and plug it back in. 65W 20V USB PD. I set the idle and solder screen to "detailed" straight out of the box; I'll see if disabling that makes any difference.

River-b, 2022-11-14:
> I found this issue page by googling "2.18.A1A569A". I have this same issue with an off-the-shelf Pv2 I just bought, with that firmware. Screen goes snowy after idle time or soldering. Only way to fix it is to unplug it, and plug it back in. 65W 20V USB PD. I set the idle and solder screen to "detailed" straight out of the box; I'll see if disabling that makes any difference.

don't need to unplug it,  just gently  flip it over and it will refresh the screen. gently flipping works better than shaking to re-draw the screen. 
 there is a 2.20 beta I tried but the issue is not gone completely.  Ralim is working on  testing lower  timing speeds to help with the noise/corrution. 
If you want to try the beta ,  go to Pinecil Wiki,  links to community chat  there,  we have the Beta  flasher in there that helps load the beta firmware.  or just wait until Ralim  makes  the next beta since current beta still has  i2C corruption.

River-b, 2022-11-16:
For most people right now,  recommend to just use firmware  beta  version  **2.20.3A96FF1**  from bottom of page of this [Action link](https://github.com/Ralim/IronOS/actions/runs/3409043351) . Get the Pinecilv2.zip and pull the Pinecilv2_EN.bin from it. you don't need anything else from the zip.
 
1. It fixed all my i2C corruption using  PD65w, 20v, 3.25 Amp chargers.  Tested while soldering.
2. Only problem is it makes all my  EPR 140W chargers reboot/not work.  as long as you don't have a PD3.1 charger, it should work. 
3.  Please try this and report in here if   **2.20.3A96FF1**  beta  works for you. it is at  bottom of this page. https://github.com/Ralim/IronOS/suites/9161524538/artifacts/426541610

You will need a Flasher to get it onto Pinecil V2. 

1. For the **Flasher** we have a command line one for Linux and one for Windows.   you can get them from  Pinecil  Community chat for now since they are not fully released. All the different community chat links are in [Pinecil Wiki ](https://wiki.pine64.org/wiki/Pinecil#Community_Chat_links).  Since I only have Discord, I posted them to PINs at the top of Pine64 Pinecil Discord. If you join by  Telegram, just ask someone in Pinecil and they could help you or search for words "Blisp Flasher"  and you will find it. 
2.  To put Pinecil  into Flashing mode. in Linux or Windows,  do this:
 Connect V2 to PC by long holding down (-)  button **first**,    then plug in usb-c cable.
 Release the (-) button after 20 seconds (do not plug in cable to  V2 until you have (-) button pressed down first).
3. V2 screen will stay BLACK/blank which  means you are in flashing mode, otherwise repeat step 2 or find another usb port/cable or PC since you are having connection issue with the PC port or cable.

mikerocklewitz, 2022-11-16:
Will try tonight/tomorrow.

River-b, 2022-11-16:
Successful Flashing of V2  in windows Poweshell looks like this

[image]

River-b, 2022-11-16:
Files  that should be in the Windows Folder and data folder for this to work. The Bin file you get from either Pinecil release zip in Github, or from Actions zip if it's an unreleased Beta,  or from Ralim or trusted source if  he made a special tester firmware he sends out.  V2  only uses the BIN  file for the language you select and the rest of the Firmware Zip can be deleted.

[image]

mikerocklewitz, 2022-11-16:
Instructions for a real os (manjaro)?

dannahan, 2022-11-16:
FYI, I am using a USAMS 240W USB C PD3.1 cable and a 140W PD3.1 PD-089PT Adapter. Both cheap.
Iron registers at 28V.
However, when I would get the snowy screen, I would rotate the iron to switch orientation and that would fix the issue. 
No more snow.

River-b, 2022-11-16:
> 

for linux there is a different  executable .  it's posted in the  Pinecil community chat links to chat are in  Pinecil Wiki page

River-b, 2022-11-16:
> te the iron to switch orientation and that would fix the issue.
> No more snow.

yes same for me and many other people rotating refreshes the screen.  but it's prefered  we don't have to do this many times while soldering.

River-b, 2022-11-16:
> Instructions for a real os (manjaro)?

there is a Linux  blisp   executable.   the screen capture I posted was for   my Windows  Powershell  install only and not for linux. 
the Executable for Linux  and instructions are posted in Discord ,  and it's pinned but if you only have Telegram, or other chat you can also get to it. 
See Pinecil Wiki for links to chat https://wiki.pine64.org/wiki/Pinecil#Community_Chat_links
and then you can see info on Linux flasher

River-b, 2022-11-16:
> Insulting the volunteers that are trying to help you for free is unlikely to help your cause. The commands provided with the bright orange square around them above are almost literally valid bash syntax. Besides, a "real OS user" would have no problem recognizing backward DOS slashes and turning them into forward UNIX path separators. You don't have to use the DOS exe extension, of course. (See how that feels?) Literally, run the blsisp executable (with the same relative path to the .bins) with the arguments like write -c bl70x --reset path_to_the_new_bin_file - the idea is very much the same in Powershell or Bash on MacOS or Linux or anything else. Be nice. Message ID: ***@***.***>
> […](#)

- There are no  flasher commands executed at all in my 2 screenshots.
- Screen shots  are simply showing for **Windows** users the files you need to do flashing.  I just did "ls" command to list what is in my folders of the flasher before   I'm about to execute  flashing  in windows.  
- I am a volunteer.   sharing my screenshots is meant to help people. don't see how it insults people. 
- I did not provide  any instructions for Linux flasher install here .  only **Windows** powershell install.

NovaViper, 2022-11-16:
Can confirm that I'm also getting the same corruption issue on my Pinecil V2 using the beta 2.19 software. I noticed it particularly when I was soldering a SMD chip for a TV circuit board, would just mess up the numbers on the screen with little dots all over (not as bad as op but definitely would make reading the numbers impossible). Sadly I don't have any pictures of it but I noticed it happening if I rotated the iron around too much (which I tend to do because I'm trying to reach different parts of whatever I'm working on).

I just installed the newest 2.20.3A96FF1 update and it seems to have gone away (at least from me just rotating it over and over again about 30 times rapidly) to see if I could trigger it again. Will update if I manage to replicate it again

Pinecil V2 info:
ID: fce95dE5
SN: 00008d7ad7fdd7c4

River-b, 2022-11-16:
> 2\. onnect V2 to PC by long holding down (-)  button **first**,    then plug in usb-c cable.
>     Release the (-) button after 20 seconds (do not plug in cable to  V2 until you have (-) button pressed down first).

I didn't notice it.

River-b, 2022-11-16:
> I just installed the newest 2.20.3A96FF1 update and it seems to have gone away

This firmware is from Action "Testing slower I2C for PinecilV2 #2388" 
if it would just work for  PD140W chargers, we would be done :)

River-b, 2022-11-16:
If anyone is willing to send Ralim some UART logs, it would help him narrow down and tweak the firmware.

1. First need to flash this special UART debugging firmware that ralim made 
version  2.20.A1510E 14-11-22  attached Pinecilv2_EN.bin

[Pinecilv2_EN_UART.zip](https://github.com/Ralim/IronOS/files/10019665/Pinecilv2_EN_UART.zip)

2. We have a command line Windows  or Linux flasher for now:
Windows Blisp Flasher is attached. 
[Windows_blisp_Flasher_Pinecilv2.zip](https://github.com/Ralim/IronOS/files/10019681/Windows_blisp_Flasher_Pinecilv2.zip)

Linux Blisp Flasher is attached (could also build from [open source code here ](https://github.com/pine64/blisp)
[Linux_Archive.tar.gz](https://github.com/Ralim/IronOS/files/10019722/Linux_Archive.tar.gz)

3. Any problems or issues with  Flasher, please go to Gamiee's repo: https://github.com/pine64/blisp
4.  Even though I don't have a linux PC ,  I did  write up instructions from different linux installs people did.  Instructions are in the "Pull Request"  section of the github  blisp  flasher linked (as they have not been merged with the blisp repo yet).
5. It is called blisp flasher because  the MCU in pinecil is the BL706.  (blisp = Bouffalo Labs ISP tool & library).

River-b, 2022-11-16:
Things needed for UART testing to get logs when V2   i2C starts corrupting screen:

1.  pinecil Break- out board from Pine store.   OR something like  this [PCB from Amazon](https://a.co/d/b1098d2) was suggested by @prokrypt  if you don't have pinecil board.
2.  A cable connector to go  from the pins of the break-out board in (1)  to the  PC.  Could [get this ](https://a.co/d/969qdvZ) that part clips onto the break-out board from (1).  Then used your own  USB cable on that to the PC. 
3. this part was recommended by @prokrypt  because  it has a real **FTDI  chip** in it which may have less issues with communication over cloned chips (especially for windows).

River-b, 2022-11-16:
Pinepony found this evidence that it is 2000000 from IronOS code. 
[image]

River-b, 2022-11-16:
@Ralim   

1. UART firmware beta  is  **not working**  for me, it's causing  V2 to reboot without stopping (kovol 140W charger)
2. version 2.20.A1510E 14-11-22  reboots fails on both EPR Apple & Kovol but different behavior
3. did you want me to just do Uart testing with PD65 or wait for you to fix one that has UART messages and can work on EPR chargers? unless you plan to release update that is Non-EPR one.
4. same exact thing happened when I tested this beta: "Testing slower I2C for PinecilV2 #2388" - it only worked for PD65 chargers so I stopped testing it. 
5.  Does this UART beta include the corrections  for the 0x00 ignore padding problem of Kovol and the Anker 717 problem?
6. Kovol 140w reboots continuously  like it did before padding fix.
7. Apple is _different_.  as you recall the Apple 140 originally worked with stock 2.18, but the PD debug menu never worked in 2.18. it just kept rebooting in PD debug without stablelizing at State 12.   In UART ver 2.20.A1510E  the PD debug  problem  re-appears (PD debug reboots and never stable) And an additional problem now. Sometimes when I plug it in screen is fine and I can heat.  every 3-4 times i plug it in the screen reboots within the first 5 seconds and it just keeps rebooting every 5 sec or so and I also see PD State messages appear every now and then when it's rebooting (even though I'm in normal screen). If I hit (+)  it sometimes heats up and stops rebooting and sometimes it reboots a few more times. If I keep plugging it in cable,  sometimes it's stable and I can Heat without more reboots, but I haven't tested it soldering to see if it would interrupt and reboot on me more during use.  this ver  A1510E    is less stable than 2.18 as now the Apple doesn't work even if I ignore PD Debug strangeness.

NovaViper, 2022-11-16:
**Reuploaded to remove sensitive information**

Can confirm that I'm also getting the same corruption issue on my Pinecil V2 using the beta 2.19 software. I noticed it particularly when I was soldering a SMD chip for a TV circuit board, would just mess up the numbers on the screen with little dots all over (not as bad as op but definitely would make reading the numbers impossible). Sadly I don't have any pictures of it but I noticed it happening if I rotated the iron around too much (which I tend to do because I'm trying to reach different parts of whatever I'm working on).

I just installed the newest 2.20.3A96FF1 update and it seems to have gone away (at least from me just rotating it over and over again about 30 times rapidly) to see if I could trigger it again. Will update if I manage to replicate it again

The PSU I used when I got the original bug was the RavPower PD Pioneer 65W 4-Port GaN Tech USB C Desktop Charger (only the iron was plugged into it at the time)

River-b, 2022-11-16:
not_daniel   suggesting that Uart firmware is failing on EPR chargers maybe because the firmware is waiting too long to send Chunk Messages.

River-b, 2022-11-16:
> I just installed the newest 2.20.3A96FF1 update and it seems to have gone away (at least from me just rotating it over and over again about 30 times rapidly) to see if I could trigger it again. Will update if I manage to replicate it again

Usually when people get the corruption, they simply flip it over gently and that causes the screen to re-draw and refresh - which gets it back to a normal screen.  flipping V2 over many times  should not cause corruption but the opposite since the acti of flipping causes the Accelerometer to refresh and draw the screen, since that is also the time it decides which angle to display the words if you have auto  orient screen on.

mikerocklewitz, 2022-11-16:
Linked firmware flashed on both my v2's  using powershell 🤣.  Had issues with getting the 2nd handshake on each one.  2nd one erased then failed. Couple tries to get both to flash - they heat up and hall sensor appears functional still.  Will report back if any issue, not doing any soldering today.

I could maybe try to give the uart thing a go. I have an official breakout board and several different USB-c breakout options as well.

River-b, 2022-11-17:
> 

I dont see link.  you tried to update using  Windows blisp flasher?  if it failed.  did you make sure to put the  .\  in front of Both the word  blisp and the  Pinecil file?    in windows Powershell, change into the Directory for the  flashing , make sure Pinecil...bin is in the folder with the blisp  executable  and Either hit tab after you type blisp word and then hit tab after typing Pinecil word,  or  just put the **".\ "**
Could you join to Discord live chat?  it's easier  to do troubleshooting it's  Easy to do takes no more than 1 min to flash this, but sometimes there is just 1 thing people are doing that is different that causes the failed flash.  in Windows doing the .\    everywhere exactly as  I have it in screenshot is important.

We also have a Linux flasher if you prefer.  
Join  Community Chat ,  Volunteers can probably Figure it out more quickly in there why flashing is not working for you. 
If  The First half of install  worked but not the 2nd part,  then I suspect:
1.  Pinecil.bin is not in same folder as  blisp.exe
2.  Pincil..bin  is  named incorrectly, use the exact name as you see in Windows Powershell screenshot above if you can just to make it easier.
3. the ".\"  was not used with the name. need the .\   in front of blisp and in front of Pinecilv2_EN.bin   refer to exact way command is written in the  Screenshot.
4. Could you  Drag Screen shot here of the Error and commands you typed into Powershell?  
if you come to Discord chat just should out you need help  installing BETA firmware for  V2.  and which OS you prefer.

mikerocklewitz, 2022-11-17:
I was able to get it to flash. I'm just stating it took multiple tries before I was able to get past the 2nd handshake. No errors except failed handshake.

I wasn't implying i linked anything. I was referring to your linked files above.

Didn't try Linux.  One of my windows machines was turned on when I flashed and it was convenient.

River-b, 2022-11-17:
> I was able to get it to flash. I'm just stating it took multiple tries before I was able to get past the 2nd handshake.

I see you got update done on 2nd try using the same command.  it could have been the cable or the port was flakey.
 
 when people run into this,  we sugest following:
-  USB-C to C cable if possible as that is better, or if not possible then switch to a different other Usb-A to C cables that might do data transfer better (some A-C  are just really bad at data transfer and no one notices as they mostly just use their cable for power only) . 
-  Avoid any usb hubs and do direct connect to PC ports or laptop.  On a PC,   the rear ports  can be better than front ports as front ones might be on an internal hub and wire and not direct connect to the motherboard.
- Another trick that has been done for very flakey ports is  hold the **(-)**   the whole time and don't let go of (-)  during the update process.  I've not seen it need to get to that, usually it is resolved  just  using different port or PC/laptop  or different cable. 
- in Windows Device manager, you may notice that BL706 MCU connects to Windows differently than the old  V1 did.  It connects as  a [Serial COM device under Ports (COM & LPT) ](https://lupyuen.github.io/articles/bl706)

mikerocklewitz, 2022-11-17:
I concur - very likely a flakey cable.

[2 more replies]

closed 2023-01-24 by River-b