Ralim/IronOS #1419: request to add the QC 2.0 (9V or 12V) maximum voltage selection function to the settings

StepanKoshman, 2022-09-27:
**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

Ralim, 2022-10-05:
Hi,
Can you please elaborate further? As of present QC2.0 is not supported and just QC3 is used.
What is the use case?
What device is this for?
Why does the existing QC3 support not work for you?

StepanKoshman, 2022-10-05:
[image]
[image]
[image]
[image]
[image]
[image]
Xiaomi powerbank 2i, pinecil v2. As you can see from the photo above, the bank does not support QC 3.0, but only QC 2.0.  up to 12 volts.  But my pinecil v2 only raises the voltage to 9 volts, while the power does not exceed 10 watts.  When I force the voltage to 12 volts with an external trigger, the power reaches 18 watts.

discip, 2023-09-23:
@StepanKoshman
Is this issue still present?
If not, please don't forget to close. 😊

StepanKoshman, 2023-09-23:
> @StepanKoshman Эта проблема все еще существует? Если нет, не забудьте закрыть. 😊

Yes, this is still relevant for me...