Ralim/IronOS #1420: Add TS101 support

rozwell, 2022-09-28:
Hopefully TS101 hardware will be easy to adapt and there won't be any bootloader issues.

Ralim, 2022-09-28:
Trying to source one. Miniware so far have not shared any details (firmware, schematics).
May or may not be worth my time to to depending on complexity, but will try.

River-b, 2022-09-28:
> Hopefully TS101 hardware will be easy to adapt and there won't be any bootloader issues.

see if the miniware community is willing  to  either mail Ralim a TS101  or easier if they can put some funds into the Sponsor link in top right of the screen.

Miniware could not be bothered to send Ralim one.  which they should since  his firmware  runs on most all their products and makes people actually like using the Miniware products more.  without IronOS the miniware firmware is not as fun or good.

I know Ralim does this for the love of  IronOS community.  however, the community doesn't send him love and coffee, (but many issues). 

See reference below, fact is there over  155,000  downloads of  open IronOS  _**just for TS100 alone**_   (yet  maybe ~24 Kofi/coffees donated to for all his volunteer work and most of those are from Pinecil community which is small % of downloads).  

If even 10% of people donated  0.25 cents it would be meaningful since Ralim has a day job to pay for rent/food.  I'd like to get that word out. And he tries to do everything he can  to only use features on Github  that don't trigger the Advertisements.

this is not pressure, just putting out some facts to help people know how much work goes into firmware, and how little the community is aware of  it, and ralim never asks,  the heart sponsor link is barely visible.   I also knows he buys a lot of equipment and chargers with his own money to test them and improve issues.

[Reference datasheet](https://docs.google.com/spreadsheets/d/e/2PACX-1vTVXQfzKYp0MsOMG7gf9u-V8gYyewQVI84YdE8i4e2ZBf9xSU--9BgFSd5JIUkVaGwjkzyVK0yPOmvt/pubhtml#)

rozwell, 2022-10-06:
@Ralim I'm actually surprised they IronOS isn't the default firmware for Miniware irons, since it's just fantastic compared to default. Were you ever in contact Miniware about it? Or them supporting the project in any way? The least they could do, is provide 1 unit...

@River-b good point. I really love IronOS (using on TS100) so I just donated 4 Koffis 😉 
It's good you pointed it out, cause I didn't actually notice that option it until now 😅 - dealing with all the daily crap can make people so dull...

rozwell, 2022-10-08:
@Ralim, @River-b Miniware responded to my YouTube comment:
> MINIWARE
> [9 hours ago](https://www.youtube.com/watch?v=e7saVBTAgEg&lc=UgweHzJqPc3SchHRdrJ4AaABAg.9gqHlzYzik09guPkiwPXbR)
> Already send to Ralim.

@Ralim Can you confirm? 😀

sorgelig, 2022-10-18:
I've just got TS101. That's great improvement over TS100. And it's really useful updates unlike Pinecil64 v2.0 (BT addition is dumbest feature ever). I like new larger screen and clean interface. PD support. New slot for tip not requiring to screw it every time you change the tip. The only problem with original FW is bad work with non-original tips with noisy temperature sensors. Otherwise FW is great.

Ralim, 2022-10-20:
Alrighty my TS101 has arrived.

First a word of warning, don't rush to open it, (1) I broke plastic clips for the back cover (there is no screw). (2) You have to remove the front glass to get the screen out (if you just pull up on the PCB, you will tear off the screen). As such as I have not done a full teardown yet.

First impression notes:

- Same STM32F103CU as the TS80P
- Screen is a tiny bit larger, looks to be same resolution as others
- Their bootloader is 'uge at 32k as they have packed storage into its region instead.
- - This means to make IronOS work either need to replace bootloader, or get creative at moving as much as we can into that area somehow.
- Features are basically same as the old PinecilV1 though less powerful CPU as still on the F1.
- Tip retention with the spring loaded ball, though nothing I would write home about, tips spin when using non-round ones.

Biggest issue is that at the moment only have ~32k of flash to work with, which is _small_.
So will need to ponder this a bit about how best to handle this; will add some complexity to bring up.

Have dumped the firmware off mine, and started poking around, need to figure out pinmaps and do all the usual bringup mess.
Once I confirm OLED+Buttons I will try and get my DFU bootloader working so we have a backup plan.

sandmanRO, 2022-10-22:
I personally don't like the new design. It's somehow bulky / non-fluid as opposed to the old TS100 design, more likely dictated by the parts size fitted in rather than the other way around. It's very unlikely I would buy this product. But then again, I already have four TS100 units so I would not be needing more portable irons any time soon.
P.S. Please don't shoot...it's just a personal opinion ;-)

SubZer01, 2022-10-22:
> I personally don't like the new design. It's somehow bulky / non-fluid as opposed to the old TS100 design, more likely dictated by the parts size fitted in rather than the other way around. It's [...]

and who is interested in your opinion?
it is about the firmware Support from the TS101.
furthermore, the new design is made softer rather than vice versa.

discip, 2022-10-22:
@SubZer01
>and who is interested in your opinion?

I'm pretty sure you wouldn't be "talking" like that if you knew that this man contributed a significant part of the code that makes your device heat so precisely.

Look here: https://github.com/Ralim/IronOS/issues/1038

> P.S. Please don't shoot...it's just a personal opinion ;-)

Furthermore he literally asked for not being shoot. 😞

Ralim, 2022-10-23:
@SubZer01 Please watch your tone, you are coming across super rude. I would rather not block anyone, but not against it.

Making some gradual progress on this. Dont have a huge amount of time at the moment, but have started poking a flash dump in Ghidra.

My rough plan is to:
- Figure out the pinout they are using
- - Mildly concerned as they are bit-banging everything like before so wondering how many signals are on hardware pins _sigh_
- Get all the basic features working
- Trace out USB-PD stuff and see how thats done
- Figure out how to use the screen (it is 128x32 so higher resolution)
- - I knew this day was inevitable but have to think about best way to handle it. Worst case will just scale possibly for now (unsure; would love ideas)
- Wonder what to do about bootloaders, their one is super wasteful. They are clearly storing the virtual disk in the bootloader area. Which means we can store settings there but not the bootup logo.

@sandmanRO 
I'm fairly similar tbh. Also dont like that tip contacts are soldered to the board; and assembly of screen under glued glass makes it muuch harder for repair. (Thus why I'm disassembling code rather than multimeter probing it).

BF3000, 2022-10-24:
> I've just got TS101. That's great improvement over TS100. And it's really useful updates unlike Pinecil64 v2.0 (BT addition is dumbest feature ever). I like new larger screen and clean interface. PD [...]

Actually  Pinecil v2 have much more improvements than TS101. BT support is dumb, but 28v PD support & short tips (3 sec to heat up from 20C to 300C) is not.

PUF52, 2022-11-07:
Just mini ware posted the firmware 1.09

PUF52, 2022-11-07:
http://www.minidso.com/forum.php?mod=viewthread&tid=4937
http://www.minidso.com/forum.php?mod=viewthread&tid=4937
firmware is here

PUF52, 2022-11-07:
really called it 1.10

River-b, 2022-11-10:
> Actually Pinecil v2 have much more improvements than TS101. BT support is dumb, but 28v PD support & short tips (3 sec to heat up from 20C to 300C) is not.

- the BL706 chip came with BT support as a side bonus and it was low cost and readily available chip unlike other chips  that are in short supply in the world.  This is one main reason the BL706 MCU was chosen for Pinecil V2. 
- Pinecil also serves  the Pine64 FOSS community as a RISC-V  development tool. It is _not just_  a soldering iron like other soldering irons. 
- Some people buy Pinecil and do not have any tip installed. They just have a USB-C  break-out board connected to the back side which helps with development and working with USB-C  libraries for example. They also have a completely different development OS running on it that is not IronOS.
- People often do not realize  that Pinecil is not meant to be just a soldering iron and then criticise these other features it has based on on idea that a soldering irons can only function as a soldering iron. If you think of it _first_  as a RISC-V  development hardware for the open source community  and second as a soldering iron  then it will make more sense why it has the  extra features for open community  to explore and have fun with. 
- See[ Pinecil Wiki here ](https://wiki.pine64.org/wiki/Pinecil) which explains how it serves as a RISC-V development tool 
- Other irons only care if they can solder.  The Pinecil is held to an additional standard - how well does it meet the needs of the open source community purposes to be a RISC-V  dev tool and could features be added that exploit some community requests for fun and science.

Ralim, 2022-11-12:
I am planning to continue work on this, motivation is lower as I'm going to have to rebuild a lot of the firmware to support this iron, so initial firmware will probably not be great.

So far all of the hardware pinouts I've found were done by an idiot, and so not going to be able to use hardware peripherals for things.
About halfway through making a pi out for the device.

Question for the community: the current bootloader wastes a bunch of space on the device. Is it worth pushing people to replace it or not?

sorgelig, 2022-11-12:
If flashing bootloader will go wrong, will it brick the device? If so, then it's better to avoid flashing bootloader unless it's absolutely required.

Ralim, 2022-11-12:
Yeah it can definitely go wrong. Its a problem that can be decided later on I guess. Just concerned since bringing in a new device also implies full feature completeness (langues etc) which is currently a fair bit chunky.

sorgelig, 2022-11-12:
will it require to flash original bootloader if for some reason an original firmware will be needed?

BF3000, 2022-11-14:
As long as there will be a way to restore it to original, I'm fine with that.
Flashing to IronOS is usually one way road.
And BIG WARNING will be nice.

Ralim, 2022-12-01:
At this point I do not know if I will bother; it would most likely be a sub-par experience due to their hardware design decisions.
Most likely answer is that I will add some basic support; but (a) it will take  time and (b) its a lower priority as I'm not paid to waste 100's of hours working around their design decisions and decompiling their firmware.

L1cardo, 2022-12-10:
They released v2 of their firmware which can go to 240w with PD3.1, that is crazy!!

River-b, 2022-12-10:
> They released v2 of their firmware which can go to 240w with PD3.1, that is crazy!!
 could you give me a link to where they say it will do  240W? 

 I think they just enabled  PD3.1  like we had in IronOS already for >4 months for Pinecil V2 (also supports EPR PD3.1).   I've been running Pinecil V2 for 4 months with IRonOS  at 28V, but with special Pine Short tip resistence the highest max would be  126W.

Miniware TS101  says  up to 28V  90W now  on their website (the 28V is  PD3.1 current chargers that are available).   There are not even any  USB-C  chargers out yet that do 240w, the highest USB-C chargers that are EPR PD3.1  on the market in 2022  can only do  28V 140W. 
 
Miniware  website says  only 90W, 28V  on USB-C. 
[image]

Miniware uses long 7.8-8.3 ohm tips,  so at  28V EPR charger, with some manufacturer tolerance that would be about the 90W they say on their new listing.   But not 240W (since that doesn't exist yet).   A Special  USB-C cable is required.  the cable  is  rated at 240W (future proof),  but the Chargers on the market are max  28V-140w right now.   I know because I've been using it in the  Pinecil V2 iron for  4 months now  a  140W-28V-5Amp charger with  special  USB-c 240w-48v  cable .   Cable  is designed to last  into new  new standards for EPR that will be coming  in the next few years  ,but the chargers are only  capable of the  lower end of the PD3.1

Ralim, 2022-12-10:
I had to poke them a bunch and give them a hand to get them to add EPR support to the unit.
I'm glad that they did it; but no idea why they didnt at launch.

sorgelig, 2022-12-12:
I more wonder why 28V can be used on USB-C input but not on DC input. Is there are difference?

PauliusMorku, 2022-12-15:
> I more wonder why 28V can be used on USB-C input but not on DC input. Is there are difference?

I was wondering the same thing, it is so annoying that we can't use a fully charged 6s battery 😑 or maybe we can? Is there anything on the DC path that could not handle 25.2 volts?

L1cardo, 2022-12-15:
@Ralim Just got my new ts101 iron, really like the tip slot. Looking forward to testing your new FW, even I should flash a bootloader.
Let me know if you need any of ny help.

L1cardo, 2022-12-15:
> > I more wonder why 28V can be used on USB-C input but not on DC input. Is there are difference?
> 
> I was wondering the same thing, it is so annoying that we can't use a fully charged 6s battery 😑 or maybe we can? Is there anything on the DC path that could not handle 25.2 volts?

I asked the miniware, they say it can handle 28v

River-b, 2022-12-15:
> I asked the miniware, they say it can handle 28v

they said for  the DC barrel you can use  up to 28V  (like from a Bench DC power supply)?   some DC barrel bricks can spike high and then it's not a good idea  to  use more  than 24V  on DC barrel side. 
or do they mean for USB-C only on EPR?

River-b, 2022-12-15:
> Let me know if you need any of ny help.

see this  https://github.com/Ralim/IronOS/issues/1420#issuecomment-1334529802

PauliusMorku, 2022-12-18:
Do all of TS101 come with STM32 chip? Or some of them use clones? I had TS100 with a clone ship and that prevented me to upload IronOS.

MaximusPanin, 2022-12-27:
> Do all of TS101 come with STM32 chip? Or some of them use clones? I had TS100 with a clone ship and that prevented me to upload IronOS.

Chip clones on the TS100 have appeared recently. This is a strange decision if it is dictated by economy. If the situation with TS101 is similar, they will lose part of the audience. Reputation is hard to build and easy to lose.

Ralim, 2022-12-27:
I can almost guarantee that they will use clone chips at some point, with all its issues, since its designed using the same STM chip.
The only question is _when_ not if.

L1cardo, 2022-12-28:
> I can almost guarantee that they will use clone chips at some point, with all its issues, since its designed using the same STM chip. The only question is _when_ not if.

Is there a way to know if it is a clone chip without open the iron case?

Ralim, 2022-12-28:
Only other hint is usually "is their DFU more dodgy than usual"; generally opening case is easiest to go by until we know other means. (For TS100 we know all made after earlier this year are clones, and DFU versions above a certain number are all clones).

MaximusPanin, 2022-12-29:
GD32F107 https://youtu.be/Lo77r6JwimU?t=1149
Compare board https://youtu.be/Lo77r6JwimU?t=850

GD32F103TBU6 _(TS101)_ -  **108** MHz
STM32F103T8U6 _(TS100)_ - **72** MHz

PUF52, 2022-12-30:
Greetings to all lovers of soldering irons ts. The question is addressed to those who have ts101.
The miniware company promised everyone that you can easily replace the regular logo with your own. How to do this is written in the user manual. I tried, but it didn't work out right. My logo appeared, but the miniware logo did not disappear either. That is, at first, when loading, I see the miniware logo, and then my own. Correspondence with the seller did not help. Here is the last answer:
The factory said this is a normal phenomenon, there is no way to solve it for the time being. Will not affect the normal use of the soldering iron.
The conclusion follows from the answer, the developers did not implement this feature in the firmware, but promised ..
Which of you managed to make the correct logo replacement and how?

MaximusPanin, 2022-12-30:
**PUF52**, firmware 2.0 ?

PUF52, 2022-12-31:
Yes, now 2.0.

MaximusPanin, 2022-12-31:
Send me your logo, I'll try after the new year. 
Happy New Year!

discip, 2022-12-31:
As in the previous miniware irons, part of the logo is in the bootloader.
So the bootloader has to be changed or rewritten.
@Ralim already did that [IronOS-dfu](https://github.com/Ralim/IronOS-dfu).
But I doubt this is going to work with devices with GD chips.

PUF52, 2023-01-01:
[image]

PUF52, 2023-01-01:
> Пришлите мне свой логотип, попробую после нового года. С Новым Годом!

А куда прислать? Дайте свою почту, вот моя: pershin 52@list.ru

L1cardo, 2023-01-26:
Sorry to bother! Any progress in this?

Ralim, 2023-01-26:
TL;DR: No, and I have no timeline.

Miniware have gotten less and less helpful as time has gone on.
No schematics or pinmaps or any help anymore really.

I did an initial investigation to figure out where I2C was, and it's been placed on pins that can't do hardware I2C.

This is super demoralising, as it means that a fair bit more work will need to be done for this to implement it all in software and make sure things work.

Can it be done: totally, but it will take a lot more work. And their dumb as a rock bootloader makes it even worse.

I did scratch out some basic support using the replacement bootloader but it would require flashing using an SWD adaptor to use, so it's not a good way forward.

Next steps is that all the signals on the board would need to be traced, and then all the components measured to figure out scaling values.

Miniware are not making this easier, I have asked for schematics and been told NO, I've asked for a basic pinmap and been told NO, I've for any help they can given and been told "nothing is available".
So only real option left would be a lot of hours either decompiling their firmware to tracing with a multimeter to pin it all out.

Which is a ton of work, and not very motivating or interesting. Couple that with (relatively) low demand and impact compared to other things I could spend my time on.

So this is in the bucket for "future" when other projects are done.

I would happily take a PR, or even just a schematic and then I would look at it again. Or even just a roughly documented pinout.

If someone in the community wants to reverse engineer their firmware or the PCB back to a pinmap I would be happy to pick this up again sooner. But until then, burning 100's of hours on this feels like a poor way to spend my time when I could achieve a lot more in other places for the same time.

L1cardo, 2023-01-26:
I asked a staff of MiniWare, and he said they offered some to you(don't know if it is true). I also asked him if he can offer more, he said he will look into it.
Following are the screenshots of the chat.

[image]
[image]
[image]

Ralim, 2023-01-26:
They sent me a single TS101 unit (with NOT FOR SALE lasered on it 🤣 ) but refused any supporting documentation when I asked.

Wuthoqquan, 2023-02-07:
Hi all! Hope this is not going to be too much off topic!
I'd like to have a better understand on how to use the 90W support made available to the TS101 with the latest FW2.0 update.

The question is if the 90W can be unlocked on the TS101 only with a charger that complies with PD3.1 standard and can output 28V, or if it possible to supply the required power also with a PD charger that has for example an output of 20V at 5A (100W).

It is really not clear to my if the TS101 with FW2.0 can reach the advertised 90W power only with PD3.1 chargers (for example: 28V at 5A = 140W), or if it is also perfectly fine to use a PD charger with a maximum rating of 20V at 5A = 100W.

It is really not clear to me how the power management has been implemented in FW2.0 and if the 90W power can only be achieved with the latest PD3.1 chargers.

Thanks a lot!
Marco

River-b, 2023-02-07:
> The question is if the 90W can be unlocked on the TS101 only with a charger that complies with PD3.1 standard and can output 28V, or if it possible to supply the required power also with a PD charger that has for example an output of 20V at 5A (100W).

You will have to ask  **Miniware** how to do this  as you are asking about  Miniware's own company firmware  and not  Ralim's IronOS firmware.   There is no Ralim's IronOS firmware for the TS101 and this is not miniware support page.   This is open source Ralim's IronOS  which  is able to be written  to  some Miniware  devices   but the  new ts101  only  has  miniware's  official firmware which  you have to talk to Miniware for support.

This ticket should probably be closed  as  Ralim  indicated above  he is probably not going to put more effort into  trying to get  IronOS to work on  TS101 because Miniware has made it too hard to do so.

Ralim said above  in 2 places, he is probably not going to work on firmware for TS101  because Miniware is not cooperative enough and made a mess of things such that reverse engineering it is going to take too much time.
see here  https://github.com/Ralim/IronOS/issues/1420#issuecomment-1334529802
and  here https://github.com/Ralim/IronOS/issues/1420#issuecomment-1404510840

River-b, 2023-02-07:
> It is really not clear to me how the power management has been implemented in FW2.0 and if the 90W power can only be achieved with the latest PD3.1 chargers.

it's not clear to anyone how Miniware implemented it since the source code for the TS101 is not open as far as I know. See if  Miniware will release the code for the TS101 firmware and email their support to find out.  
Since it's not open, no way to know and only they can answer this.

River-b, 2023-02-07:
if you really want to use a  28V 140W PD3.1 charger,  I recommend  getting an Iron that supports it  using an open  firmware like Ralim's IronOS.  
Pinecil V2 version (with green thub grip)  is the only Iron I know of that both  supports PD3.1 and has been  tested by many people with PD3.1 chargers at this point  and also **can use open source firmware** - Ralim's IronOS firmware as opposed to  a closed proprietary firmware. 
Being able to use open  source firmware  is an important  feature to look for in Iron's if you want to use  Github IronOS on them, not all irons can do this or the manufacturer makes it too hard to bother trying. 
Pinecil V2  is confirmed by many to work with 28V although theoretically the maximum you could get after subtracting out the resistance of the Tip  is  about 124 Watts on Pinecil V2.  and  in practice using official 240W  PD3.1 cable with a PD3.1 140w 28V charger,  I've  tested up to 110Watts  putting Pinecil V2 into extreme  testing.    
Besides the tip  having  resistance which you have to  use  ohm calculator to account for,  also consider there are other things causing resistance too and subtract for that (pcb components, cable itself,  more resistance as pcb gets hotter).  For Miniware,  their tips are about  8.0 ohms but you should check that out yourself using  multimeter, then add  a little more resistance in the calculation for other  little things, like the usb-c external tester and such. 
https://ohmslawcalculator.com/ohms-law-calculator

For Pinecil V2, we subtract  6.2ohms  for ohms calculator because the new V2  has  specially designed Pine64 shorter  tips which are more ergonomic but also have lower resistance.   However the V2 can also use the normal longer ts100 tips as well, in which you use the 8.0 ohms  in the  Ohm calculator.

 whereas  the Miniware  can not use  the newly designed Pine64 short tips and therefore,  just  use  8.0 normal length tips in the calculations.  
 For miniware then the theoretical maximum is  98W but you might not see more than 85W with resistance of other things added in,  but you have to test this for yourself. 
 
[image]

The Pinecil V2 on the other hand  was designed with  lower resistance shorter  tips of 6.2-6.5 ohms in mind (although it can also use  the  older tips of 8.0 ohms as well if you unplug between tip swapping). 
and as you can see in  ohm calculator,  Pinecil V2  much better utilizes  the PD3.1 charger as it has a theoretical  max  of   126W  using  a real EPR PD 3.1  charger  of 28V  and also  using a real  PD 3.1 240W  USB-C cable.   The cable is important,  if people use  old  normal  usb-C cable  they will not get more than 20V  plugged into PD3.1 charger.  that cable has a special emark chip in it which opens  negotiation for 28V to an IRon. 
[image]

Wuthoqquan, 2023-02-07:
@River-b Thank you so much! I was not expecting such a comprehensive reply to my question about the difference between 100W supplied by a charger capable of delivering 28V and 100W supplied by a charger capable of delivering only 20V but at a higher current.
I do not own a certified PD3.1 charger, but a 100W USB-C charger working at 20V - 5A, hence why I was curious to understand if I could possibly still take advantage of the 90W offered by the TS101 with FW2.0.
I do apologize if I asked the question here, as recognized in my earlier post, where I admitted my query could be off topic.
But I was asking just from a general technical point of view, since I really do not know if 100W can be different, depending on the different combination of originating voltages and currents.
In my specific case, I’ve purchased a 100W PD charger that can supply a maximum of 20V at 5A and I’m now questioning whether I would be better served by a certified PD3.1 charger capable of delivering up to 28V.
I appreciated your insights about the need of proper cables to anyway be able to fully utilize the higher 28V attainable by PD 3.1. Thanks for that as well.

River-b, 2023-02-07:
> but a 100W USB-C charger working at 20V - 5A,
@Wuthoqquan 
unfortunately   20V 5amp  100W charger will not make difference.  if you look at the Ohm Calculator math you will see this.
Enter simply  20V  and  8ohm  for the Tip.  and you  get  your answer on maximum watts.
then "try"  to clear  ohm calculator  ,  enter  20V,   5Amps,  8 ohm resistance.   you will see  the science auto corrects and you have exact  same  Power watts.

In this situation, you need to  increase the Voltage as  increase of  amps  rating of your charger doesn't really  do anything significant  as you can see for yourself if you play with  Ohm calculator online.

you might try  20V 5amp  100w charger with a 100W  PD usb-c cable and  get a tiny bit more  than using USB-C   normal  PD65W cable,   but I'm  not sure  it will even show on  the miniware device.
I have tested this on Pinecil devices myself  and  no matter if I use    different cable  PD100w cable or PD60 cable with  either a PD65W 20V 3.25amp charger or a PD100W 20V 5amp charger,   the heating time and performance and  also watts  used  is approxiamately the same in real life.

The only true significant change I could see and feel in performance was upgrading to full 28V  charger  PD3.1 and combining this  with  a PD240W  EPR PD3.1 cable  ( many  work  you dont need  certified ones).  I recently got some  $5  cables from aliexpress of  PD3.1  240w kind and they work  fine  just like my  $15 -$50  USB-C  PD3.1 cables.   for the charger,  if you want a budget one that has proven to be  pretty good and passed  test,   the Rocoren  140w 28V one  has been tested by many and is about  $44  on Aliexpress (use store coupons)  and also  is on  Amazon too.

Wuthoqquan, 2023-02-08:
@River-b WOW…. Thank you! Hats off! Your reply was again super informative! Now all makes sense to me. The way you have explained how the voltage impacts on the overall supplied power has clarified why PD 3.1 should be the preferred option in such use cases, given all the other variables that reduce the actual power output.
Guess I will send back the 100W charger and replace it with a proper PD 3.1 140W one.
Many thanks again!

River-b, 2023-02-08:
@Wuthoqquan 
PD3.1 is also more future proof.  and the USB-C  240W cables also future proof.
All  100W cables were officially deprecated by the USB-IF  organization in December 2021  because they want to simplify cables into just 2 kinds. One category is  SPR ( standard power)  PD60 cables - cheaper because  no emark smart chip inside.   and the 2nd category is  EPR  PD3.1 cable (extended power  rating)  which is  the 240W cables that come in different data speed.  for  Firmware  even the lowest 480 mb/s  is fast enough data speed so we only care about the 240W rating of the cable.
Be aware there are some fake  240W cables on the market  that I purchased and others as well, and tested.
I have both certified and non-certified.
following cheap cables worked at  28V  on usb-tester for me if you want to save some money.

- https://www.aliexpress.us/item/3256804889230539.html
- https://www.aliexpress.us/item/3256804901287548.html

they are not certified by USB-IF but they work and show me correct 28V on USB-C tester and on Pinecil Iron Detailed screen view. 

yes 100W cables are still sold. I dont see reason to buy them unless you get it dirt cheap same price as SPR PD60 cable.  USB-IF said manufacturers who have certification on 100W cables can continue making them, but  when they run out of  100W emark chips, they too will probably switch to selling  EPR 240W emark cables.
reference:  https://compliance.usb.org/index.asp?UpdateFile=Cables%20and%20Connectors&Format=Standard#:~:text=Effective%20Immediately%20100%20Watt%20USB,on%20December%2022nd%2C%202021.

rozwell, 2023-02-08:
> This ticket @rozwell - maybe consider closing as Ralim indicated above he is probably not going to put more effort into trying to get IronOS to work on TS101 because Miniware has made it too hard to do so.

Looks like TS101 is a no-go :unamused: 
And this thread went off-topic so... I agree, closing.

violavon, 2023-02-17:
@rozwell please, re-open.

piit79, 2023-02-17:
@violavon Why...? It was explained adding support would be complicated, and there is no plan to do it. You can blame Miniware for that for being ignorant.

violavon, 2023-02-17:
@piit79 As said, support may be done in the future, although it isn't a high priority.

River-b, 2023-02-17:
and it will be basic because... read all comments from ralim in this ticket, miniware has laid it out in a way that it's too much work to reverse engineer and they haven't shared with Ralim the schematics or anything to make it easier.

[66 more replies]

closed 2023-07-23 by Ralim