Ralim/IronOS #1421: Which dev board is for the pinecilV1?

Laar3, 2022-09-29:
Gday, im writing a flashing tool for the pinecil and i have been told by river that you used one of these https://www.seeedstudio.com/Sipeed-Longan-Nano-RISC-V-GD32VF103CBT6-Development-Board-p-4205.html but i was wondering which version of that board to get as they have both a V1 and V1.1 or if either would work?

Thanks.

River-b, 2022-09-29:
I found this available to buy in stock on Aliexpress. would this be good to test  simulate V1 for someone who doesn't own a V1 ?
**https://www.aliexpress.us/item/3256803280749848.html**

I think the seed studio is out of stock and they prob charge shipping too.

Ralim, 2022-10-05:
I used what I assume was the V1 as it was the only version at the time.

Basically any will work for what you want, you just need to be able to short boot to enter into bootloader + a working USB port.
For just testing you can look at a working flash being one if it disconnects from the usb bus and doesnt enumerate (i.e the code started running; and it didnt fall back into bootloader). If you corrupt the flash it should drop back into the bootloader.

Also you can validate by doing a write + read test and comparing; should get same out as went in.
The Sipeed Longan  is what I started on back when they were < $5 shipped.