Ralim/IronOS #1422: TS80P not heating up, thermal runaway?

kiararey, 2022-09-30:
I was getting an error on my iron saying "low-vol" and I updated to version 2.19 and now the iron heats up extremely slowly, not even reaching 300C, before saying "thermal runaway" and cooling down.

How can I fix this?

Ralim, 2022-09-30:
"Low-Vol" is an error from the Miniware software. This firmware doesnt have that limit in that it will let you heat up on 5V, but if you are doing something that tries to pull tooo much heat you will trip thermal runaway (which is there to detect hardware failures).

Most likely you are not running the device at 9/12V but instead at 5V which is not really supported.

kiararey, 2022-09-30:
Yeah low vol was on the previous firmware before I updated. How can I check at what voltage I am running the device?

kiararey, 2022-09-30:
Ok I figured out how to check Vin and indeed it is 5V. Not sure how to fix this though as multiple cables and multiple plugs, including all the ones I used in the past, are showing 5V :(

Ralim, 2022-09-30:
I would start by providing the information in the template that was not filled out:
What charger are you using + what cable type are you using with it?

Usually the main suspect is using a QuickCharge adaptor rather than the far superior USB-PD.

kiararey, 2022-09-30:
Sure. Like I said before, I’ve tried several cables and several doctors including:
- The USB-C cable that came with my Nintendo switch (used to work before) 
- The USB-C cable that came with my iron (used to work before) 
- The USB-A to C cable that came with my playdate (used to work before) 
- The USB-C cable that came with my old Macbook (used to work before) 
- A USB-A to C cable I found in my junk drawer 

As for the converters or power source, I’ve tried:
- an Apple USB-A to USB-C adapter (used to work before) 
- An Apple USB-C to USB-C adapter (used to work before) 
- The wall plugs (used to work before) 
- A surge protector (used to work before) 
- Straight from my computer

Ralim, 2022-09-30:
When you installed IronOS did you get a warning about the PD chip?

I would be wondering if you have a hardware issue.

An Apple C to C cable with the Apple USB-C power adaptor is known to work totally fine. It should at the least give you 9V.
Also the Nintendo Switch USB-C charger is known to work as well.

I strongly suggest not using any USB-A connections anywhere; as that will prevent you being able to use USB-PD.

If none of the USB-PD sources are working; I would try a forced settings reset in the menu (menu -> Advanced).
If that still doesnt work; its quite possibly a hardware issue to be suspect of.

kiararey, 2022-09-30:
I did a factory reset and the problem is still occurring. I'll try to check the internals of the iron.

River-b, 2022-09-30:
@kiararey   a lot of low cost PD USB-C chargers online  that would give more power and better performance than trying to get the minimal power to  the iron ($10-$20).  
if it's PD usb-C charger  it negotiates to lower power or higher power for whatever the iron asks for so you could get a higher PD45w charger too.  just saw ones on amazon and aliexpress for $10 (search for TS80P usb-c charger, many types work). 
 using a C<-->C cable would be better as it will work with PD (Power Delivery on usb-C chargers)  which smartly does PD negotiation with the Iron. 

Straight from your laptop/PC is most likely just low 5V port and too low to do anything  more than just firmware updates. 
most phone chargers are also too low at only 5V.  

Also recommend contacting Miniware or the seller directly if  there is an issue with the hardware not working with their [published minimum voltage chargers](https://www.miniware.com.cn/product/ts80p-soldering-iron-main/) listed on the Miniware website. 
QC and PD are different,  PD  is using a usb-C to usb-C  cable end to end (no wider usb-A port at all).

discip, 2023-02-15:
@kiararey
Is your issue solved by now?
If so please close. 😊

thanks

kiararey, 2023-02-16:
I had to reset to factory OS but it worked then with a new USB-C PD charger. I’ll go ahead and close the issue.

closed 2023-02-16 by kiararey