Ralim/IronOS #1424: Pinecil debug menu reports incorrect power mode

wolfpackmars2, 2022-10-03:
**Describe the bug**
When connected to PD power source, iron reports "PWR   QC" in debug menu. Everything else works as expected. The i

**To Reproduce**
Steps to reproduce the behavior:
1. Start iron while holding + to enter PD Debug
2. Notice that PD Debug reports iron has negotiated PD 20v successfully
3. Exit PD Debug and go to normal debug menu (Long press B- from main screen)
4. Click on A+ button until "PWR    QC" is displayed

**Expected behavior**
Debug menu should display "PWR    PD No VBus" or just "PD No VBus"

**Details of your device:**
 - Device: Pinecil V2
 - Release: V2.18.A1A569A (17-07-22) and V2.20.0DDBE80 (25-09-22)
 - Power adapter being used: Anker 737

**Additional context**
Using new Anker 737 power bank [https://www.amazon.com/dp/B09VPHVT2Z]
Using new Anker 3.3' USB-C 100w cable [https://www.amazon.com/dp/B093GGVB89]
New Pinecil V2 [https://www.amazon.com/dp/B096X6SG13]

It is noted that iron performs as expected. It reports 20v, it operates at PD spec, the PD Debug menu shows the correct information, the iron quickly negotiates the PD mode. The ONLY problem is the regulard debug menu reports "PWR   QC".

Have a different USB cable [https://www.amazon.com/gp/product/B09PB8DJ28] ordered, will update if it performs differently.

River-b, 2022-10-03:
Please attach the video you made in here. I think you can just drag and drop into github videos and photos.

JoeyNinja, 2022-10-03:
I have the same 737 140W battery. Using [Inter9ers EPR cable](https://www.amazon.com/Transfer-Compatible-Thunderbolt-DisplayPort-Extended/dp/B09ZY2LSR1) I get the normal "No VBus" messages in both the (-) Debug menu as well as the (+) PD Debug menu.

River-b, 2022-10-03:
We have 2 different behavior  for  (-) Debug menu with Anker 737.  
- Using a certain cable,  Rojo got "DC"  in debug menu,  and Wolf got  "QC"  in debug menu. 
- changing to a better cable, Rojo eliminated his issue with inconsistent messages.

River-b, 2023-02-23:
@wolfpackmars2  did you try the different cable and did it fix the issue? if yes please close the ticket.