Ralim/IronOS #1425: Display glitches on DC power

JaxTheWolf, 2022-10-05:
**Describe the bug**

When the iron (Pinecil V2) is left idle for a couple minutes, the display gets jumbled. Rotating the display resets it. 

**To Reproduce**
Steps to reproduce the behavior:
1. Power your Pinecil 
2. (Probably optional) heat it up to working temp 
3. Let it idle
4. After some time, the screen should glitch out 

**Expected behavior**

The screen should remain stable and without glitching. 

**Details of your device:**
 
- Device: Pinecil v2
- Release: 2.20.something (cannot access my iron right now, I'll clarify later.)
- Power Adapter: generic 19V laptop power supply, ungrounded

**Additional context**

I have installed the hall sensor in my unit, though I doubt that's an issue.

Ralim, 2022-10-05:
Duplicate of #1417

closed 2022-10-05 by Ralim

JaxTheWolf, 2022-10-05:
@Ralim My bad, should've checked for similar issues. Though, my screen doesn't get completely jumbled up, sometimes there's remains of UI elements shifted elsewhere