Ralim/IronOS #1428: Pinecil v2 hall sensor menu setting

mikerocklewitz, 2022-10-19:
In the past ive added a hall sensor to my v1's.  But i dont see a menu setting for a hall sensor in the v2.  Is this missing or just not enabled yet?  Or is it a smart menu item hidden til the sensor is added? I see hall sensor in the debug menu.....

More insight please. Thank you! Also capable of doing testing on v2 for hall sensor if needed.

Thanks!

Ralim, 2022-10-19:
Hia,

The menu option should be hidden unless a sensor is fitted.

It is working on my one device here I have added one to. But small sample size, so interested to hear if you have any dramas.

mikerocklewitz, 2022-10-19:
> Hia,
> 
> The menu option should be hidden unless a sensor is fitted.
> 
> It is working on my one device here I have added one to. But small sample size, so interested to hear if you have any dramas.

Will report back tomorrow.....maybe tonight if i have the energy after work. I have an extra hall sensor in stock.

mikerocklewitz, 2022-10-20:
Hall sensor added to pinecil v2.  Works as its supposed to and menu appears in sleep settings.

Excellent work.  Looking forward to more to come.

Really wish they would use less sticky adhesive to mount the oled.  Impossible to do any modifications near the mcu without risking destruction.

closed 2022-10-20 by mikerocklewitz