Ralim/IronOS #1448: Calibration method is too confusing even for EE experts

River-b, 2022-10-25:
**Is your feature request related to a problem? Please describe.**
SDG who is a youtube and EEVblog  reviewer  posted a terrible video on Pinecil V2 today replete with errors and misunderstandings on how the V2 works.  Here is  an  electronics hobby expert with a wide  following,   where  Calibration screen is too confusing which leads him to think  and tell the world it is broken either the firmware or the Pinecil.

[Video here ](https://www.youtube.com/watch?v=-TosWhPsru4)

Please watch the  whole video, but based on several incorrect things he tells people, that calibration software doesn't work and other things about the tips, many people in comments are reacting  and saying how they are disappointed with V2. This is  a matter of several misunderstandings in how to use V2 and errors he makes because he does not understand how the menu works or how calibration works.  I'm also not sure he knows that  he needs to unplug the pinecil between swapping tips as the Resistance changes significantly between short and long tips 

**Describe the solution you'd like**
Some of these things below I know you can not do in github but could pass along to Pine64 people.

1. Some of the mistakes could be avoided by  adding some information to the Pine Store webpage.  Not everyone knows about Pinecil Wiki.  I would like to see  a link to Pinecil Wiki  on the Pine Store webpage.  Would also like  Pine Store to tell people to unplug Pinecil when swapping tips of different lengths.   It needs to be noted that many people will never know to go to Pinecil Wiki or Github and just use the pinecil as they get it with only the information they see on the website when they buy it.

2. Pinecil V2  mini user manual is so small it is not readable at all.  on front page should have huge letters with link to Pinecil Wiki, and the Github/IronOS,  and to the PDF version of the User manual.   It comes in such a tiny font that people will just put it aside, but if it had those 3 links in Large readable font on the front, people would have a chance to go to real instructions pages.  FYI, it already includes  webpage linkg to Github-IronOS  but it is so tiny, you can not see it or any of the manual without a good magnefying glass or taking a macro photo. it could be so much better for end-users than it is now.

3. when performing a calibration,  when the Offset number appears, make it more clear you are done. In SDG's video example  it could be  
`833   offset.  Done.  `  or `833  offset. Complete.`
 that would be more clear and avoid the  huge misunderstanding that he showed to whole world today. 

4.  SDG is not the only person who does not understand how calibration works,  it is unclear to everyday people when is calibration Done or Complete?

River-b, 2022-10-26:
5. It has been pointed out that the currect directions for calibration area little  confusing. Even if you knew about Github and came here to read instructions.  I know how to calibrate because I have direct communication with Ralim, but after reading this _below_ I was more confused than before reading it.  I have no idea what this is trying to say, it needs some clarity.   try to read the below it as someone brand new to Pinecil and see if it makes sense. All the sections where calibrate are mentioned should be reviewed for clarity because as they stand, it would leave any new or experienced user a little confused. 

https://ralim.github.io/IronOS/Settings/#setting-calibrate-cjc-at-next-boot

_"This is used to calibrate the offset between ADC and Op-amp of the tip at next boot (Idealy it has to be done at boot, before internal components get warm.). But this setting is not permanent! It resetes after the calibration is completed (At next boot the checkbox will be unchecked!). If you need to repeat the calibration however, you have to set the checkbox again, unplug your device and let it cool down to room/ambient temperature & power it up, idealy while it sits on the desk."_

River-b, 2022-10-26:
I see also in   Beta 2.20  Calibration is  different than in older firmware.  
Steps seem to now be: 

1. go to advanced,  press (+)  until you check a box 
2. use minus (-)  to get back to main screen
3. unplug and plug pinecil back in
4.  you see  ....  and then an offset number  when you reboot.

If this is the new method, then it would still be good on start up  screen to show some word that says calibration is done.
`850 offset, calibrate done`

Ralim, 2022-10-26:
Yes calibration has been reworked to run at boot to try and get people to do it correctly.

We can probably stop showing the number entirely if that is less confusing?

River-b, 2022-10-26:
> We can probably stop showing the number entirely if that is less confusing?
Is there room to show  offset number and also word  Calibrated? 
like  `833 offset Calibrated`   that would be clear it was done.

discip, 2022-10-27:
As you may have found out, SDG uses the old firmware.

> 3\. when performing a calibration,  when the Offset number appears, make it more clear you are done. In SDG's video example  it could be
>     `833   offset.  Done.  `  or `833  offset. Complete.`

Regarding the latest firmware:
This is not strictly necessary as the **`....`** is only displayed during the calibration and the **`offset value`** indicates that the calibration is complete.
Calibration, in turn, is only triggered when *a tip is installed* **and** *the temperature delta between tip and handle is less than 10°C*.

So if neither is displayed, no calibration was performed.

> * what  is the intent of the above paragraph,  if I understood that, I could probably rewrite this for clarity.  what is meant by this is not permanent?

I already created a [PR](https://github.com/Ralim/IronOS/pull/1451) for that one.

discip, 2022-10-27:
As you may have found out, SDG uses the old firmware.

> 3\. when performing a calibration,  when the Offset number appears, make it more clear you are done. In SDG's video example  it could be
>     `833   offset.  Done.  `  or `833  offset. Complete.`

> Is there room to show offset number and also word Calibrated? like `833 offset Calibrated` that would be clear it was indeed calibrated.
> 
> or if not on that screen then when it's `.....calibrating now` `833`

Regarding the latest firmware:
This is not strictly necessary as the **`....`** is only displayed during the calibration and the **`offset value`** indicates that the calibration is complete.
Calibration, in turn, is only triggered when *a tip is installed* **and** *the temperature delta between tip and handle is less than 10°C*.

So if neither is displayed, no calibration was performed.

> * what  is the intent of the above paragraph,  if I understood that, I could probably rewrite this for clarity.  what is meant by this is not permanent?

I already created a PR for that one. https://github.com/Ralim/IronOS/pull/1451

River-b, 2022-11-09:
> 
@Ralim 
1. rethinking your suggestion that _maybe removing Offset number from view_  might be  less confusing.   say Calibration Done!  or something clear.
2.  SDG  said  something in comments that implied  why would it keep showing large offset number if  you repeat calibration? this happens to me. If I wait  30-60 min unplugged and repeat calibration (everything cool and I only touch pinecil with 1 finger to not heat handle up with hand), why is there always a large offset?  I did not know answer to this. 
3. SDG also said in lower comments that after seeing   833 Offset number.  He still had  15C  off  on the tip using the  _119  temperature tester_ to check.  I suggested he open a ticket here, maybe he has defective tip or something.   and we could diagnose this easier with  him directly.

PR #1473 (2022-11-22): reworked CJC calibration appearance
@River-b
Please check if this is sufficient.

resolves #1448
closed 2022-11-22 by Ralim